aaronwmorris/indi-allsky #337: Request for Displaying Star Counts on Image Text

pimmoon, 2022-08-31:
Hello. 

Thank you so much for creating such a great software. I would like to request for a feature. I'm not sure though if this is the right place to do it. If not, please forgive me.

Currently I use the "Extra Image Text File" to add custom text to the allsky image. That text is my observatory name. However, It would be nice If I could add Star count on the image file as well (so people who view it publicly can see star counts too). I'm not sure how to add dynamic text (how to grab star count info and how to refer to it in the "Extra Image Text File").

Just FYI, I am currently using the Detection Mask feature and the star count seems accurate so far.

Thanks once again.

Looking forward to hearing from you soon.

aaronwmorris, 2022-09-01:
I have merged #343 to customize the image text label.  Star count is one of the available data points.

aaronwmorris, 2022-10-22:
Closing this for now.  Let me know if there is anything else I can do for you.

closed 2022-10-22 by aaronwmorris