aaronwmorris/indi-allsky #339: RPi HQ Camera, Libcamera IMX477 interface, RPi 4 2GB, fresh install, not showing photos

pimmoon, 2022-08-31:
Hello,

I just set up a new indi-allsky. This is my second install. This time I would like to use the RPi HQ Camera on my Raspberry Pi 4 2gb. I set up using the libcamera option.

I have finished installation and verified that the camera worked via using 'libcamera-jpeg' command. I was able to see the image captured as jpg file.

However, I am not able to see image on the INDI-AllSky web interface and here is the log.

Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky.run() #640: Camera last ready: 3.7s
Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky.run() #641: Exposure state: READY
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception: Traceback (most recent call last):
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 149, in run
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception:     self.saferun()
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception:   File "/home/pi/indi-allsky/indi_allsky/image.py", line 248, in saferun
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception:     raise Exception('*** rawpy module not available ***')
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception: Exception: *** rawpy module not available ***
Aug 31 23:13:54 raspberrypi [ERROR] MainProcess allsky._startImageWorker() #457: Image worker exception: 
Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky._startImageWorker() #464: Starting ImageWorker process
Aug 31 23:13:54 raspberrypi [WARNING] MainProcess image._load_detection_mask() #1726: No detection mask defined
Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky.detectNight() #961: Sun altitude: -62:55:38.0
Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky.detectMoonMode() #978: Moon altitide: -27:36:30.5, phase 18.9%
Aug 31 23:13:54 raspberrypi [INFO] MainProcess allsky._queueManualTasks() #1547: Checking for manually submitted tasks

Thank you so much for your help!!

aaronwmorris, 2022-08-31:
Are you running 32-bit or 64-bit Raspbian?

pimmoon, 2022-08-31:
Oh you are right. Thanks. My bad. I'm running 32-bit Raspbian.

I went with libcamera as it got better rating in the "Tested Hardware" section, but I was not careful with the Distibution support list.  For Raspberry Pi HQ Camera with Raspberry Pi 4 2GB,  I should be able to run on 64-bit right? Would you recommend running 64-bit with libcamera?

Thanks again and sorry for a such a silly mistake.

aaronwmorris, 2022-08-31:
It is no big deal.  32-bit will work, but only if you select JPEG or PNG mode for libcamera.  DNG raw mode will only work with 64-bit.

pimmoon, 2022-09-01:
It works now that I changed the file to JPEG. Thank you.

However, I am now having problem with white balance. The auto WB option gives me very green picture. When I tried to set it manually I can not get the right balance. It's over saturated either in Green, or Purple (Red + Blue). I have been playing with WB setting for a couple of hours now but couldn't get it to look natural. I tested the libcamera-still command and from the command the picture looks right (with natural WB).

I have also noticed that the gain setting does't seem to work. The text overlay on picture always show Gain of 1 no matter what gain I set (I have set it to 0,1,50,100 for example)

Please kindly advise.

aaronwmorris, 2022-09-01:
I would try putting the green manual balance around 0.8.  Setting this value will require you to "Reload Config" in the system section to activate.  I have removed the IR/Cut filter from my imx477, so I do not have a good test case for you.

The gain range for the IMX477 is 1 to 16.  Changing this setting requires a full restart of the indi-allsky service.

aaronwmorris, 2022-09-01:
I have been considering allowing the auto white balance option to be enabled with the manual options.  The auto WB works pretty good as long as the colors are not too far off to begin with.  So, if we can manually tone down the green, the auto WB can probably fix the rest of it.

pimmoon, 2022-09-01:
And this is green 0.5, red and blue 1.0. notice purple starts to show. At this point if I reduce blue and red further to try to reduce purple, it will just become more green and darker.

[image]

aaronwmorris, 2022-09-01:
When you run libcamera-still manually, I believe it is taking multiple pictures in the background to determine the most optimal white balance settings.  I found that enabling AWB in libcamera-still added about 5 seconds of additional run time for each image.

indi-allsky works best when dealing with RAW bayer data and RAW bayer data will not have any white balance settings applied because it is supposed to be the RAW image data.

Obviously, there is work to be done on adjusting colors.  My manual color adjustment is just applying a universal multiplier to the color data which is a very simplistic approach and probably not the best.

aaronwmorris, 2022-09-01:
I have merged #345 which will apply the auto WB settings after manual WB (if enabled).

pimmoon, 2022-09-01:
Thank you so much. Now I get why libcamera command is different.

I really appreciate your quick response and really appreciate the software.

aaronwmorris, 2022-09-01:
I forgot to mention that the AWB setting actually have no effect when you get raw data from the camera via libcamera-still.

spensireli, 2022-09-09:
I'm also noticing a green hue with the Pi Camera HQ; additionally during the day everything is purple. AWB doesn't seem to work for the Pi Camera HQ (too much gain it looks like?). If I come across any settings that work out I'll report back. I'm also interested if you had any luck with this since @pimmoon .

pimmoon, 2022-09-14:
Hi Spensireli,

@spensireli I have updated to the latest code where the Auto White Balance is allowed after manual White Balance. I have tried many combination of the white balance settings. (from 0.1 - 2 for each red, green, and blue) but I am still not able to get accurate color, it's still either very green or very purple, or both.

aaronwmorris, 2022-09-16:
I have merged #355 with a new option to apply some SNCR algorithms borrowed from PixInsight.  These seem to do a good job removing the green bias.

Make sure you return your WB settings the default.  SCNR is applied before the white balance settings.

https://github.com/aaronwmorris/indi-allsky/wiki/Image-order-of-operations

pimmoon, 2022-09-17:
Thanks so much for a quick action. I just updated the new code with SCNR (green reduction) setting in the config page. 

Here are the results:

When disabled

[image]

when set to average neutral
[image]

when set to maximum neutral
[image]

 The Maximum Neutral one looks the best, however the image is grainy.

This is Maximum Neutral + AWB
[image]

Just want to report the results. Looking forward to hearing from you soon :)

aaronwmorris, 2022-09-17:
Those look much better.

The grain could be because you are using JPEG mode.  The problem with JPEG mode is the image from the camera is already a lossy compressed JPEG.  indi-allsky reads the JPEG, processes it, adds the text, and then re-compresses it again as a JPEG.  This can lead to a lot of noise.

If you installed Raspbian 11 64-bit and used the DNG raw mode (and take dark frames), the result will be much better.

spensireli, 2022-09-20:
Great, I will give this a try and post results.

aaronwmorris, 2022-12-07:
Closing this issue for now.  Let me know if I can help with anything else.

closed 2022-12-07 by aaronwmorris