aaronwmorris/indi-allsky #340: Stretching images

t-ara-fan, 2022-08-31:
Hi,
I just started reading about indi-allsky. Nice work - it looks really good, I am installing it right now.

I have a question I could not find an answer to while scrolling through the project.  Do you have an option to automatically "stretch" images to boost contrast of features like the Milky Way?  I have dabbled with some AllSkyCamera software.  I use openCV to stretch the image, and it really boosts the Milky Way, aurora, and faint meteors.

The stretch is quite simple. A sample photo with Perseid meteor is attached. 

[image]

t-ara-fan, 2022-08-31:
Great, I got it up and running. (Off topic: after step 6 in your instructions "Login to the indi-allsky web application https://raspberrypi.local/"  it took me a couple minutes to figure out the user/password.  I was trying my "pi" user. Then I remembered I saw admin/secret somewhere and that worked.)

OK so now I have it running, and I see that you have CLAHE contrast enhancement available. Right now my camera is inside my house, and CLAHE gives an impressive contrast boost. I will try CLAHE under the stars when I get a chance.

aaronwmorris, 2022-09-01:
The CLAHE definitely produces some interesting results.  I find that daytime contrast enhancing is pretty interesting, especially with clouds.  It is not really a proper stretching algorithm.

I am actually interested in implementing a stretching feature, but I have not been able to find a good reference to implement one.  If you can point me at some code or an example, I will see about getting it integrated.

t-ara-fan, 2022-09-01:
Hi Aaron,
Thanks for your prompt response.  I will send you my code that does the "stretch" along with an explanation.

aaronwmorris, 2023-01-02:
I have made some progress testing the stretching algorithms included with Astropy.
https://docs.astropy.org/en/stable/visualization/normalization.html

"Progress" insomuch as I now know how to use them... mostly.  :-)  Some of the algorithms are pretty expensive in terms of CPU.

I would still be happy to take a look at any code you would be willing to contribute.

aaronwmorris, 2023-02-17:
Closing this issue for now.

closed 2023-02-17 by aaronwmorris