aaronwmorris/indi-allsky #342: Dark frames affecting day imaging?

laterrible, 2022-09-01:
Hello - encountered a fresh error, probably user created. I created my first batch of dark frames this morning, process worked perfectly. However all of my subsequent day images were black. The details of each image (exposure, gain time) was still imprinted, the service was functioning just fine.

I flushed the dark images, and immediately I returned to regular day time imaging. Is there something I overlooked in the process?

SV305Pro
Pi4 on Ubuntu, freshly updated

aaronwmorris, 2022-09-01:
I would have to see an example dark frame, but I have never had an issue with daytime darks.  I would try to take the dark frames again and see what happens.

laterrible, 2022-09-01:
Thanks Aaron, will do so tonight, weather permitting. Let you know. I'm wondering if perhaps the images were simply being under exposed, but I'll test.

Aside from that, running beautifully 👍

aaronwmorris, 2022-12-07:
Closing this issue for now.  Let me know if I can help with anything.

closed 2022-12-07 by aaronwmorris