aaronwmorris/indi-allsky #353: setup.sh fails with cryptography build error

AllSkyPiRA, 2022-09-13:
RPi 3B+, latest Astroberry image, trying to add indi-allsky

I have seen some recent commits which seem to address this issue but I assume they've been picked up during the fetch and update process.

I've tried a few workarounds but can't seem to get this to work.

Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  Building wheel for cryptography (pyproject.toml) did not run successfully.
  exit code: 1

.
.
.

running build_rust  /tmp/pip-build-env-vdj3frqa/overlay/lib/python3.7/site-packages/setuptools/command/build_py.py:202: SetuptoolsDeprecationWarning:     Installing 'cryptography.hazmat.bindings._rust' as data is deprecated, please list it in `packages`.      !!          ############################      # Package would be ignored #      ############################      Python recognizes 'cryptography.hazmat.bindings._rust' as an importable package,      but it is not listed in the `packages` configuration of setuptools.        'cryptography.hazmat.bindings._rust' has been automatically added to the distribution only      because it may contain data files, but this behavior is likely to change      in future versions of setuptools (and therefore is considered deprecated).        Please make sure that 'cryptography.hazmat.bindings._rust' is included as a package by using      the `packages` configuration field or the proper discovery methods      (for example by using `find_namespace_packages(...)`/`find_namespace:`      instead of `find_packages(...)`/`find:`).        You can read more about "package discovery" and "data files" on setuptools      documentation page.      !!      check.warn(importable)        =============================DEBUG ASSISTANCE=============================      If you are seeing a compilation error please try the following steps to      successfully install cryptography:      1) Upgrade to the latest pip and try again. This will fix errors for most         users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip      2) Read https://cryptography.io/en/latest/installation/ for specific         instructions for your platform.      3) Check our frequently asked questions for more information:         https://cryptography.io/en/latest/faq/      4) Ensure you have a recent Rust toolchain installed:         https://cryptography.io/en/latest/installation/#rust        Python: 3.7.3      platform: Linux-5.10.103-v7+-armv7l-with-debian-10.13      pip: n/a      setuptools: 65.3.0      setuptools_rust: 1.5.1      rustc: 1.41.1      =============================DEBUG ASSISTANCE=============================    error: Rust 1.41.1 does not match extension requirement >=1.48.0  [end of output]    note: This error originates from a subprocess, and is likely not a problem with pip.  ERROR: Failed building wheel for cryptographyFailed to build cryptographyERROR: Could not build wheels for cryptography, which is required to install pyproject.toml-based projects

The script exited abnormally, please try to run again...

aaronwmorris, 2022-09-13:
Merged #354 to restrict the version of the cryptography module.

Looks like newer versions of the cryptography module are not building on Debian 10/Python 3.7.

AllSkyPiRA, 2022-09-14:
Excellent. indi-allsky built. Thanks for really swift response.

closed 2022-09-14 by AllSkyPiRA