aaronwmorris/indi-allsky #356: Feature: indi-allsky Docker Container

spensireli, 2022-09-20:
I'd be willing to give a crack at Dockerizing this application and writing a compose file for it. Before I did I wanted to check if this wasn't already started somewhere.

aaronwmorris, 2022-09-20:
I have not done any work related to containerization yet.

Elliot-Ford, 2023-01-11:
I did some digging into this as this is a feature-set that I'd like to see as well. The biggest initial hurdle that I encountered would be removing and replicating the systemd/dbus calls found in a couple of places (namely the `views.py`).
I've seen some discussion on how one might go about replicating this with a container, but haven't done any experimentation with it yet

aaronwmorris, 2023-02-02:
I have done some work toward using docker.  You can see what I have in the `docker_testing/` folder.  Nothing ground breaking there.  I also updated the setup.sh script to run non-interactive using settings from environment variables.

@Elliot-Ford is right about the systemd stuff, but with a slightly different spin from my point of view.  Trying to combine all of the services into a single container is not really a [docker] best practice.  It can be done, but it is not the docker way.

indi-allsky could be split into as many as 4 containers:

1. indiserver
2. indi-allsky capture process
3. gunicorn web app
4. Apache/Nginx reverse proxy (this could be optional)

(with shared volumes between 2, 3, and 4)

I suppose nobody would really care about combining all of these into one container if it is just for testing.

I do think that using the setup.sh in the docker build is not going to work.  I think I will just write a custom setup process for the docker container and try to keep the setup.sh and docker builds in sync.