aaronwmorris/indi-allsky #358: Extra text on image

alexdumitriu, 2022-09-23:
Hi,

Thank you for creating this project, it's absolutely amazing!

I have just a quick question - I'd like to add a bit of extra text on the image (moon alt and moon phase in percentage). I see the information is in views.py, could you point me in a direction of making them available on the image?

Many thanks
Alex

aaronwmorris, 2022-09-23:
Let me see if I can add that info to the template.

aaronwmorris, 2022-09-23:
Merged #359 with a few extra variables for the image label.

The variables are documented here:  https://github.com/aaronwmorris/indi-allsky/wiki/Image-Labels

alexdumitriu, 2022-09-23:
Hi Aaron,

That's fantastic, thank you so much for the fast implementation of these!

Is there a way to get you a coffee anywhere? :)

Cheers
Alex

alexdumitriu, 2022-09-23:
Hi Aaron,

I'm getting some errors when implementing the changes:

Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: Traceback (most recent call last):
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 152, in run
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     self.saferun()
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 405, in saferun
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     self.image_text(scidata, exposure, exp_date, exp_elapsed, blob_stars, image_lines)
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 1112, in image_text
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     'sun_moon_sep' : sun_moon_sep,
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: KeyError: 'sun_alt'
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: 

**Then, after removing the 'sun_moon_sep' in label_data, I get the following:**

Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: Traceback (most recent call last):
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 152, in run
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     self.saferun()
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 405, in saferun
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     self.image_text(scidata, exposure, exp_date, exp_elapsed, blob_stars, image_lines)
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:   File "/home/allsky/indi-allsky/indi_allsky/image.py", line 1112, in image_text
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception:     'sun_moon_sep' : sun_moon_sep,
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: KeyError: 'sun_alt'
Sep 23 20:51:19 allsky [ERROR] MainProcess allsky._startImageWorker() #453: Image worker exception: 

If I remove the new variables in configuration, the images start being displayed again.

Thanks,
Alex

aaronwmorris, 2022-09-23:
Did you perform a full restart of indi-allsky or just a reload?

alexdumitriu, 2022-09-23:
Full system restart now, all working!

Thanks a lot :)
Alex

closed 2022-09-26 by alexdumitriu