aaronwmorris/indi-allsky #361: timelapse

dave21850, 2022-09-25:
bonjour a tous j'ai ce défaut qui apparait 5 2022-09-25 08:35:32 VIDÉO ÉCHEC Le dossier d'image n'existe pas : /var/www/html/allsky/images/20220925/night pouvez vous m'aider merci 
[image]

aaronwmorris, 2022-09-25:
I think you tried to generate the video too early since there were no night time images yet for 2022-09-25 (today)

aaronwmorris, 2022-09-25:
This may not be clear, but the date for a night time timelapse is always the date at the start of the night (not the end of the night).

dave21850, 2022-09-26:
MERCI POUR VOS REPONSE MAIS LA CREATION DU TIMELAPSE NE MARCHE TOUJOURS PAS 
[image]
[image]
[image]

aaronwmorris, 2022-09-26:
If you are using the Raspi HQ camera (IMX477) or the IMX378, you probably need to enable the FFMPEG Scaling option to make the resulting video smaller.

[image]

Once you enable that, delete the existing keogram/startrails and and try to regenerate.

dave21850, 2022-09-26:
j'ai changé les paramètres mais toujours pas de timelapse !
[image]
[image]

aaronwmorris, 2022-09-26:
Make sure you "Reload" the configuration or just restart your raspberry pi after to change the config options.  The options are not changed until you reload the config.

dave21850, 2022-09-26:
bonsoir c'est ce que j'ai fait mais toujours rien désolé

aaronwmorris, 2022-09-26:
You will have to look at the logs at `/var/log/indi-allsky/indi-allsky.log` and look for error messages.

What type of Raspberry Pi do you have?

dave21850, 2022-09-27:
UN Raspberry Pi  4 2 G RAM

dave21850, 2022-09-27:
Sep 27 07:03:23 raspberrypi [WARNING] ImageWorker003 image.calibrate() #880: Temperature matched bad pixel map not found: -273.15c
Sep 27 07:03:23 raspberrypi [WARNING] ImageWorker003 image.calibrate() #898: Bad Pixel Map not found: ccd1 16bit 4.6067233s gain 1 bin 1 -273.15c
Sep 27 07:03:23 raspberrypi [INFO] ImageWorker003 image.calibrate() #910: Searching for dark frame: gain 1, exposure >= 4.6, temp >= -273.1c
Sep 27 07:03:23 raspberrypi [WARNING] ImageWorker003 image.calibrate() #927: Temperature matched dark not found: -273.15c
Sep 27 07:03:23 raspberrypi [WARNING] ImageWorker003 image.calibrate() #945: Dark not found: ccd1 16bit 4.6067233s gain 1 bin 1 -273.15c

aaronwmorris, 2022-09-27:
After trying to generate a timelapse, see if you have any out of memory (OOM) messages in the system log using `sudo dmesg`.

2GB of RAM may not be enough memory to generate a 4056×2304 timelapse.  I may have to add another smaller resolution for your situation.

dave21850, 2022-09-27:
merci pour votre aide pouvez faire une plus petite résolution ?

aaronwmorris, 2022-09-27:
Not a problem.  I merged #363 with two additional down-scaling options.

dave21850, 2022-09-28:
SUPER CA MARCHE MERCI BEAUCOUP POUR VOTRE AIDE !!!!!!

aaronwmorris, 2022-10-22:
Closing this issue for now.  Let me know if I can help again!

closed 2022-10-22 by aaronwmorris