aaronwmorris/indi-allsky #366: Day--> Night transition stops image capture

pterodyne, 2022-09-30:
Repost from issue 225:

Im not sure whether or not to reopen this or create a new issue. I am using an ASI ZWO 178mm/pi4, and have this issue with transitioning to night. Strangely I had the same issue with TJacquin's allsky. In both cases turning off daytime imaging solves the problem permanently. BUT of course no daytime images. Is this an SDK issue? Seems strange that it would be present in indi and a SDK native application. In the other project people suggested a hardware problem, but this setup will operate for months, night after night without issue as long as I'm not transitioning from day to night. Seems to point to software, libraries, sdk or algorithms. At any rate, I don't really need daytime images. It would be nice if it displayed a NO IMAGES DAYTIME or something during the day if disabled. Mine just makes an over exposed image all day, maybe not good for the sensor, but who knows. It's been deployed for years.

Attached is my config.json

[config.json.txt](https://github.com/aaronwmorris/indi-allsky/files/9687293/config.json.txt)

aaronwmorris, 2022-09-30:
Your minimum exposure is way too high for day time captures.  I would set your minimum capture down to 0 which means it will be auto-detected.  The minimum for the camera is probably 0.000032 or 0.000064.

You can also set your daytime gain to 0, that will help a little.

pterodyne, 2022-10-01:
Yes, sorry, that is just what it is currently set to.  The problem occurs when I set it correctly, then at night time it is stuck at .000032.  And that is using auto for minimum.  So then Id set the minimum to .000032 and it would still stick.  So I had to set it higher to get it to be stable.  Just to clarify to myself, I set the minimum and starting to .000064, and Ill report back

aaronwmorris, 2022-10-01:
You can always leave the default at 0.1 and set the minimum to 0.  The default exposure is the starting exposure.  I can then scale up and down from the default.

pterodyne, 2022-10-03:
So it has been stable for a few days.  It seems like when it was broken was with the default and minimum both set to .000034

closed 2022-10-03 by pterodyne

aaronwmorris, 2022-10-03:
I am glad to hear it is working!