aaronwmorris/indi-allsky #373: Timelapse fine, but StarTrail and Keogram fail

StarGeezerPhil, 2022-10-17:
Hi everyone,

IMX378, with following video settings:

- Crop: 471,0 (x1,y1) 3620,3404 (x2,y2)
- ADU/Region of Interest - all using a mask from a raw image.
- Image file type: JPEG, 98%
- FFMPEG: 25fps, 4000k bitrate, V 1520px (imx477) scaling

StarTrail and Keograms fail to produce every day/night, but the timelapse generates absolutely fine.

Crop/ADU/Region of Interest - all using a mask from a raw image.
Image file type: JPEG, 98%
FFMPEG: 25fps, 4000k bitrate, V 1520px (imx477) scaling

Any idea what the cause here is:

```
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 timelapse.generate() #70: Timelapse generated in 215.8123 s
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 timelapse.generate() #72: FFMPEG output: b"ffmpeg version 4.3.4-0+deb11u1+rpt3 Copyright (c) 2000-2021 the FFmpeg developers\n  built with gcc 10 (Debian 10.2.1-6)\n  configuration: --pre>Oct 17 22:06:35 stargeezer [WARNING] VideoWorker001 video.uploadVideo() #246: Video uploading disabled
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 video.generateKeogramStarTrails() #336: Found 736 images for keogram/star trails
Oct 17 22:06:35 stargeezer [WARNING] VideoWorker001 miscDb.addKeogram() #264: File not found: /var/www/html/allsky/images/20221016/allsky-keogram_ccd1_20221016_night.jpg
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 miscDb.addKeogram() #267: Adding keogram /var/www/html/allsky/images/20221016/allsky-keogram_ccd1_20221016_night.jpg to DB
Oct 17 22:06:35 stargeezer [WARNING] VideoWorker001 miscDb.addStarTrail() #298: File not found: /var/www/html/allsky/images/20221016/allsky-startrail_ccd1_20221016_night.jpg
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 miscDb.addStarTrail() #301: Adding star trail /var/www/html/allsky/images/20221016/allsky-startrail_ccd1_20221016_night.jpg to DB
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 video.generateKeogramStarTrails() #374: Processed 0 of 736 images
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 keogram.processImage() #116: New Shape: (2939, 1, 3)
Oct 17 22:06:35 stargeezer [INFO] VideoWorker001 keogram.processImage() #119: New dtype: uint8
Oct 17 22:06:36 stargeezer [INFO] MainProcess allsky.run() #636: Camera last ready: 53.9s
Oct 17 22:06:36 stargeezer [INFO] MainProcess allsky.run() #637: Exposure state: BUSY
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception: Traceback (most recent call last):
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:   File "/home/stargeezer/indi-allsky/indi_allsky/video.py", line 74, in run
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:     self.saferun()
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:   File "/home/stargeezer/indi-allsky/indi_allsky/video.py", line 153, in saferun
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:     self.generateKeogramStarTrails(task, timespec, img_folder, timeofday, camera_id)
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:   File "/home/stargeezer/indi-allsky/indi_allsky/video.py", line 395, in generateKeogramStarTrails
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:     stg.processImage(p_entry, image)
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:   File "/home/stargeezer/indi-allsky/indi_allsky/starTrails.py", line 113, in processImage
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:     m_avg = cv2.mean(image_gray, mask=self._sqm_mask)[0]
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception: cv2.error: OpenCV(4.6.0) /io/opencv/modules/core/src/matrix_iterator.cpp:71: error: (-215:Assertion failed) A.size == arrays[i0]->si>Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:
Oct 17 22:06:36 stargeezer [ERROR] MainProcess allsky._startVideoWorker() #503: Video worker exception:
Oct 17 22:06:36 stargeezer [INFO] MainProcess allsky._startVideoWorker() #510: Starting VideoWorker process
```

aaronwmorris, 2022-10-17:
Okay, I see the problem.

It is trying to apply the [full size] mask to the cropped image to generate the star trails.  I need to crop the mask to match the cropped image.

aaronwmorris, 2022-10-18:
Merged #374 to fix this.

This should work in theory, but I have not done much testing.  Please let me know if it works.

StarGeezerPhil, 2022-10-23:
Can't thank you enough for the prompt response and fix 👍💖

[image]

Couple of, nice to have, ideas - don't know if it's easily feasible or you'd be interested:
- an adhoc generate GIF function (select period - 15/30/60 mins/hour etc)
- time lapse addition generation, this is a beautiful method I've seen some using, where the video shows the startrail building and fading (e.g. combining 1-3 hours max then the older images fading out of the stack). No scientific benefit, but beautiful.

closed 2022-10-23 by StarGeezerPhil

aaronwmorris, 2022-10-24:
> an adhoc generate GIF function (select period - 15/30/60 mins/hour etc)
This is something I have been wanting to add for a while.  I just need to figure out the best way to implement.

> time lapse addition generation, this is a beautiful method I've seen some using, where the video shows the startrail building and fading (e.g. combining 1-3 hours max then the older images fading out of the stack). No scientific benefit, but beautiful.

I have seen these on facebook.  It should not be too difficult to create something like this.

aaronwmorris, 2022-10-25:
I still have a few commits to make before I officially release this...  Here are some star trail timelapses I generated from two of my cameras.

https://youtu.be/pLJbTzlyBkM
https://youtu.be/QhQZURdiTPU

StarGeezerPhil, 2022-10-25:
Absolutely stunning - yes, exactly this. 

Can't wait! 🔥