aaronwmorris/indi-allsky #378: Indi-AllSky stops working when using an Altair GPCam 290C

vastmeridian, 2022-10-24:
Processor:  **SECO picoITX industrial computer c/w Intel Atom N2800 - 4GB RAM and 32G SSD**
Camera:     **Altair GPCam3 290C**
Operating System: **Ubuntu 20.04.5 LTS (GNU/Linux 5.4.0-131-generic x86_64)**
indi-allsky: **Release 5.2 (from #377)**

**Background**: I was originally able to get the Altair GPCam3 290C to run on the indi-allsky with full functionality, but changed camera to a ZWO ASI120MC.  This proved to not function reliably - and was too noisy - so the decision was made to return to the 290C.

**Issue**: Following discussions on CloudyNights with Aaron, who has kindly developed an extension to the remote file upload function, I updated and downloaded first #375 and then #377, but, in each case, the camera is detected, runs for a few seconds and then stops.  No images are taken or stored.  I have even re-installed the operating system, thinking to make sure the environment is clear of any leftovers from using the ASI120MC, but the effect is the same.

Reviewing the indi-allsky log file (attached) from /var/log/indi-allsky shows three processes caught the TERM signal and shut down:
VideoWorker001 allsky.sigterm_handler() #227: Caught TERM signal, shutting down
ImageWorker001 allsky.sigterm_handler() #227: Caught TERM signal, shutting down
FileUploader001 allsky.sigterm_handler() #227: Caught TERM signal, shutting down

[20221023 - indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/9850729/20221023.-.indi-allsky.log)

Happy to help debug, but I would need some guidance as to what to try regarding settings.
Thanks
Martin

aaronwmorris, 2022-10-25:
Can you post the output of `indi_getprop` so I can see what options are available for your camera?

Also, try adding these settings in the indi config at the bottom.

```
{
    "PROPERTIES" : {},
    "SWITCHES" : {
        "CCD_CAPTURE_FORMAT" : {
            "on"  : ["INDI_RAW"],
            "off" : ["INDI_RGB"]
        }
    }
}
```

vastmeridian, 2022-10-27:
[20221027 output of indi_getprop.txt](https://github.com/aaronwmorris/indi-allsky/files/9882794/20221027.output.of.indi_getprop.txt)

I've attached the output of indi_getprop for the 290C.  I will try the indi config suggestions and report back.  I've three other Altair cameras so I can pass indi-getprop outputs for each of them, if this might help you support Altair cameras.  Let's get this issue resolved first though.

vastmeridian, 2022-10-27:
I tried the suggested config above and it still fails - but the failure is now different.  I've attached the indi-allsky log as before.  Everything is in the log from the reboot.

[20221027 - indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/9883143/20221027.-.indi-allsky.log)

I have tried the camera once more on Windows (NINA, Altair Capture under Ascom) and it works fine.  I have also tried a beefier power supply and different USB leads too, just in case.

vastmeridian, 2022-10-27:
Now I have restarted indi-allsky, we have a picture!  Hooray!  So it looks very much like the same issue as #191, Aaron

vastmeridian, 2022-10-27:
Do you want the issues tracked as one Aaron?  Or do we keep them separate for the moment?

aaronwmorris, 2022-10-27:
I am okay keeping them separate.  The issues are cross referenced, so I think it is fine.

vastmeridian, 2022-10-27:
OK.  In the meantime, I'll try my other Altair cameras for you.  My complete list is:

- GPCAM2 130M (mono)
- GPCAM3 290C
- Hypercam 178C
- Hypercam ProTEC 533C

I'll happily test these for you once you have progressed at your end.

aaronwmorris, 2022-10-27:
Awesome, I appreciate the info.  I am guessing the Hypercam models probably use the touptek server.

vastmeridian, 2022-10-27:
I'll add that to my checks.  I know the GPCam3 290C wasn't found when I tried the Touptek drivers (**indi.findCcd() found 0 CCDs**)

Would running indi-allsky in a Virtualbox VM be a valid test environment?  It'll be easier than playing with the rather slow Atom-based SBC I have - and much faster to switch configs too.

aaronwmorris, 2022-10-28:
You can always manually start the indisever with the server you want.

```
systemctl --user stop indiserver

indiserver indi_altair_ccd

indiserver indi_toupcam_ccd
```

You can also have multiple config files for indi-allsky and start the process manually and override the config used.

```
./allsky.py -c /etc/indi-allsky/config2.json
```

vastmeridian, 2022-10-28:
Good shout. I'll keep that in mind.

aaronwmorris, 2022-11-02:
Opened https://github.com/indilib/indi-3rdparty/issues/681

vastmeridian, 2022-11-03:
Magic.  Thank you.

vastmeridian, 2022-11-04:
Just to let you know, I generated a Virtualbox VM and set up a test bed to do the other Altair cameras on indi-allsky.  I've tested the USB3-based Hypercam 178C and it works fine out of the box, with no changes to the default config - apart from not reporting the sensor temperature and needing the gain tweaked etc.

If it helps, here's the indi config:
```
Indi_config:
{
    "PROPERTIES": {},
    "SWITCHES": {}
}
```
and here's the output from indi_getprop:
[output of indi_getprop (Altair Hypercam 178C).txt](https://github.com/aaronwmorris/indi-allsky/files/9939714/output.of.indi_getprop.Altair.Hypercam.178C.txt)

It has run overnight perfectly for almost 24 hours, capturing day, moon and night mode.

vastmeridian, 2022-11-04:
The Altair GP130M wasn't so successful.  It allows indi-allsky to load and run, but no images.  Maybe it needs a different start-up setting.  I've not looked further into this I'm afraid.

I get very little from indi_getprop:
```
username@allskyvm:~/indi-allsky$ indi_getprop
No *.*.* from localhost:7624
username@allskyvm:~/
```
I've attached the indi-allsky log for when the GP130M was connected
[allskyLog-GP130M.log](https://github.com/aaronwmorris/indi-allsky/files/9942502/allskyLog-GP130M.log)

aaronwmorris, 2022-11-05:
Very interesting about the 178C.  I almost purchased the 178C as my test camera because I figured they would all have similar behavior.  I am glad I got the 224C for testing.  (I also do not have another IMX224 based camera, which is one of the reasons I picked it)

As for the GP130M... you may want to run from the command line.  There may be an exception that is not logged.

```
source virtualenv/indi-allsky/bin/activate
./allsky.py run
```

aaronwmorris, 2022-11-05:
Also, if you get a chance, can you run `indi_getprop` again with the 290C but after you have run it with indi-allsky once.  The camera needs to be in a "connected" state with the indiserver to report full information.

vastmeridian, 2022-11-05:
> Very interesting about the 178C. I almost purchased the 178C as my test camera because I figured they would all have similar behavior. I am glad I got the 224C for testing. (I also do not have another IMX224 based camera, which is one of the reasons I picked it)
> 
Ah.  About the 178C.  Mine is a model that's no longer made.  It's their Hypercam 178C.  It's fan-cooled..
If you'd bought a 178C, you'd have bought a GPCam3 178C, which may be a different beast, in terms of support.

I'll try the suggestions regarding the GP130M and I'll restart the 290C from a cold power on and do the indi_getprop too.  It might be worth trying the 290C from the command line too, in case there's an exception there, too.

vastmeridian, 2022-11-06:
> Also, if you get a chance, can you run `indi_getprop` again with the 290C but after you have run it with indi-allsky once. The camera needs to be in a "connected" state with the indiserver to report full information.

Here you go.  Two files for you.  First file is the **indi_getprop** after a power-on of the system
[Power on output of indi_getprop (Altair GPCam290C).txt](https://github.com/aaronwmorris/indi-allsky/files/9945907/Power.on.output.of.indi_getprop.Altair.GPCam290C.txt)

Second file is the **indi_getprop** after a restart of indi_allsky from the web site
[ServiceRestart output of indi_getprop (Altair GPCam290C).txt](https://github.com/aaronwmorris/indi-allsky/files/9945908/ServiceRestart.output.of.indi_getprop.Altair.GPCam290C.txt)

There are a couple of immediate differences I can see, but how relevant they are I do not know.

I can also start a dialogue with Altair here in the UK if you want, now that you've raised the issue with the indi team. Let me know if you want any more info.

vastmeridian, 2022-11-06:
> Opened [indilib/indi-3rdparty#681](https://github.com/indilib/indi-3rdparty/issues/681)

I notice that issue you raised has been closed.  Please let me know when there is something you'd like me to test here with indi_allsky and the 290C.

vastmeridian, 2022-11-06:
> As for the GP130M... you may want to run from the command line. There may be an exception that is not logged.
> 
> ```
> source virtualenv/indi-allsky/bin/activate
> ./allsky.py run
> ```
I have now done this and attach the screenscrape from the PuTTY session:
[GP130M - Running indi-allsky from command line.txt](https://github.com/aaronwmorris/indi-allsky/files/9946261/GP130M.-.Running.indi-allsky.from.command.line.txt)

You can see where I logged on over SSH and stopped both services then tried to start allsky.py and got an error.
I then restarted indiserver and tried again.  That's better... Until you can see a line in the screenscrape where it says **CCD_CAPTURE_FORMAT: defSwitchVector with no valid members**  - I don't recall seeing that in the log file from my previous attenpt.
```
2022-11-06 16:14:35,658 [INFO] MainProcess indi.newSwitch() #225: new Switch CONNECTION for device Altair ALTAIRGP130M
CCD_CAPTURE_FORMAT: defSwitchVector with no valid members
2022-11-06 16:14:36,429 [INFO] MainProcess indi.updateCcdBlobMode() #248: Set BLOB mode
2022-11-06 16:14:36,581 [INFO] MainProcess indi.newSwitch() #225: new Switch DEBUG for device Altair ALTAIRGP130M
2022-11-06 16:14:37,434 [INFO] MainProcess indi.configureDevice() #444: Setting switch CCD_FRAME_TYPE
2022-11-06 16:14:37,556 [INFO] MainProcess indi.newSwitch() #225: new Switch CCD_FRAME_TYPE for device Altair ALTAIRGP130M
2022-11-06 16:14:38,758 [INFO] MainProcess indi.getCcdInfo() #356: Detecting bayer pattern
```

aaronwmorris, 2022-11-07:
I think you have custom indi options left over from your other cameras which causes the GP130M initialization hang.

You can use `#` to "comment out" configs in the json.  Like so....

```{
    "PROPERTIES" : {},
    "SWITCHES" : {},
    "#SWITCHES" : {
        "CCD_CAPTURE_FORMAT" : {
            "on"  : ["INDI_RAW"],
            "off" : ["INDI_RGB"]
        }
    }
}
```

vastmeridian, 2022-11-07:
I thought that, so I checked:

```
{
    "PROPERTIES": {},
    "SWITCHES": {}
}
```
Still doesn't work.  indi-allsky is up and running but no images.  I'll continue the investigation.

aaronwmorris, 2022-11-07:
The good news is Jasem able to fix the issue with the Altair cameras.  I assume the fix will be included in the next release of indi, v1.9.9.

You can test now by compiling indi and indi-3rdparty from the HEAD branch in git.  If you are using the apt repo, you will have to remove those packages before compiling.

vastmeridian, 2022-11-08:
Which would help you more?  Me compiling my own version of the indi driver/server? Or helping you test the updated versions of indi-allsky?

aaronwmorris, 2022-11-08:
I think validating the cameras are working with the latest indi would be the most helpful.  You will also require the latest version of indi-allsky to do so.  Two birds with one stone.  :-)

vastmeridian, 2022-12-17:
I will rebuild the indi_allsky with the latest version and try this out, Aaron.

vastmeridian, 2022-12-21:
Thanks Aaron.  It's up and running now which is a giant step in the right direction  I'll need to re-think the config to make use of your new features, so I'll do that and report back.  I'm doing things to the computer remotely at the moment, so I will need to get in front of it to complete my testing.  I will let you know as soon as I can.

FYI - the INDI config I am using for the Altair 290C is:
```
{
    "PROPERTIES": {},
    "SWITCHES": {
        "CCD_RESOLUTION": {
            "on": [
                "1920 x 1080"
            ]
        },
        "CCD_CAPTURE_FORMAT": {
            "on": [
                "INDI_RAW"
            ]
        }
    }
}
```

vastmeridian, 2022-12-27:
Just thought I'd post an image I took under test (i.e. shot through a double-glazed window as the system is laid out on my desk at the moment ... )  You can even see the orange Betelgeuse
[image]

It's looking good, Aaron, apart from one small thing - switching from day mode to night mode when I have switched off daytime capture.  I just don't get any images when the system switches back to night mode.  The system seems to stay 'Running' but all I get is 'Image out of date'.  I've not really had a chance to investigate with the holiday season upon us, but I will do so as soon as I can.  I vaguely remember there is a setting or an investigation relating to this, but can't remember where I saw it.

aaronwmorris, 2022-12-27:
Very nice!

Are you sure you are not seeing new images?  The "Image out of date" message is set when the page is first loaded (based on the timestamp of the last image).  If you let it sit there until night time, it should start loading the next image, but the message will remain until the page is reloaded.

vastmeridian, 2022-12-27:
Yes - that was what I expected but it doesn't seem to re-start.
As I said, holiday duties (including the birth of a grandson on Boxing Day 👍) will delay any proper investigation.

vastmeridian, 2023-01-02:
SIgh.  Not sure if this is related, but the camera stopped logging images in the last day or so.  I'm seeing this in the log:
`Jan  2 21:25:58 allsky [ERROR] VideoWorker002 video.saferun() #155: Image folder does not exist: /var/www/html/allsky/images/20230102/day`
I'm not out of disk space (at least, not according to my monitoring) just - no image directroy created - and nothing uploaded to my FTP server either (I suspect this is related to the above!)

aaronwmorris, 2023-01-02:
Is this happening for night time videos or just day time?  I would have to see more of the log to make any determinations.

I recently merged #503 which includes a notification system which might help stay on top of camera issues.

vastmeridian, 2023-01-02:
The log file reports that  *just* the day folder doesn't exist - but the image folder doesn't have folders for 20230101 or for 20230102.

I ran this command on the gzipped log files:
```
username@allsky:/var/log/indi-allsky$ zgrep ERROR indi-allsky.log-20230101.gz | more
Dec 31 16:50:36 allsky [ERROR] VideoWorker002 video.saferun() #155: Image folder does not exist: /var/www/html/allsky/images/20221231/day
Dec 31 16:50:36 allsky [ERROR] VideoWorker002 video.saferun() #155: Image folder does not exist: /var/www/html/allsky/images/20221231/day
username@allsky:/var/log/indi-allsky$
```

Here's the result from today for ls /var/www/html/allsky/images
```
username@allsky:/var/www/html/allsky/images$ ls
20221221  20221222  20221223  20221224  20221225  20221226  20221227  20221228  20221229  20221230  20221231  darks  export  latest.jpg
username@allsky:/var/www/html/allsky/images$
```
I'll do some digging to get the log file from around the time the first error was logged.

vastmeridian, 2023-01-02:
Here's the log file for the last couple of minutes of 2022
[indi-allsky.log-last-two-minutes-of-2022.txt](https://github.com/aaronwmorris/indi-allsky/files/10333693/indi-allsky.log-last-two-minutes-of-2022.txt)
And here's a log from around the time of latest.jpg, whose ls info reads:
`-rw-r--r--  1 username username 183148 Jan  1 11:33 latest.jpg`
[indi-allsky.log-around 11.33 (latest.jpg time).txt](https://github.com/aaronwmorris/indi-allsky/files/10333714/indi-allsky.log-around.11.33.latest.jpg.time.txt)

aaronwmorris, 2023-01-02:
```
Jan  1 11:34:02 allsky [INFO] MainProcess allsky.detectNight() #1297: Sun altitude: 15:37:39.8
Jan  1 11:34:02 allsky [INFO] MainProcess allsky.detectMoonMode() #1314: Moon altitide: -9:40:38.4, phase 74.4%
Jan  1 11:34:02 allsky [INFO] MainProcess indi.getCcdTemperature() #778: Sensor temperature: 21.1
Jan  1 11:34:02 allsky [INFO] MainProcess indi.getTelescopeRaDec() #763: Telescope Coord: RA 18.17, Dec 50.99
Jan  1 11:34:02 allsky [INFO] MainProcess allsky._queueManualTasks() #2066: Checking for manually submitted tasks
Jan  1 11:34:02 allsky [INFO] MainProcess indi.removeDevice() #201: remove device Altair ALTAIR290C3(USB2.0)
Jan  1 11:34:03 allsky [INFO] MainProcess indi.newDevice() #198: new device Altair ALTAIR290C3(USB2.0)
```

It looks like something caused the camera to disconnect and that put the system in a hung state.  This kind of behavior is common with a bad USB cable.

vastmeridian, 2023-01-03:
Interesting. I have seen that before in previous log views but not understood the relevance.  The cable in place is the cable that came with the camera; a decent USB3 cable less than a metre long.  Even though this is still under test, it's velcro-strapped so it doesn't move.  I have tried other USB cables previously so I'm wondering if it's the USB2 ports on the SBC.  I'll go over to a decent USB2 cable and see if that makes a difference.  The alternative is a different computer (like an Intel NUC or similar).

vastmeridian, 2023-01-08:
I think we're close to being able to resolve this issue, Aaron.  I'll see how the USB2 cable - and your suggestion of the bandwidth restriction - goes over the next week and, if everything is stable, I'll close as 'resolved'.

vastmeridian, 2023-01-29:
Think we're good with this one now, Aaron.  I've not had issues since restricting the USB bandwidth.  Thanks

closed 2023-01-29 by vastmeridian