aaronwmorris/indi-allsky #380: Dark frames creation fails

topdancer, 2022-10-26:
Hi Aaron,

I tried to create dark frames following the procedure in your wiki. It fails, please the following traceback:

pi@raspi-allsky:~ $ systemctl --user stop indi-allsky
pi@raspi-allsky:~ $ cd indi-allsky/
pi@raspi-allsky:~/indi-allsky $ source virtualenv/indi-allsky/bin/activate
(indi-allsky) pi@raspi-allsky:~/indi-allsky $ ./darks.py sigmaclip
2022-10-26 09:56:04,070 [INFO] MainProcess fake_indi.__init__() #42: creating an instance of FakeIndiClient
2022-10-26 09:56:04,071 [INFO] MainProcess darks._initialize() #142: Connecting to indiserver
2022-10-26 09:56:12,080 [WARNING] MainProcess darks._initialize() #160: Connecting to device libcamera_imx477
2022-10-26 09:56:12,288 [INFO] MainProcess miscDb.addCamera() #44: Camera DB ID: 1
2022-10-26 09:56:12,290 [INFO] MainProcess darks._initialize() #193: CCD CFA: BGGR
2022-10-26 09:56:12,293 [INFO] MainProcess darks._take_exposures() #534: Temp folder: /tmp/tmp0j1473tj
2022-10-26 09:56:12,295 [INFO] MainProcess darks.shoot() #229: Taking 15.00000000 s exposure (gain 12)
2022-10-26 09:56:12,297 [INFO] MainProcess libcamera.setCcdExposure() #111: image command: libcamera-still --immediate --nopreview --encoding jpg --quality 100 --denoise off --awbgains 1,1 --gain 12 --shutter 15000000 --output /tmp/tmpvcxeva6y.jpg
2022-10-26 09:56:31,998 [INFO] MainProcess darks._take_exposures() #546: Exposure received in 19.7033 s
Traceback (most recent call last):
  File "/home/pi/indi-allsky/./darks.py", line 88, in <module>
    action_func()
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 347, in sigmaclip
    self._run(IndiAllSkyDarksSigmaClip)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 487, in _run
    self._take_exposures(exposure, dark_filename_t, bpm_filename_t, ccd_bits, stacking_class)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 549, in _take_exposures
    hdulist = self._wait_for_image(exposure)
  File "/home/pi/indi-allsky/indi_allsky/darks.py", line 305, in _wait_for_image
    return hdulist
UnboundLocalError: local variable 'hdulist' referenced before assignment

Did I anything incorrectly?

Thanks
Kurt

aaronwmorris, 2022-10-26:
It looks it is trying to create darks using JPGs which will not work.  What kind of raspberry pi are you running?

topdancer, 2022-10-26:
I was using this on a raspy 3 B+

aaronwmorris, 2022-10-26:
Merged #385.

I have recently tuned the code the code to allow RAW (DNG) frames using libcamera with Raspberry PIs with 1GB of memory.  You can change the libcamera format to ```DNG``` at the bottom of the config page.

You will also have to set the FFMPEG Scaling to ```25%``` or you **WILL** run out of memory when it tries to build the timelapse.

topdancer, 2022-10-26:
Thank you, Aaron. I entered the recommended changes and the process completed successfully.

You can close the issue.

aaronwmorris, 2022-10-26:
Great news!  Just reopen if there is anything else I can do for you.

closed 2022-10-26 by aaronwmorris