aaronwmorris/indi-allsky #388: Getting OSError: [Errno 28] No space left on device when disk is 56% used

vastmeridian, 2022-10-30:
I'm getting no images stored or transferred and no updates on the web page.  The out-of-date image (attached) has no info on the image but is timestamped (18::47:31) at the bottom right of the image.  
[image]

If I look in indi-allsky.log at this time I see an uncaught exception **Type: <class 'OSError'> Value: Not enough space on disk**.  The disk is 56% used and has 8% inodes used:

Here's the output of df -i:
[20221030 - disk_space.txt](https://github.com/aaronwmorris/indi-allsky/files/9897058/20221030.-.disk_space.txt)

Here's the log extract from around that time (the full log from the day is 70MB+):
[20221030 - allsky_logextract1.txt](https://github.com/aaronwmorris/indi-allsky/files/9897059/20221030.-.allsky_logextract1.txt)

Looking further at the log, this has happened three times on Oct 30th but not before (i.e. on 29th).  There are also a considerable number of log entries along these lines too (this is the first such occurrence) so this may be related to the generation of the video timelapse?:
[20221030 - allsky_logextract2.txt](https://github.com/aaronwmorris/indi-allsky/files/9897064/20221030.-.allsky_logextract2.txt)

Are there any further logs I need to supply?  Do you need any further info from me at this stage?

Thanks

aaronwmorris, 2022-10-31:
You do not appear to be using a tmpfs for /tmp, but my first guess is that may be running out of space.

indi-allsky first writes images to temporary file, normally at /tmp but python might pick another location based on the tempfile module.

I will google a bit more to see if this exception could be raised due to another [masked] issue.

aaronwmorris, 2022-10-31:
How much space do you actually have?

vastmeridian, 2022-10-31:
Here is the output from those commands:

```
username@allsky:~$ source virtualenv/indi-allsky/bin/activate
-bash: virtualenv/indi-allsky/bin/activate: No such file or directory
username@allsky:~$ python3 -c "import tempfile; print(tempfile.gettempdir());"
/tmp
username@allsky:~$
```

vastmeridian, 2022-10-31:
I'll add some memory and disk usage monitoring and see what comes back (I use Xymon elsewhere for monitoring and graphing)

vastmeridian, 2022-10-31:
Well that didn't take long.
indi-allsky is not reading the config (or not implementing it) as I get the same image with the timecode on the bottom right.  Memory utilisation is as per graph below:
[image]
And an ls of /tmp sees lots of .fit files (a small extract is below).  Not expected, from my config, at all.
```
-rw-------  1 username username 4150080 Oct 31 19:04 tmpzw8rnzn7.fit
-rw-------  1 username username 4150080 Oct 31 19:23 tmpzwaehn6i.fit
-rw-------  1 username username 4150080 Oct 31 18:32 tmpzwicw0fh.fit
-rw-------  1 username username 4150080 Oct 31 18:21 tmpzw_mxu38.fit
-rw-------  1 username username 4150080 Oct 31 19:19 tmpzx5jgeig.fit
-rw-------  1 username username 4150080 Oct 31 18:41 tmpzxzzydwp.fit
-rw-------  1 username username 4150080 Oct 31 18:49 tmpzy2k34rj.fit
-rw-------  1 username username 4150080 Oct 31 18:21 tmpzy_8m5ae.fit
-rw-------  1 username username 4150080 Oct 31 18:35 tmpzzew5m9h.fit
-rw-------  1 username username 4150080 Oct 31 19:17 tmpzzg9foc8.fit
-rw-------  1 username username 4150080 Oct 31 18:50 tmpzzwpiutc.fit
```

aaronwmorris, 2022-10-31:
Are you saving images as PNG?

vastmeridian, 2022-11-01:
Nope - only JPGs according to the settings on config.json.  But allsky is behaving as if it's not taking the config.json in properly, so I'll save my config an clear what's there.

aaronwmorris, 2022-11-01:
It looks like the image processing is getting backed up and the filesystem is full of the pre-processed images.

Check the logs for messages like `Image processed in` and `Image compressed in` and see the average times of these metrics.

```
grep "Image processed in" /var/log/indi-allsky/indi-allsky.log | tail -n 50

grep "Image compressed in" /var/log/indi-allsky/indi-allsky.log | tail -n 50
```

vastmeridian, 2022-11-01:
Thanks Aaron.  8,262 entries for both and the average times are as shown.:
```
Oct 31 17:14:28 allsky [INFO] ImageWorker001 image.saferun() #415: Image processed in 0.7404 s
Oct 31 20:21:59 allsky [INFO] ImageWorker001 image.write_img() #777: Image compressed in 0.2565 s
```

aaronwmorris, 2022-11-01:
That is a pretty standard time.  Did any of them have any values above 1-2 seconds?

vastmeridian, 2022-11-01:
I'll have a longer look.  As the log file is zipped, I've popped it onto my main computer where I have a bit more grunt.  I'll see what I find.

vastmeridian, 2022-11-01:
Mmm.  5000-odd of the "image processed" log entries recorded 1.6-1.7s.  One I saw took 1.9s

aaronwmorris, 2022-11-02:
That should not be a problem.  This is odd.

vastmeridian, 2022-11-03:
Finally got time back on the camera last night.  indi-allsky was marked as 'Down' still, so I cleared the images and videos, validated the database and then shut down the system and powered it off.  This morning I powered it on, restarted indiserver and indi-allsky (as part of dealing with #378) and it looks like we're back in business.  However, the real RAM allocation (/tmp location?) is already at 93% and climbing, whereas disk usage is pretty flat.

[image]
[image]

I'll need to update the disk monitoring to explicitly show /tmp, but I'll run it as-is now and see what happens when we run out of RAM.  Currently it looks as if /tmp contains a number of .json files as this short extract shows.

```
-rw-r--r--  1 user user    588 Nov  3 07:59 tmp23n1jrm9.json
-rw-r--r--  1 user user     585 Nov  3 10:21 tmp241xv8ay.json
-rw-r--r--  1 user user     591 Nov  3 12:16 tmp24j798ps.json
-rw-r--r--  1 user user     577 Nov  3 10:46 tmp25bzyp0n.json
-rw-r--r--  1 user user     602 Nov  3 07:43 tmp25y5ik_l.json
-rw-r--r--  1 user user     576 Nov  3 08:19 tmp25_yf7mz.json
-rw-r--r--  1 user user     603 Nov  3 09:49 tmp25z7k3fh.json
```
one of the .json files is printed below:
```
username@allsky:/tmp$ cat tmpzvtbitxh.json
{
    "device": "Altair ALTAIR290C3(USB2.0)",
    "night": 0,
    "temp": 26.100000381469727,
    "gain": 100,
    "exposure": 0.00024401000700891018,
    "stable_exposure": 0,
    "target_adu": 80.0,
    "current_adu_target": 93.22837567304497,
    "current_adu": 119.6176270922184,
    "adu_average": 0.0,
    "sqm": 445076.9311118737,
    "stars": 0,
    "time": "1667469889",
    "sqm_data": {
        "max": 450532.36953468726,
        "min": 201140.8000350515,
        "avg": 332016.59792684705
    },
    "stars_data": {
        "max": 0,
        "min": 0,
        "avg": 0.0
    }
}
username@allsky:/tmp$
```

aaronwmorris, 2022-11-03:
Can you graph CPU utilization as well?

vastmeridian, 2022-11-03:
I can, yes.
[image]
I can even keep an eye on http, inodes, files and processes too.  At present this is 'straight out of the box', so no customisation yet.

aaronwmorris, 2022-11-03:
Memory usage hitting 100% is not unusual.  Most memory will be used by file cache.  The "actual" number is probably memory usage minus file cache.  That looks normal.

I would be interested in seeing the actual CPU utilization (as opposed to load).  That load seems awfully spiky.

vastmeridian, 2022-11-03:
I'll drill down and see what I can find

Since you've opened the bug regarding the Altair CCD driver within indilib, I'll test the non-GPCam cameras I have in case it affects them too.  I also have a mono GPCam2 camera so I should try that too, as that may be different to the colour ones.  I'll let you know here first, then, if you want, I can also add these cameras to your issue on indilib.

aaronwmorris, 2022-11-03:
Sure, you can test the mono camera.  I already had a report of a GPCam2 290M that worked without too much issue (other than indi-allsky was less mature at that point).  I think mono cameras are set in RAW mode by default since there is no bayer matrix to deal with.

https://github.com/aaronwmorris/indi-allsky/issues/7

vastmeridian, 2022-11-03:
> I would be interested in seeing the actual CPU utilization (as opposed to load). That load seems awfully spiky.

This do?  Same timescale as before.. The spike this time was me doing some testing.  I'll leave it for now.
[image]

aaronwmorris, 2022-11-03:
Indeed.  That looks normal.

vastmeridian, 2022-11-05:
Aaron.
This issue hasn't repeated itself and may well have been a glitch in the matrix.  I'm happy to close the issue, if you are, and to concentrate on getting the GPCam cameras to run out of the box.

Thoughts?

aaronwmorris, 2022-11-05:
I say close it and re-open if you experience the issue again.

closed 2022-11-05 by vastmeridian