aaronwmorris/indi-allsky #390: Apache issues? Internal Server Error / Can't reach this page / AH00526: Syntax error on line 27

StarGeezerPhil, 2022-10-31:
Hi Aaron,

Strange issue today - I was using the web interface happily, then suddenly received "Internal Server Error" warning in browser.

Can SSH in all-ok.

I've restarted the device and fully powered down.

Looks like apache is refusing to start - I just queried `systemctl status apache2.service` with the following error:

```
Oct 31 22:21:47 pi systemd[1]: Starting The Apache HTTP Server...
Oct 31 22:21:47 pi apachectl[3084]: AH00526: Syntax error on line 27 of /etc/apache2/sites-enabled/indi-allsky.conf:
Oct 31 22:21:47 pi apachectl[3084]: <Directory> directive requires additional arguments
Oct 31 22:21:47 pi apachectl[3081]: Action 'start' failed.
Oct 31 22:21:47 pi apachectl[3081]: The Apache error log may have more information.
Oct 31 22:21:47 pi systemd[1]: apache2.service: Control process exited, code=exited, status=1/FAILURE
Oct 31 22:21:47 pi systemd[1]: apache2.service: Failed with result 'exit-code'.
Oct 31 22:21:47 pi systemd[1]: Failed to start The Apache HTTP Server.
```

aaronwmorris, 2022-10-31:
Can you paste the full contents of `/etc/apache2/sites-enabled/indi-allsky.conf`?

StarGeezerPhil, 2022-11-01:
Hi Aaron, please see below:
```
Listen *:80
Listen *:443

<VirtualHost *:80>
    RewriteEngine On

    ### Comment this section to permit HTTP access to indi-allsky
    RewriteCond "%{HTTPS}" off
    RewriteRule "^/(.*)" "https://%{SERVER_NAME}:443/$1" [R,L]
    ###

    RewriteRule "^/$" "/indi-allsky/" [R]

    ProxyPreserveHost On
    ProxyPass /indi-allsky/static !
    ProxyPass /indi-allsky/images !
    ProxyPass /indi-allsky unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock|http://localhost/indi-allsky
    ProxyPassReverse /indi-allsky unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock|http://localhost/indi-allsky
    ProxyTimeout 180

    <Directory /home/stargeezer/indi-allsky>
        Require all granted
    </Directory>

    <Directory >
        Require all granted
        Options +Indexes
    </Directory>

    <Location />
        AuthType Basic
        AuthName "Allsky"
        AuthUserFile "/etc/indi-allsky/apache.passwd"
        Require valid-user
        #SetEnv proxy-chain-auth On
    </Location>

    # Allow non-authenticated access to latest image
    <Location /indi-allsky/public>
        Require all granted
    </Location>

    <Location /indi-allsky/static>
        Require all granted
    </Location>

    <Location /indi-allsky/images/latest.jpg>
        Require all granted
    </Location>

    <Location /indi-allsky/images/latest.png>
        Require all granted
    </Location>

    Alias /indi-allsky/images
    Alias /indi-allsky/static /home/stargeezer/indi-allsky/indi_allsky/flask/static
</VirtualHost>

<VirtualHost *:443>
    RewriteEngine On
    RewriteRule "^/$" "/indi-allsky/" [R]

    ProxyPreserveHost On
    ProxyPass /indi-allsky/static !
    ProxyPass /indi-allsky/images !
    ProxyPass /indi-allsky unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock|http://localhost/indi-allsky
    ProxyPassReverse /indi-allsky unix:/var/lib/indi-allsky/gunicorn-indi-allsky.sock|http://localhost/indi-allsky
    ProxyTimeout 180

    <Directory /home/stargeezer/indi-allsky>
        Require all granted
    </Directory>

    <Directory >
        Require all granted
        Options +Indexes
    </Directory>

    <Location />
        AuthType Basic
        AuthName "Allsky"
        AuthUserFile "/etc/indi-allsky/apache.passwd"
        Require valid-user
        #SetEnv proxy-chain-auth On
    </Location>

    # Allow non-authenticated access to latest image
    <Location /indi-allsky/public>
        Require all granted
    </Location>

    <Location /indi-allsky/static>
        Require all granted
    </Location>

    <Location /indi-allsky/images/latest.jpg>
        Require all granted
    </Location>

    <Location /indi-allsky/images/latest.png>
        Require all granted
    </Location>

    Alias /indi-allsky/images
    Alias /indi-allsky/static /home/stargeezer/indi-allsky/indi_allsky/flask/static

    SSLEngine on
    SSLCertificateFile /etc/apache2/ssl/indi-allsky_apache.pem
    SSLCertificateKeyFile /etc/apache2/ssl/indi-allsky_apache.key

    #Header always set Strict-Transport-Security "max-age=604800; includeSubDomains"  # 1 week
</VirtualHost>
```

aaronwmorris, 2022-11-01:
I am wondering if you started with a [much older] version of indi-allsky that allowed the IMAGE_FOLDER variable to be blank, which means the images would be stored inside the git checkout folder.

Merged #391 and #392 with some minor changes and checks to attempt to correct this situation.

Pull down the latest code and re-run setup.sh and see if that corrects the issue.

StarGeezerPhil, 2022-11-01:
Thank you Aaron, I'll check and let you know

StarGeezerPhil, 2022-11-02:
Hi @aaronwmorris, thank you again for the always quick responses.

Sadly, getting a different issue in the setup.py script:

```
**** Setup HTDOCS folder ****
**** Setup image folder ****
**** Setup DB ****
**** Backup DB prior to migration ****
Traceback (most recent call last):
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/bin/flask", line 8, in <module>
    sys.exit(main())
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 1047, in main
    cli.main()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 354, in decorator
    app = __ctx.ensure_object(ScriptInfo).load_app()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 312, in load_app
    app = locate_app(import_name, None, raise_if_not_found=False)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 218, in locate_app
    __import__(module_name)
  File "/home/pi/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/pi/indi-allsky/indi_allsky/__init__.py", line 4, in <module>
    from .allsky import IndiAllSky
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 28, in <module>
    from . import camera as camera_module
  File "/home/pi/indi-allsky/indi_allsky/camera/__init__.py", line 1, in <module>
    from .indi import IndiClient as indi
  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 26, in <module>
    app = create_app()
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 92, in create_app
    event.listen(db.engine, 'connect', _sqlite_pragma_on_connect)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_sqlalchemy/__init__.py", line 998, in engine
    return self.get_engine()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_sqlalchemy/__init__.py", line 1017, in get_engine
    return connector.get_engine()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask_sqlalchemy/__init__.py", line 592, in get_engine
    sa_url = make_url(uri)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/url.py", line 725, in make_url
    return _parse_rfc1738_args(name_or_url)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/sqlalchemy/engine/url.py", line 786, in _parse_rfc1738_args
    raise exc.ArgumentError(
sqlalchemy.exc.ArgumentError: Could not parse rfc1738 URL from string ''

The script exited abnormally, please try to run again...
```

aaronwmorris, 2022-11-02:
Can you attach a copy of your config file `/etc/indi-allsky/config.json`

StarGeezerPhil, 2022-11-03:
Hi Aaron, that file appears to be blank.

StarGeezerPhil, 2022-11-03:
@aaronwmorris, I did try setup.py other day before you merged more changes and it failed - is it likely it failed to copy the original config, so is now blank?

Just a thought on this - I'm hoping to mount the unit out of reach... when these types of issues arise (hopefully they won't), is it possible to do a fully clean install remotely (without needing to reimage the SD)?

aaronwmorris, 2022-11-03:
How long has this camera been in operation?  I wonder if your sdcard might be failing... just a thought.

You can always re-setup without a full install.  You may just want to delete `/etc/indi-allsky/config.json` and re-run the setup.sh script.

If your images are stored in `/home/stargeezer/indi-allsky/html/images`, you can change that setting after the setup completes (and restart again).

You can even blow away the database and re-import your images to the DB.  There is a function to do that in indi-allsky.  I can walk you through this if you want.

aaronwmorris, 2022-11-03:
The uninstall wiki page tells you just about everything you would need to delete to cleanup without having to reformat your card.

https://github.com/aaronwmorris/indi-allsky/wiki/Uninstall-indi-allsky

StarGeezerPhil, 2022-11-03:
Clearing the config.json and rerunning setup.py has done the trick, thanks Aaron. Wish I'd taken a backup though - I was still tweaking settings 🤣

closed 2022-11-03 by StarGeezerPhil