aaronwmorris/indi-allsky #400: error: b'ERROR: *** no cameras available ***\n'

jcauthen78, 2022-11-06:
Excited about this package. Currently have Thomas' AllSky running solid on a permanent install, so giving this one a shot on my test/camping rig.

**Equipment**: 
- Raspberry Pi 4 Model B Rev 1.4, 8gb
- Fresh Debian GNU/Linux 11 (bullseye)
- ASI178MC. 

**The main problem - ASI178MC not being found.** 
- Update & full upgrade ran today (6 nov) prior to Indi-Allsky install.
- Rebooted a few times - no joy.
- Log File should be attached here : [indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/9946802/indi-allsky.log)
- ran 'lsusb' and the 178 is visible (see end of above log).

I've played pretty heavily with Thomas's AllSky software for about 4 years now, through all of its updates, so the basics are pretty familiar - it's entirely possible I'm missing something obvious.

_Also of note:_ Not related to this problem (i think), but had to run the setup.sh twice (second after reboot) for the setup to fully complete - not sure if it's intended to be run twice, the main page doesn't say to check.  The second run appeared to complete successfully.
[image]

jcauthen78, 2022-11-06:
From the System Tab: 

>Release - 5.4
> System
> Python - 3.9.2
> pyindi - 1.9.7
> cv2 - 4.6.0
> numpy - 1.22.4
> ephem - 4.1.3
> astropy - 5.1.1
> flask - 2.2.2
> ccdproc - 2.3.1
> paramiko - 2.12.0
> paho.mqtt - 1.6.1
> dbus - 1.3.2
> pycurl - PycURL/7.45.1 libcurl/7.74.0 GnuTLS/3.7.1 zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3

aaronwmorris, 2022-11-06:
You have selected the libcamera interface during the setup.  Re-run setup.sh and select `indi` and the `indi_asi_ccd` driver during the setup.  I would suggest a full reboot after do that and everything should work for you.

jcauthen78, 2022-11-06:
Thank you for the quick response - That fixed it (with a tweak, see notes below) - Closing :)

Notes: 
that first **indi / libcamera** wasn't as intuitive in my head this afternoon. maybe stick (zwo/atik/etc) after the Indi text in the selection? Or maybe in the instructions on the main page in section 4 as a note? I know others have it successfully installed, so no worries. It's one of those _'now i know'_ moments. 

Tried re-running the **setup.sh** then reboot - no joy. 

renamed (mv) indi-allsky folder, then started back with the original GIT. 
Went a little faster since most the dependencies were already there, and it installed a few more that the re-run didn't load. After that, rebooted again, BOOM!!! We've got images!  

Now to go play with the configs a bit before throwing it outside for some testing under partly cloudy & rainy sky tonight. 

Thanks again for this project - looking forward to playing with it,
/Jim

jcauthen78, 2022-11-06:
oops, forgot to click close.

closed 2022-11-06 by jcauthen78

aaronwmorris, 2022-11-07:
Thanks for the feedback!  I do need to clarify the indi vs libcamera configuration options.