aaronwmorris/indi-allsky #408: Getting segmentation fault

sifank, 2022-11-12:
Top part of the setup output:
Distribution: Ubuntu
Release: 22.04
Arch: aarch64
INDI_DRIVER_PATH: /usr/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: true
HTTP_PORT: 80
HTTPS_PORT: 443

Last part of the stdout from "./allsky.py run":
2022-11-12 15:42:32,570 [INFO] MainProcess indi._findCcds() #414: Found device ZWO CCD ASI294MC Pro
2022-11-12 15:42:32,570 [INFO] MainProcess indi._findCcds() #419:  Detected ccd
2022-11-12 15:42:32,571 [INFO] MainProcess indi.findCcd() #429: Found 1 CCDs
2022-11-12 15:42:32,571 [WARNING] MainProcess allsky._initialize() #364: Connecting to device ZWO CCD ASI294MC Pro
2022-11-12 15:42:32,638 [INFO] MainProcess miscDb.addCamera() #45: Camera DB ID: 1
2022-11-12 15:42:32,638 [INFO] MainProcess indi.configureDevice() #439: Setting switch DEBUG
Segmentation fault (core dumped)

This is a freshly build RPI4 using Ubuntu desktop with update to KDE.  I had two issue getting this far:
1) At first I tried having your software install INDI.  It blew up on pyindi-client.  So I built the indi3rdparty software manually.
2) Then, when installing, it found the indi drivers but still tried to install pyindi-client (which was already installed).  I found the spot in your software where you rebuilt it to the indi drivers.  So I downloaded the current pyindi-client (from the updated git location), placed it within the allsky folders and had your software build/install it from there.  That worked and got me to this point.  The web site comes up, everything starts, but the web systems/services page says that 'allsky' failed.  So I stopped the service, started the virtenv and ran allsky.py manually and attached that output above.

How can I debug this further?

I can see that in ~/indi-allsky/indi_allsky/camera/indi.py in the def for configureDevice, where it is trying to get the info for the SWITCHES seems to the last place before it core dumps.  Where could I place logger.info statements to continue my dive into this?

Thx,
Sifan

sifank, 2022-11-13:
Additional info:
- I ran a remote KSTARS/EKOS against the indiserver on this rpi and successfully downloaded images. 
- I also started a manual python3 session and typed in a few pyindi commands successfully
Hope that helps
Sifan

aaronwmorris, 2022-11-13:
I am wondering if you have multiple versions of indi on your system.  If you built the indi-3rdparty stuff manually, you should probably install the indi core library manually, too.

Make sure you uninstall all of the indi packages from the system before doing that.

I have a script, `./misc/build_indi.sh` that will take care of compiling everything.  It will also refuse to build if you have any indi packages pre-installed.

sifank, 2022-11-13:
Thanks Aaron!

When I run the./misc/build_indi.sh, it gets tons of symbol not found errors
- like 5 pages full.  And all of those are from the pyindi-client.  That's
why I downloaded pyindi-client from the new github location, placed it
under ~allsky and modified the script to build it in place.   And that
worked, as far as continuing and finally being able to complete and get the
web up.

The same errors occur when I do a pip3 install pyindi-client.  And that's
why I got the version from github.

I did run the python3 test examples suggested in the indi site for
pyindi-client and those worked and came back with the telescope simulator
and my ASI294 cam.

I also checked (after a reboot) that only one indiserver was running.  I'll
do a ps for indi and allsky and reply to the email shortly.

Sifan
Carpe Nocturnum ...

sifank, 2022-11-13:
I'll build the core libs manually as well.

aaronwmorris, 2022-11-13:
I will break out my pi4 with an Ubuntu image to see if there are any dependencies that need updating.

sifank, 2022-11-13:
The only indi pkgs that were installed was libindi1 and libindi-data.  I removed both of those (apt remove) then cloned the latest indi-core, built and installed it.  But I'm getting the same errors (segmentation fault after Setting switch DEBUG).

Interestingly, after a reboot I'm seeing 2 indi-allsky processes: (psg is a ps -ef | grep -i)
psg allsky
sifan       1142     997  2 11:47 ?        00:00:01 /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/python /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
sifan       1266    1142 17 11:47 ?        00:00:06 /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/python /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi

When I try to stop it using the web (system/services) I get (in /var/log/allsky)
Nov 12 19:45:38 sqm [INFO] MainProcess views.dispatch_request() #1861: Command return: /org/freedesktop/systemd1/job/346

and the there are still 2 services:
psg allsky
sifan       1142     997  0 19:42 ?        00:00:01 /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/python /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
sifan       2039    1142  4 19:44 ?        00:00:05 /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/python /home/sifan/Projects/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi

I have to kill them manually before running 'allsky.py run' (kill 1142 gets them both).  Stdout from the allsky.py run (after activate):
(indi-allsky) [sifan@sqm:399]# ./allsky.py run
2022-11-12 19:50:28,609 [INFO] MainProcess allsky._initialize() #315: indi-allsky release: 5.5
2022-11-12 19:50:28,610 [INFO] MainProcess allsky._initialize() #316: indi-allsky config version: 20221106.0
2022-11-12 19:50:28,610 [INFO] MainProcess allsky._initialize() #318: Python version: 3.10.6
2022-11-12 19:50:28,611 [INFO] MainProcess allsky._initialize() #319: Platform: aarch64
2022-11-12 19:50:28,611 [INFO] MainProcess allsky._initialize() #321: System CPUs: 4
2022-11-12 19:50:28,612 [INFO] MainProcess allsky._initialize() #326: System memory: 3789 MB
2022-11-12 19:50:28,612 [INFO] MainProcess allsky._initialize() #329: System uptime: 470s
2022-11-12 19:50:28,613 [INFO] MainProcess indi.__init__() #111: creating an instance of IndiClient
2022-11-12 19:50:28,613 [INFO] MainProcess indi.__init__() #118: PyIndi version: 1.9.9
2022-11-12 19:50:28,614 [INFO] MainProcess allsky._initialize() #346: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2022-11-12 19:50:28,615 [INFO] MainProcess indi.serverConnected() #241: Server connected (localhost:7624)
2022-11-12 19:50:28,616 [INFO] MainProcess indi.newDevice() #156: new device ZWO CCD ASI294MC Pro
Dispatch command error(-1): INDI: Could not find property CCD_COOLER_POWER in ZWO CCD ASI294MC Pro
<setNumberVector device="ZWO CCD ASI294MC Pro" name="CCD_COOLER_POWER" state="Idle" timeout="60" timestamp="2022-11-13T03:50:28">
    <oneNumber name="CCD_COOLER_VALUE">
0
    </oneNumber>
</setNumberVector>
2022-11-12 19:50:28,619 [INFO] MainProcess indi.newDevice() #156: new device Telescope Simulator
Dispatch command error(-1): INDI: Could not find property CCD_COOLER_POWER in ZWO CCD ASI294MC Pro
<setNumberVector device="ZWO CCD ASI294MC Pro" name="CCD_COOLER_POWER" state="Idle" timeout="60" timestamp="2022-11-13T03:50:28">
    <oneNumber name="CCD_COOLER_VALUE">
0
    </oneNumber>
</setNumberVector>
Dispatch command error(-1): INDI: Could not find property CCD_COOLER_POWER in ZWO CCD ASI294MC Pro
<setNumberVector device="ZWO CCD ASI294MC Pro" name="CCD_COOLER_POWER" state="Idle" timeout="60" timestamp="2022-11-13T03:50:28">
    <oneNumber name="CCD_COOLER_VALUE">
0
    </oneNumber>
</setNumberVector>
2022-11-12 19:50:28,642 [INFO] MainProcess indi.newText() #232: new Text SCOPE_CONFIG_NAME for device Telescope Simulator
2022-11-12 19:50:36,620 [INFO] MainProcess indi._findCcds() #414: Found device Telescope Simulator
2022-11-12 19:50:36,620 [INFO] MainProcess indi._findCcds() #419:  Detected telescope
2022-11-12 19:50:36,620 [INFO] MainProcess indi._findCcds() #419:  Detected guider
2022-11-12 19:50:36,621 [INFO] MainProcess indi._findCcds() #414: Found device ZWO CCD ASI294MC Pro
2022-11-12 19:50:36,622 [INFO] MainProcess indi._findCcds() #419:  Detected ccd
2022-11-12 19:50:36,622 [INFO] MainProcess indi.findCcd() #429: Found 1 CCDs
2022-11-12 19:50:36,623 [WARNING] MainProcess allsky._initialize() #364: Connecting to device ZWO CCD ASI294MC Pro
2022-11-12 19:50:36,695 [INFO] MainProcess miscDb.addCamera() #45: Camera DB ID: 1
2022-11-12 19:50:36,696 [INFO] MainProcess indi.configureDevice() #439: Setting switch DEBUG
Segmentation fault (core dumped)

The 'Could not find property CCD_COOLER_POWER in ZWO CCD ASI294MC Pro' is a new entry - did not see that in the logs before building indi-core.  Is there a config I need to reapply?  Or do I need to reinstall (rerun setup.sh)?

I could burn a new sim and start again if you think that will make a difference.  Also, if you would like, I can sprinkle a few logger.info's around in places you suggest.

Thx for your help!
Sifan

sifank, 2022-11-13:
BTW, I'm working with the Oregon chapter of the IDA (International DarkSky Association) to create a calibrated allsky sqm using the ASI294mc pro.  They have someone who calibrates cameras.  So once I have this portable unit (includes a GPS, env sensors, etc) working, I'll be handing it over to them for calibration.

One question I have, is it possible to create FITs images?  It looks like you have the option for raw:  what format is the raw file in?  Perhaps that could be wrapped into a FITs image and saved?

Thx
Sifan

aaronwmorris, 2022-11-13:
You may want to re-image your raspberry pi and start from scratch.  The first thing you should do with Ubuntu is run the `./misc/build_indi.sh` and then run `setup.sh`.

There is both an option to save the FITS data directly and an option to save RAW data as PNG prior to any processing (other than dark frame calibration).

This wiki page documents the order of operations on images:
https://github.com/aaronwmorris/indi-allsky/wiki/Image-order-of-operations

sifank, 2022-11-13:
After a fresh install of ubuntu-22.04.1-preinstalled-desktop-arm64+raspi.img, then installing kde and updating, then I ran 
mkdir Projects; cd Projects
git clone https://github.com/aaronwmorris/indi-allsky.git
cd indi-allsky
./misc/build_indi.sh

And this happened:
TASK [build_indi : fail] *************************************************************************************************************************
fatal: [localhost]: FAILED! => {"changed": false, "msg": "The following packages need to be removed before continuing: libindi-data, libindi-dev, libindi-plugins"}

No other indi packages were installed.

I am now re-imaging again, but this time I will not install kde, only update then clone and run ./misc/build_indi.sh and see if I get the same error.

Thx for the info re FITS!

aaronwmorris, 2022-11-13:
You just need to remove the default indi system related packages before building from source.

`apt-get remove libindi-data libindi-dev libindi-plugins`

I am downloading the same Ubuntu image you used so I can verify the steps.

sifank, 2022-11-13:
Ok, fresh install of: ubuntu-22.04.1-preinstalled-desktop-arm64+raspi.img
- full upgrade
- installed net-tools, openssh-server, git
- modified netplan for my ip addrs
- git cloned indi-allsky
- run .misc/build_indi.sh

And .... it's working (after a reboot)!!

I'll follow this with what I had installed previously on the failed attempts for that everyone will have an idea of what could have made this fail.  (If I have time I'll try to find the pkg that did it ...)

Much thanks Aaron!

Sifan

sifank, 2022-11-14:
Here is a list of the other packages that were added in my first attempts.  So one of these (or more) may have caused the issues I was seeing.  All of the attempts started with 'ubuntu-22.04.1-preinstalled-desktop-arm64+raspi.img' as a base.

apt install:
i2c-tools muon locate xterm vlc daemontools lshw screen software-properties-common dbus-x11 rpi.gpio-common libi2c-dev python3-smbus p7zip-full python3-psutil python3-smbus python3-pip python3-numpy libindi-dev update-notifier-common vim vim-scripts vim-doc vim-latexsuite vim-gui-common xz-utils ntpdate ntp

pip3 install:
mysql.connector adafruit-blinka hidapi 

And then:
apt -y install kde-standard

Hope this help.  If I can narrow it down further, I'll post that here.

aaronwmorris, 2022-11-14:
It's the `libindi-dev` package that caused the problem.  When you compile from source, it will install the latest indi development headers, so this package was not needed.

sifank, 2022-11-14:
Ah ha!  Thanks again for all your help!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-07:
Closing this issue for now.

closed 2022-12-07 by aaronwmorris