aaronwmorris/indi-allsky #409: Indi-allsky fails to start with IP camera configured yet works fine after restarting the service

vastmeridian, 2022-11-14:
**Processor**: Virtualbox VM configured with 2GB RAM and 20GB SSD
**Camera**: IP camera permanently running with indi-webcam driver (downloaded from ppa:mutlaqja/ppa)
**Operating System**: Ubuntu 20.04.5 LTS (GNU/Linux 5.4.0-131-generic x86_64)
**indi-allsky**: Release 5.5

**Issue**: When the server is initially started, the indi-allsky service stops after a short while and remains DOWN.  If the service is then restarted from the web interface, the service starts and remains running properly.
----
Files attached from period after switch-on until service goes 'down' (the broken phase)
[indi_getprop from power-on until indi_allsky failure.txt](https://github.com/aaronwmorris/indi-allsky/files/10005636/indi_getprop.from.power-on.until.indi_allsky.failure.txt)

[indi-allsky logfile from power-on until indi_allsky failure.txt](https://github.com/aaronwmorris/indi-allsky/files/10005672/indi-allsky.logfile.from.power-on.until.indi_allsky.failure.txt)
----
Files attached from when indi-allsky service is restarted  (the working phase)
[indi_getprop after service restarted.txt](https://github.com/aaronwmorris/indi-allsky/files/10005678/indi_getprop.after.service.restarted.txt)
[indi-allsky logfile after service restarted.txt](https://github.com/aaronwmorris/indi-allsky/files/10005680/indi-allsky.logfile.after.service.restarted.txt)
----
if it helps, here's the config.json
[IP camera config.json.txt](https://github.com/aaronwmorris/indi-allsky/files/10005687/IP.camera.config.json.txt)

aaronwmorris, 2022-11-14:
The problem is the `FRAME_TYPE` property does not become valid until a connection to the camera has been successful.

I do not have a good way of dealing with this, but I do have a workaround.

Once the webcam is successfully taking pictures (after the second run), run the following in the console:

```
indi_setprop -s "INDI Webcam".CONFIG_PROCESS.CONFIG_SAVE=On
```
This will save the existing settings in the indiserver to the XML files in `~/.indi/`.  The next time the indiserver starts, it will load these as a default and it should not hang on the first run.

vastmeridian, 2022-11-15:
Thanks Aaron.  I'll try that shortly.

Looks like we may have an issue switching from day mode to night mode (or 'sleeping' to 'running' according to the web site) I get no new images tonight.  I'll take a look at the logs and see where we are.  The service is 'running', just no images.

vastmeridian, 2022-11-15:
I've done that and I'll test shortly.  Would an alternative be a startup script that uses wget or curl to get an image from the camera? Or does the connection need to be made from within indi?

aaronwmorris, 2022-11-15:
That error means a camera was not detected.  Are you sure the indiserver is running with the indi_webcam_ccd driver?

vastmeridian, 2022-11-15:
yes.  I've even re-run ./setup.sh to re-install the driver. I even recreated a new indiserver instance, as prompted in the script. No different.

Curioser and curioser ...  Here's the indi-allsky log after a reboot.
[indi-allsky log showing 'remove device'.txt](https://github.com/aaronwmorris/indi-allsky/files/10016790/indi-allsky.log.showing.remove.device.txt)

aaronwmorris, 2022-11-16:
You can delete the XML files in `~/.indi/` to reset to the defaults.

vastmeridian, 2022-11-16:
Did that and restarted the services.  This is the log file.   It's still 'removing the device' I think I'll uninstall indi-allsky and re-install from scratch.
[indi-allsky logfile after rm of indiserver XML files.txt](https://github.com/aaronwmorris/indi-allsky/files/10020948/indi-allsky.logfile.after.rm.of.indiserver.XML.files.txt)

I have been pondering.  I need to work some magic on the images straight from the camera, so I may put together a separate pre-processor to gather the image from the camera, process it (stretch and contrast) and leave it somewhere within the local web server.  I can then just point indi-allsky at itself then to get the images.

aaronwmorris, 2022-11-16:
Its not indi-allsky causing that behavior.  The indiserver is misbehaving.  Did you restart the indiserver after deleting the XMLs?

vastmeridian, 2022-11-17:
I have stopped both services; made sure XML files have been deleted; restarted the services.  No change.
I have also stopped both services (after a reboot); made sure XML files are gone and re-run ./setup.sh. No change.  During setup, I replaced the indiserver definition too.

```
Nov 17 23:06:52 allskyvideo [INFO] MainProcess indi._findCcds() #419:  Detected telescope
Nov 17 23:06:52 allskyvideo [INFO] MainProcess indi._findCcds() #419:  Detected guider
Nov 17 23:06:52 allskyvideo [INFO] MainProcess indi.findCcd() #429: Found 1 CCDs
Nov 17 23:06:52 allskyvideo [WARNING] MainProcess allsky._initialize() #364: Connecting to device INDI Webcam
Nov 17 23:06:52 allskyvideo [INFO] MainProcess miscDb.addCamera() #45: Camera DB ID: 2
Nov 17 23:06:52 allskyvideo [INFO] MainProcess indi.configureDevice() #439: Setting switch DEBUG
Nov 17 23:06:53 allskyvideo [INFO] MainProcess indi.updateCcdBlobMode() #248: Set BLOB mode
Nov 17 23:06:53 allskyvideo [INFO] MainProcess indi.configureDevice() #439: Setting switch CAPTURE_DEVICE
Nov 17 23:06:53 allskyvideo [INFO] MainProcess indi.configureDevice() #439: Setting switch ONLINE_PROTOCOL
Nov 17 23:06:53 allskyvideo [INFO] MainProcess indi.removeDevice() #159: remove device INDI Webcam
```
There are no new XML files in ~/.indi/ either.

vastmeridian, 2022-11-22:
I've managed to get #425 working and loaded the extra driver indi_webcam.  My camera is working fine on its URL, but indi-allsky is still not working properly.  The website says 'DOWN' and the service says Failed, but the log says very little:

`Nov 22 17:41:23 allskyvideo [INFO] MainProcess views.dispatch_request() #1871: Command return: /org/freedesktop/systemd1/job/106
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #317: indi-allsky release: 5.5
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #318: indi-allsky config version: 20221106.0
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #320: Python version: 3.9.5
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #321: Platform: x86_64
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #323: System CPUs: 2
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #328: System memory: 1983 MB
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #331: System uptime: 1071s
Nov 22 17:41:25 allskyvideo [INFO] MainProcess indi.__init__() #112: creating an instance of IndiClient
Nov 22 17:41:25 allskyvideo [INFO] MainProcess indi.__init__() #119: PyIndi version: 1.9.9
Nov 22 17:41:25 allskyvideo [INFO] MainProcess allsky._initialize() #348: Connecting to indiserver
Nov 22 17:41:25 allskyvideo [INFO] MainProcess indi.serverConnected() #245: Server connected (localhost:7624)
Nov 22 17:41:25 allskyvideo [INFO] MainProcess indi.newDevice() #157: new device INDI Webcam
Nov 22 17:41:25 allskyvideo [INFO] MainProcess indi.newDevice() #157: new device Telescope Simulator
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi._findCcds() #430: Found device INDI Webcam
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected ccd
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi._findCcds() #430: Found device Telescope Simulator
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected telescope
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected guider
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi.findCcd() #445: Found 1 CCDs
Nov 22 17:41:33 allskyvideo [WARNING] MainProcess allsky._initialize() #365: Connecting to device INDI Webcam
Nov 22 17:41:33 allskyvideo [INFO] MainProcess miscDb.addCamera() #45: Camera DB ID: 1
Nov 22 17:41:33 allskyvideo [INFO] MainProcess indi.configureDevice() #459: Setting switch DEBUG
`
That's it.  Nothing.  I'll try and fire up the Altair camera server (separate system), update it to #425 and see if that gets off the ground.

aaronwmorris, 2022-11-28:
indi-allsky is crashing due to some issues between indi 1.9.9 and pyindi-client.

https://github.com/indilib/pyindi-client/issues/26

vastmeridian, 2022-11-28:
> indi-allsky is crashing due to some issues between indi 1.9.9 and pyindi-client.
> 
> [indilib/pyindi-client#26](https://github.com/indilib/pyindi-client/issues/26)

Thanks for investigating.  I'll wait and try again when indi 1.9.9 is updated.

vastmeridian, 2022-12-04:
> indi-allsky is crashing due to some issues between indi 1.9.9 and pyindi-client.
> 
> [indilib/pyindi-client#26](https://github.com/indilib/pyindi-client/issues/26)

Any update on this, Aaron?

aaronwmorris, 2022-12-04:
The fixes are in a PR and we are just waiting on a merge.  I will go see if I can get some movement on it.

vastmeridian, 2022-12-17:
I've been trying the later builds as you've been releasing them, but I am getting the system defaulting to /dev/video, despite the config set up for an IP camera.

Extract from logfile:
```
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi._findGpss() #592:  Detected telescope
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi._findGpss() #592:  Detected guider
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.findGps() #602: Found 0 GPSs
Dec 17 00:32:26 allskyvideo [WARNING] MainProcess allsky._initialize() #408: Connecting to CCD device INDI Webcam
Dec 17 00:32:26 allskyvideo [WARNING] MainProcess allsky._initialize() #412: Connecting to Telescope device Telescope Simulator
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.configureDevice() #620: Setting switch TELESCOPE_SLEW_RATE
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-17T00:32:26: [INFO] Trying to connect to: /dev/video0, on device: video4linux2,v4l2 with 640x480 at 30 frames per second
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-17T00:32:26: [INFO] Attempting to connect
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-17T00:32:26: [INFO] Failed to open source. Check your settings: No such file or directory
Dec 17 00:32:26 allskyvideo [INFO] MainProcess indi.newSwitch() #270: new Switch CONNECTION for device INDI Webcam
```

And the config setup I've got (just edited for privacy)
```
{
    "PROPERTIES": {},
    "TEXT": {
        "ONLINE_PATH": {
            "URL_PATH": "http://username:password@ip_address/jpg/image.jpg"
        }
    },
    "SWITCHES": {
        "CAPTURE_DEVICE": {
            "on": [
                "IP Camera"
            ]
        },
        "ONLINE_PROTOCOL": {
            "on": [
                "HTTP"
            ],
            "off": [
                "CUSTOM"
            ]
        }
    }
}
```
The URL_PATH works just fine when pasted into a browser.
I'm wondering if this issue should be closed and a new one opened.  Happy either way.

vastmeridian, 2022-12-21:
Closing this as I have a new issue with the camera that I will raise separately.

closed 2022-12-21 by vastmeridian