aaronwmorris/indi-allsky #412: Dark Frames | Migrating installs

StarGeezerPhil, 2022-11-15:
Is it possible to migrate an install (in this case simply to a larger SD card) and reuse the darks already collected?

I have a set of FITS darks at gain 0, 8 and 16 collected on my install on a 64gb card.

I've copied these into the new install on a 256gb card, but they don't appear to be picked up and I can't find any mention of them in the log file (I'm sure I remember it previously mentioning the darks).

The reference "ccd1" is the same as freshly generated images.

Is this simply not possible, or is there something I need to do on the new install to enable them?

Many thanks, 
Phil

aaronwmorris, 2022-11-15:
The dark frames just need to be registered in the database.  I will write a script to help import those tonight.

StarGeezerPhil, 2022-11-15:
Ah - that's brilliant Aaron. It'd be a big help and useful addition - can imagine others building up their dark library and subsequently upgrading the SD.

I was just trying to figure out if I was going mad or if the image quality had just suddenly gone terrible 🤣

StarGeezerPhil, 2022-11-17:
Hi @aaronwmorris, sorry did you make any progress or is there anything I can do this end to register the existing darks?

aaronwmorris, 2022-11-17:
I am just running behind.  I will try to work on it tonight.

StarGeezerPhil, 2022-11-17:
No trouble - understand what it's like. Thanks again 👍

aaronwmorris, 2022-11-19:
This took a little longer than I was expecting, but it is ready.  Merged #420 with the new import utility.

`./misc/import_darks_frames.py`

Just put the dark frames (and bad pixel maps) in `/var/www/html/allsky/images/darks/` and run the utility above.  The import is interactive and will try to auto detect the metadata, but will ask you if it cannot be read from the FITS headers.

StarGeezerPhil, 2022-11-19:
Hi Aaron, thank you for this.

I've updated indi-allsky, but when running this script get the following error:

```
Traceback (most recent call last):
  File "/home/pi/indi-allsky/./misc/import_darks_frames.py", line 12, in <module>
    from astropy.io import fits
ModuleNotFoundError: No module named 'astropy'
```

aaronwmorris, 2022-11-19:
You have to enable the python virtual environment before running.

```
source virtualenv/indi-allsky/bin/activate
```

StarGeezerPhil, 2022-11-19:
Ah brilliant - I'd not clicked 👍

It didn't manage to pick up the gain, bin mode or temperature from it's own files - so might need some tweaking, but looks like the process was successful manually putting them in.

Filename for reference (assume this is very similar for most installs?):
`/var/www/html/allsky/images/darks/dark_ccd1_16bit_30s_gain1_bin1_-273c_20220823_211702.fit`

aaronwmorris, 2022-11-20:
The automatic detection is only performed against the internal FITS headers, which is optional data in the files.  INDI automatically creates the headers when it generates FITS files, but I did not do a very good job of populating the FITS headers when indi-allsky converted the RAW data from the libcamera DNG files (to FITS).

I did go back and program the dark frame module to populate more headers such as gain and bin mode to the libcamera sourced darks.

https://github.com/aaronwmorris/indi-allsky/commit/3c7eea79d8ba6361539f9c9df8d051f273f3c068

StarGeezerPhil, 2022-11-20:
Ah brill. Once I've made the final tweaks and happy with everything I intend to clear out and take a wider range of fresh darks - so should all get the new headers.

Thanks again Aaron 👍

closed 2022-11-20 by StarGeezerPhil