aaronwmorris/indi-allsky #413: Image quality and TIF/Raw [discussion/queries]

StarGeezerPhil, 2022-11-15:
Not an issue so much as optimisation discussion.

I'm just experimenting with image formats for the "Image file type" (standard save) and "Export raw image type".

I've noticed that, despite whatever settings are used for [image file type] compression, the "export raw image" file quality is just immeasurably better.

**Example** below: Export (TIF) on left .vs. "Image" TIF LSW 5 
(I couldn't attach TIF here so screenshot simply showing the clear data differences)
[image]

I'm assuming the heavy green in the export is the debayer pattern for the IMX378.

- Is there a way to save RAW for the images used in the timelapse, startrails etc.?
- Is there a way to correct the debayering/colour for raw/export images?

Another thing I've found with my install (default Raspberry Pi Imager: Raspberry Pi Os (32-bit)) is that this time around I couldn't leave the _libcamera image type_ as "DNG (raw)", as I'm sure I used previously - instead I have had to set it to "PNG".

As per other discussion, I'm not currently having the dark frames attached, but:

- Is the libcamera image type (PNG currently) reducing the image quality of my (standard) captured images?

aaronwmorris, 2022-11-16:
> I'm assuming the heavy green in the export is the debayer pattern for the IMX378.
> 
>     Is there a way to save RAW for the images used in the timelapse, startrails etc.?
>     Is there a way to correct the debayering/colour for raw/export images?

I am not sure there is an easy way to correct the green bias.  At least not easily.  The green bias in the IMX477 and 378 was so bad it really motivated me to implement the SCNR algorithms from PixInsight to help.  The algorithm is almost magic.

> Another thing I've found with my install (default Raspberry Pi Imager: Raspberry Pi Os (32-bit)) is that this time around I couldn't leave the libcamera image type as "DNG (raw)", as I'm sure I used previously - instead I have had to set it to "PNG".

The `rawpy` python module I used to decode the DNG raw files will not build on a 32-bit OS.  This is the only thing preventing using the raw functionality on 32-bit OSs.

> Is the libcamera image type (PNG currently) reducing the image quality of my (standard) captured images?

Yes.  I had hoped that the PNG files would be lossless which is common, but libcamera implemented JPEG compression with their PNGs so they are exactly the same quality as JPEGs.

StarGeezerPhil, 2022-11-16:
Brill, thanks Aaron. So it really is a case of reinstalling.

Can you recommend the best 64-bit distro for this?

It's weird - sure I ran the same previously, but probably mistaken as it was some time back now 🤣

aaronwmorris, 2022-11-16:
If you want to use libcamera, I think you have to use Raspbian 11 64-bit (bullseye).

StarGeezerPhil, 2022-11-17:
Brill - thanks Aaron. Hadn't realised they'd renamed it Raspberry Pi OS, but seem to be operational again with libcamera using RAW images - it does appear to make quite a considerable difference to the final images. 👍

aaronwmorris, 2022-12-07:
Closing this issue for now.

closed 2022-12-07 by aaronwmorris