aaronwmorris/indi-allsky #414: indi-allsky stops and doesn't recover when disk is full

Vertikar, 2022-11-16:
**Describe the bug**
When the disk is full, image capture stops, and doesn't restart once disk space is freed up

**To Reproduce**
Steps to reproduce the behavior:
Let disk fill up

**Expected behavior**
Hopefully exceptions could be handled and image capture could restart once there is enough disk space

**Screenshots**
[image]

**Additional context**
system info:
```
Release 5.4
Uptime 2 days, 9 hours, 42 minutes, 39 seconds
<snip>
System
Python 3.9.2
pyindi 1.9.7
cv2 4.6.0
numpy 1.22.4
ephem 4.1.3
astropy 5.1.1
flask 2.2.2
ccdproc 2.3.1
paramiko 2.12.0
paho.mqtt 1.6.1
dbus 1.3.2
pycurl PycURL/7.45.1 libcurl/7.74.0 GnuTLS/3.7.1 zlib/1.2.11 brotli/1.0.9 libidn2/2.3.0 libpsl/0.21.0 (+libidn2/2.3.0) libssh2/1.9.0 nghttp2/1.43.0 librtmp/2.3
```
git info:
```
karl@allsky:/opt/indi-allsky $ git show
commit 4ae6eb4003872c99c9b23a893fe363d67499d4b5 (HEAD -> master, origin/master, origin/HEAD)
Merge: 7864aac 5508f4e
Author: Aaron W Morris <aaron@aarmor.net>
Date:   Sat Nov 5 14:23:53 2022 -0400

    Merge pull request #399 from aaronwmorris/dev

    add link to indi config wiki

karl@allsky:/opt/indi-allsky $ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
Log of error from indi-allsky.log below:

```
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image._convert_16bit_to_8bit() #1592: Resampling image from 16 to 8 bits
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.calculate_histogram() #1302: Greyscale mean: 80.38
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.calculate_histogram() #1313: Brightness average: 80.38
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.calculate_histogram() #1351: Current target ADU: 82.75 (72.75/92.75)
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.calculate_histogram() #1352: Current ADU history: (6) [81.07, 80.96, 81.26, 80.98, 79.96, 80.38]
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.calculate_histogram() #1356: ADU average: 80.77
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.white_balance_manual_bgr() #1518: Applying manual color balance settings
Nov 16 08:04:03 allsky [INFO] ImageWorker003 image.saferun() #415: Image processed in 0.3216 s
Nov 16 08:04:10 allsky [INFO] MainProcess allsky.run() #673: Camera last ready: 7.3s
Nov 16 08:04:10 allsky [INFO] MainProcess allsky.run() #674: Exposure state: OK
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception: Traceback (most recent call last):
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception:   File "/opt/indi-allsky/indi_allsky/image.py", line 864, in write_status_json
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception:     f_indi_status.flush()
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception: OSError: [Errno 28] No space left on device
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception:
Nov 16 08:04:10 allsky [ERROR] MainProcess allsky._startImageWorker() #481: Image worker exception: During handling of the above exception, another exception occurred:
[... 106 lines cut ...]
karl@allsky:/opt/allsky/images $ systemctl status --user gunicorn-indi-allsky
● gunicorn-indi-allsky.service - indi-allsky gunicorn daemon
     Loaded: loaded (/home/karl/.config/systemd/user/gunicorn-indi-allsky.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-11-14 03:07:18 AEDT; 2 days ago
TriggeredBy: ● gunicorn-indi-allsky.socket
   Main PID: 620 (gunicorn)
     Status: "Gunicorn arbiter booted"
      Tasks: 10 (limit: 779)
        CPU: 5min 31.781s
     CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/gunicorn-indi-allsky.service
             ├─   620 /opt/indi-allsky/virtualenv/indi-allsky/bin/python /opt/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi
             └─628661 /opt/indi-allsky/virtualenv/indi-allsky/bin/python /opt/indi-allsky/virtualenv/indi-allsky/bin/gunicorn --config /etc/indi-allsky/gunicorn.conf.py indi_allsky.wsgi

Nov 16 01:15:40 allsky gunicorn[626519]: [INFO] MainProcess views.dispatch_request() #1261: Wrote new config.json
Nov 16 01:15:41 allsky gunicorn[626519]: [INFO] MainProcess glogging.info() #264: Worker exiting (pid: 626519)
Nov 16 01:15:42 allsky gunicorn[628661]: [2022-11-16 01:15:42 +1100] [628661] [INFO] Booting worker with pid: 628661
Nov 16 01:15:42 allsky gunicorn.indi_allsky.wsgi.error[628661]: [628661] Booting worker with pid: 628661
Nov 16 01:16:07 allsky gunicorn[628661]: [INFO] MainProcess views.dispatch_request() #1861: Command return: /org/freedesktop/systemd1/job/73
Nov 16 01:16:07 allsky gunicorn[628661]: [INFO] MainProcess views.dispatch_request() #1861: Command return: /org/freedesktop/systemd1/job/73
Nov 16 08:14:36 allsky gunicorn[628661]: [INFO] MainProcess forms.__init__() #1739: Dates processed in 0.0213 s
Nov 16 08:14:36 allsky gunicorn[628661]: [INFO] MainProcess forms.__init__() #1739: Dates processed in 0.0213 s
Nov 16 12:40:25 allsky gunicorn[628661]: [INFO] MainProcess forms.__init__() #1739: Dates processed in 0.0066 s
Nov 16 12:40:25 allsky gunicorn[628661]: [INFO] MainProcess forms.__init__() #1739: Dates processed in 0.0066 s

```

aaronwmorris, 2022-11-16:
Merged #416 to make the main process loop more resilient to out of space conditions.

Vertikar, 2022-11-17:
Great thanks, I can probably test relatively soon and report back

Vertikar, 2022-11-20:
Looks all good, and recovers fine after clearing the space.

Here's the log output of after i filled the disk

```
Nov 20 15:35:11 allsky [INFO] MainProcess indi.newBLOB() #172: new BLOB CCD1
Nov 20 15:35:11 allsky [ERROR] MainProcess indi.newBLOB() #192: OSError: Not enough space on disk: requested 2545408, available 0. Not enough space on disk: requested 2545408, available 0. [Errno 28] No space left on device
Nov 20 15:35:11 allsky [INFO] MainProcess allsky.run() #783: Exposure received in 0.8239 s (0.8076)
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.shoot() #1276: Taking 0.01635809 s exposure (gain 0)
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.run() #841: Total time since last exposure 10.0442 s
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.run() #679: Camera last ready: 9.4s
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.run() #680: Exposure state: OK
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.detectNight() #1054: Sun altitude: 52:26:02.4
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.detectMoonMode() #1071: Moon altitide: 4:54:03.3, phase 16.9%
Nov 20 15:35:21 allsky [INFO] MainProcess indi.getCcdTemperature() #489: Sensor temperature: 26.7
Nov 20 15:35:21 allsky [INFO] MainProcess allsky._queueManualTasks() #1760: Checking for manually submitted tasks
Nov 20 15:35:21 allsky [INFO] MainProcess indi.newText() #236: new Text DRIVER_INFO for device ZWO CCD ASI224MC
Nov 20 15:35:21 allsky [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-20T04:35:21: [WARNING] Telescope focal length is missing.
Nov 20 15:35:21 allsky [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-20T04:35:21: [WARNING] Telescope aperture is missing.
Nov 20 15:35:21 allsky [INFO] MainProcess indi.newBLOB() #172: new BLOB CCD1
Nov 20 15:35:21 allsky [ERROR] MainProcess indi.newBLOB() #192: OSError: Not enough space on disk: requested 2545408, available 0. Not enough space on disk: requested 2545408, available 0. [Errno 28] No space left on device
Nov 20 15:35:21 allsky [INFO] MainProcess allsky.run() #783: Exposure received in 0.6740 s (0.6576)
Nov 20 15:35:31 allsky [INFO] MainProcess allsky.shoot() #1276: Taking 0.01635809 s exposure (gain 0)
Nov 20 15:35:31 allsky [INFO] MainProcess allsky.run() #841: Total time since last exposure 10.0429 s
Nov 20 15:35:31 allsky [INFO] MainProcess indi.newText() #236: new Text DRIVER_INFO for device ZWO CCD ASI224MC
Nov 20 15:35:31 allsky [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-20T04:35:31: [WARNING] Telescope focal length is missing.
Nov 20 15:35:31 allsky [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-20T04:35:31: [WARNING] Telescope aperture is missing.
Nov 20 15:35:31 allsky [INFO] MainProcess indi.newBLOB() #172: new BLOB CCD1
Nov 20 15:35:31 allsky [ERROR] MainProcess indi.newBLOB() #192: OSError: Not enough space on disk: requested 2545408, available 0. Not enough space on disk: requested 2545408, available 0. [Errno 28] No space left on device
Nov 20 15:35:31 allsky [INFO] MainProcess allsky.run() #783: Exposure received in 0.6705 s (0.6541)
Nov 20 15:35:32 allsky [INFO] MainProcess allsky.run() #679: Camera last ready: 0.5s
Nov 20 15:35:32 allsky [INFO] MainProcess allsky.run() #680: Exposure state: OK
```

closed 2022-11-20 by Vertikar