aaronwmorris/indi-allsky #424: pyindi-client - can't build

jkoenig72, 2022-11-22:
HI,

just run the installer on a new checkout fresh from the repo. 

Distribution: Ubuntu
Release: 22.04
Arch: x86_64

INDI_DRIVER_PATH: /usr/bin
INDISERVER_SERVICE_NAME: indiserver
ALLSKY_SERVICE_NAME: indi-allsky
GUNICORN_SERVICE_NAME: gunicorn-indi-allsky
ALLSKY_ETC: /etc/indi-allsky
HTDOCS_FOLDER: /var/www/html/allsky
DB_FOLDER: /var/lib/indi-allsky
DB_FILE: /var/lib/indi-allsky/indi-allsky.sqlite
INSTALL_INDI: true
HTTP_PORT: 80
HTTPS_PORT: 443

Building wheels for collected packages: pyindi-client, pycurl, pygifsicle, inotify, dbus-python, paho-mqtt, astroscrappy, support_developer
  Building wheel for pyindi-client (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [760 lines of output]
      /home/ubuntu/indi-allsky/virtualenv/indi-allsky/lib/python3.10/site-packages/setuptools/dist.py:530: UserWarning: Normalizing 'v1.9.1' to '1.9.1'
        warnings.warn(tmpl.format(**locals()))
      running bdist_wheel
      running build
      running build_ext
      building '_PyIndi' extension
      swigging indiclientpython.i to indiclientpython_wrap.cpp
      swig -python -v -Wall -c++ -threads -I/usr/include -I/usr/include/libindi -I/usr/local/include/libindi -o indiclientpython_wrap.cpp indiclientpython.i
      Language subdirectory: python
      Search paths:
         ./
         /usr/include/
         /usr/include/libindi/
         /usr/local/include/libindi/
         ./swig_lib/python/
         /usr/share/swig4.0/python/
         ./swig_lib/
         /usr/share/swig4.0/
      Preprocessing...
      Starting language-specific parse...
      /usr/include/libindi/indiwidgettraits.h:38: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:49: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:60: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:69: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indiwidgettraits.h:83: Warning 325: Nested struct not currently supported (UpdateArgs ignored)
      /usr/include/libindi/indipropertyview.h:257: Warning 302: Identifier 'fill' redefined (ignored),
      /usr/include/libindi/indipropertyview.h:251: Warning 302: previous definition of 'fill'.
      /usr/include/libindi/indipropertyview.h:275: Warning 302: Identifier 'fill' redefined (ignored),
      /usr/include/libindi/indipropertyview.h:251: Warning 302: previous definition of 'fill'.
      /usr/include/libindi/indipropertyview.h:354: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:359: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:486: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:491: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:650: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:655: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:779: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:784: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:908: Warning 362: operator= ignored
      /usr/include/libindi/indipropertyview.h:913: Warning 362: operator= ignored
      /usr/include/libindi/indipropertybasic.h:116: Warning 389: operator[] ignored (consider using %extend)
      /usr/include/libindi/indipropertyswitch.h:39: Warning 325: Nested struct not currently supported (NewValues ignored)
      indiclientpython.i:181: Warning 326: Deprecated %extend name used - the struct name '_IBLOB' should be used instead of the typedef name 'IBLOB'.
      /usr/include/libindi/indiproperties.h:61: Warning 389: operator[] ignored (consider using %extend)
[... 49 lines cut ...]
      /usr/include/libindi/indiproperty.h:86: Warning 508: Declaration of 'getName' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:86: Warning 508: previous declaration of 'getName'.
      /usr/include/libindi/indiproperty.h:87: Warning 508: Declaration of 'getLabel' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:87: Warning 508: previous declaration of 'getLabel'.
      /usr/include/libindi/indiproperty.h:88: Warning 508: Declaration of 'getGroupName' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:88: Warning 508: previous declaration of 'getGroupName'.
      /usr/include/libindi/indiproperty.h:89: Warning 508: Declaration of 'getDeviceName' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:89: Warning 508: previous declaration of 'getDeviceName'.
      /usr/include/libindi/indiproperty.h:90: Warning 508: Declaration of 'getTimestamp' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:90: Warning 508: previous declaration of 'getTimestamp'.
      /usr/include/libindi/indiproperty.h:91: Warning 508: Declaration of 'getState' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:91: Warning 508: previous declaration of 'getState'.
      /usr/include/libindi/indiproperty.h:92: Warning 508: Declaration of 'getStateAsString' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:92: Warning 508: previous declaration of 'getStateAsString'.
      /usr/include/libindi/indiproperty.h:93: Warning 508: Declaration of 'getPermission' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:93: Warning 508: previous declaration of 'getPermission'.
      /usr/include/libindi/indiproperty.h:96: Warning 508: Declaration of 'isEmpty' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:96: Warning 508: previous declaration of 'isEmpty'.
      /usr/include/libindi/indiproperty.h:97: Warning 508: Declaration of 'isValid' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:97: Warning 508: previous declaration of 'isValid'.
      /usr/include/libindi/indiproperty.h:99: Warning 508: Declaration of 'isNameMatch' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:99: Warning 508: previous declaration of 'isNameMatch'.
      /usr/include/libindi/indiproperty.h:99: Warning 508: Declaration of 'isNameMatch' shadows declaration accessible via operator->(),
      /usr/include/libindi/indiproperty.h:100: Warning 508: previous declaration of 'isNameMatch'.
      /usr/include/libindi/indiproperty.h:102: Warning 508: Declaration of 'isLabelMatch' shadows declaration accessible via operator->(),

vastmeridian, 2022-11-22:
Hi

I'm getting the same error, Aaron.  I won't upload the logs as they are the same as above - unless there's something specific you'd like us to try?  This was a fresh build attempted today on a virtual machine running Ubuntu 20.04.5 LTS

aaronwmorris, 2022-11-22:
There is definitely a problem with compiling pyindi-client right now.  I am investigating.

vastmeridian, 2022-11-22:
> There is definitely a problem with compiling pyindi-client right now. I am investigating.
Let us know if there are any tests you need doing.

aaronwmorris, 2022-11-22:
Merged #425 .  There are some upstream changes to pyindi-client that have not been added to an official release.  I updated the pip requirements files to pull from git instead.

vastmeridian, 2022-11-22:
> Merged #425 . There are some upstream changes to pyindi-client that have not been added to an official release. I updated the pip requirements files to pull from git instead.

I'll do a git pull origin master and let you know.

vastmeridian, 2022-11-22:
Thanks Aaron.  The ./setup.sh completes without error, now - but I forgot to go and get **indi_webcam_ccd**. Any chance you could include that in the list of drivers as it's needed for IP cameras?

aaronwmorris, 2022-11-22:
It should automatically include any drivers found.  What did it list?

vastmeridian, 2022-11-23:
indi-allsky lists every driver it finds, no problem.  I think it's when indiserver is installed, it doesn't pull down all the drivers.  I had to install indi_webcam separately and re-run setup.sh.

aaronwmorris, 2022-11-23:
The compile of pyindi-client is working with 1.9.9, but it is still not working.  I will have to open an issue with the pyindi-client team.

aaronwmorris, 2022-11-24:
Merged #426 so that the webcam driver is installed by default.

vastmeridian, 2022-11-27:
I tried an install of #426 which went fine, but, even after a reboot, the build didn't read the config properly and tried to initialise /dev/video.  Here's an extract from the indi-allsky log:

`Nov 27 21:52:01 allskyvideo [INFO] MainProcess indi.serverConnected() #245: Server connected (localhost:7624)
Nov 27 21:52:01 allskyvideo [INFO] MainProcess indi.newDevice() #157: new device INDI Webcam
Nov 27 21:52:01 allskyvideo [INFO] MainProcess indi.newDevice() #157: new device Telescope Simulator
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi._findCcds() #430: Found device INDI Webcam
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected ccd
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi._findCcds() #430: Found device Telescope Simulator
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected telescope
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi._findCcds() #435:  Detected guider
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi.findCcd() #445: Found 1 CCDs
Nov 27 21:52:09 allskyvideo [WARNING] MainProcess allsky._initialize() #365: Connecting to device INDI Webcam
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-27T21:52:09: [INFO] Trying to connect to: /dev/video0, on device: video4linux2,v4l2 with 640x480 at 30 frames per second 
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-27T21:52:09: [INFO] Attempting to connect 
Nov 27 21:52:09 allskyvideo [INFO] MainProcess indi.newMessage() #242: new Message 2022-11-27T21:52:09: [INFO] Failed to open source. Check your settings: No such file or directory `

aaronwmorris, 2022-12-07:
Merged #445 with the updated code to build the latest pyindi-client.  indi 1.9.9 is now officially released.

aaronwmorris, 2023-01-27:
Closing this issue for now.

closed 2023-01-27 by aaronwmorris