aaronwmorris/indi-allsky #428: Add a download for fits on the Image page

sifank, 2022-11-24:
We will be using this in the field for a calibrated sqm.  The rpi will have a hotspot that a tablet  will connect to, or a desktop once brought back in.  But we need a way to download 3 or 4 fits images.  How can we do that?

aaronwmorris, 2022-11-25:
I will have to add an additional database table in order to link to the FITS files.

sifank, 2022-11-25:
Thanks Aaron !!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-03:
Merged #439 to track FITS and RAW images in the database.  Links to those assets are now made available in the image viewer.

Note:  Only images created after the code is upgraded will be shown in the image viewer.

sifank, 2022-12-04:
Works great !

aaronwmorris, 2022-12-07:
Closing this issue for now.