aaronwmorris/indi-allsky #430: request for instructions on how to update software

pimmoon, 2022-11-24:
hi. I forgot what command to use to update indi-allsky software. I tried to go back and searched on issues and on main page and found no instructions.

Could you please let me know how I can update my current install to latest code? And if possible, I think it may be a good idea to add a section on how to update software on the main page or on Wiki so that it can be easily accessible.

Thank you so much!

StarGeezerPhil, 2022-11-24:
I always forget too...

Aaron added a page to the wiki for this:
https://github.com/aaronwmorris/indi-allsky/wiki/Updating-indi-allsky

pimmoon, 2022-11-25:
Oh I see. It is already on Wiki.. I didn't click "more"..

Found it and updated the code. Thank you so much!

aaronwmorris, 2022-11-25:
Eventually, I will organize the wiki a little better.

aaronwmorris, 2022-11-29:
Closing this for now.  Let me know if there is anything else I can do for you!

closed 2022-11-29 by aaronwmorris