aaronwmorris/indi-allsky #441: Additional fields for config page

sifank, 2022-12-03:
In our field sqm project, it would be nice to have 2 additional fields added:  observer and location.  Would that be possible?  These would be added as attributes in the fits headers

aaronwmorris, 2022-12-04:
I am working on a change for the location.  I decided to go ahead and integrate the GPS functionality that is already in indi for this.  Hopefully I will be finished soon.

sifank, 2022-12-05:
Thanks.  There is a GPS unit attached to these as well.  Lat/long and gps
time (local and utc) are some of the extra fields I'm adding to the fits
headers.  That would be great if gpsd functionality will be included.
Please note that the 'Location' field is a description that they add while
out in the 'field' ... (separate from gps info).

Sifan
Carpe Nocturnum ...

sifank, 2022-12-06:
> What are the field names you envision? Obviously, there is OBSERVER. COMMENT is probably a good one. I will allow them to be user defined, but I want to put in some obvious defaults.

Hi Aaron!

Yes, "Observer" and either "Site Location" or just "Location", and
"Comment" !  If there could be a field in the config for these and for
these to be added to the fits headers, that would be great.  I also
added or set the standard fields for  'OBJECT' for "AllSky", and the
'TELESCOP' field to "AllSkySqm".  I would love to also have your
'jSQM' field added.

In addition to those, I put some code in image.py after the "if
filename_p.suffix in ['.fit]:" line to check if a file exists and if
so, reads in its lines and creates fields for them in the fits
headers.  These come from environmental sensors that are also in the
unit (GPS, HTU31D temp/hum, MS8607 for Pressure, a MLX90614 for IR sky
temp, and a TLS2591 lux sensor):
ENVTEMP
ENVHUM
ENVDEW
ENVPRES
GPSLAT
GPSLONG
GPSTIME
SKYTEMP
SKYLUX

Hope that helps!

And thanks for all you do!!!  Much appreciated.

Sifan
Carpe Nocturnum ...

sifank, 2022-12-06:
> Do you see a difference between SITELAT/SITELONG vs GPSLAT/GPSLONG?
>
> There is no real standard for FITS headers, but there are some accepted
> defaults. Much of which seems to come from
> https://cdn.diffractionlimited.com/help/maximdl/FITS_File_Header_Definitions.htm
>
> If you have a better reference, I would welcome it. I admit I am not a
> full expert the FITS format.

Oops, I took a closer look at my code - yes I've used SITELAT and SITELONG
...  And that's the url I used as well.

Thx !!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-06:
I had assumed gpsd would automatically update system time, but that may be a poor assumption.  I need to order a cheap GPS adapter and fully test this.

sifank, 2022-12-06:
gpsd only supplies the information on request.  One needs to either setup
ntsd or timedatectl to take it to the next step.  Couple that with the fact
that the rpi does not have a rtc, and oh boy, messy.  Adafruit has an
'ultimate gps hat' that also has a pps (pulse per sec) output and a button
battery that together make a rtc for the pi (after setup).  One could also
use a usb gps but then you would probably have to have a crontab job to
update the system time every so often.  One problem with that, is the first
time could be set way off, so you would have to wait for it to stabilize.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-07:
Merged #445 with the GPS functionality.  I will work on the time synchronization when I get the hardware available.

sifank, 2022-12-07:
I've downloaded 445 and it looks great.

Would it be possible for the Configuration web page to ask for
Observer and Location (or Site)?  Or is that still coming?

Thanks!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-11:
Merged #454 to allow syncing the system time with GPS.

I will be working on the additional fits headers soon.

sifank, 2022-12-11:
That's wonderful, I'll check it out shortly!

Thx!

Sifan
Carpe Nocturnum ...

sifank, 2022-12-12:
Confirmed.  I had to think about how I could test this!!  Decided to shut
down the services then set the time to something else, then reboot.  On
reboot, it sets the system time correctly.

I also tested it by removing the gps antenna and putting the unit indoors,
so no gps signal.  That works as well.

But like the camera cooling, if I completely disabled the gps daemon, but
still have it enabled in the config, indi-allsky will halt and stop.  But
again, that would only happen on 'operator' error ...  Plus the log gives
the details on why.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-15:
The time sync certainly works, but it is a bit of a minefield as far as potential problems.  Syncing time to a local NTP daemon synced to GPS is still probably a better option.

closed 2023-01-22 by sifank