aaronwmorris/indi-allsky #443: crashes if cooling set but pwr cable to cam cooler disconnects

sifank, 2022-12-04:
I repeated this test a number of times:  I have the Indi-config, in the config page set to camera cooling.  This works when the power to the camera cooling is attached (ASI294mcPro).  But when/if that cable comes off or is disconnected or power stops it to, indi-allsky stops taking images and the log gives a timeout error trying to set the cam cooler temp.

Would it be possible for this to be a warning and for indi-allsky to just continue without the cooler?

sifank, 2022-12-05:
I think I have more information on the cooler problem.  In watching the logs, I see that it use the ramp parameter to command the camera to the temp in a step-wise fashion.  But it seems that if the temperature didn't reach the expected step for the next iteration, it errors out:
Dec  4 22:44:32 allskycam [INFO] MainProcess indi.configureDevice() #464: Setting property (number) CCD_TEMPERATURE
Dec  4 22:44:32 allskycam [INFO] MainProcess indi.newSwitch() #229: new Switch CCD_COOLER for device ZWO CCD ASI294MC Pro
Dec  4 22:44:32 allskycam [INFO] MainProcess indi.newMessage() #242: new Message 2022-12-05T06:44:32: [INFO] Setting temperature to 6.40 C. 
Dec  4 22:45:32 allskycam [INFO] MainProcess indi.newSwitch() #229: new Switch CCD_COOLER for device ZWO CCD ASI294MC Pro
Dec  4 22:45:32 allskycam [INFO] MainProcess indi.newMessage() #242: new Message 2022-12-05T06:45:32: [INFO] Setting temperature to 2.60 C. 
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #33: An uncaught exception occurred:
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #34: Type: <class 'indi_allsky.exceptions.TimeOutException'>
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #35: Value: Timeout error while changing property CCD_TEMPERATURE: elapsed=65.0649516582489, timeout=65.0, status=BUSY
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/allsky.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/allsky.py", line 661, in run\n    self._initialize()\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/allsky.py", line 393, in _initialize\n    self.indiclient.configureCcdDevice(self.config[\'INDI_CONFIG_DEFAULTS\'])\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/camera/indi.py", line 478, in configureCcdDevice\n    self.configureDevice(self._ccd_device, *args, **kwargs)\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/camera/indi.py", line 465, in configureDevice\n    self.set_number(device, k, v)\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/camera/indi.py", line 737, in set_number\n    self.__wait_for_ctl_statuses(c, timeout=timeout)\n'
Dec  4 22:45:37 allskycam [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/sifan/Projects/indi-allsky/indi_allsky/camera/indi.py", line 825, in __wait_for_ctl_statuses\n    raise TimeOutException(\'Timeout error while changing property {0}: elapsed={1}, timeout={2}, status={3}\'.format(ctl.getName(), elapsed, timeout, self.__state_to_str_p[ctl.getState()] ))\n'

If I keep clicking on Start indi-allsky (services) it will continue, then error, etc, until it reaches it's setpoint.

There is a different error that occurs if the cam's cooler pwr is not connected - I'll try to capture and report that next.

sifank, 2022-12-05:
Starting without power to the cam cooler but with INDI Configuration field set for cooling as per the wiki, I get the same set of errors messages as reported above.

I also tried setting the indi config to just the temp (without the ramp) and had the same results:
{
    "SWITCHES": {},
    "PROPERTIES": {
        "CCD_TEMPERATURE": {
            "CCD_TEMPERATURE_VALUE": -10.0
        }
    },
    "TEXT": {}
}
It seems that in this case, if I set the temp to -10, it would waited the same amount of time, check and see that it didn't reach the target, and boom, this error.

I've never seen this issue in Kstars/EKOS, it will let me take images even if the cam is not at it's setpoint.  It's up to the observer to wait or continue.  Sometimes I want to continue, as in that case I can do platesolving and other preps using the cam before I'm actually going to take serious images.

Would it be possible for something like that here?  Allow it to continue and it's up to the user to check the ccd temp before they start doing serious data taking?  In our case, this would allow the observer to take/see allsky images and be able to reposition in the field (again this application is for measuring dark skies), check sky conditions, etc. before actually taking their meaningful data.

Thanks!

aaronwmorris, 2022-12-05:
I will probably have to create a separate function to enable cooling.  I have been planning on creating a separate config option to enable cooling anyway.

aaronwmorris, 2022-12-09:
Merged #447 to better support camera cooling.

You will need to remove the target temperature configuration from your custom configuration and configure the cooling parameters in the config.  You can keep the temperature ramp configuration in your custom config.

sifank, 2022-12-09:
Thanks Aaron, I'll check it out shortly !!

Sifan
Carpe Nocturnum ...

sifank, 2022-12-10:
All checks out well.

One issue I noted, not sure if this makes a difference:  if I replace a
cooled camera (ASI294mc) with a non-cooled one (ASI120mc) I get errors in
the log if I forget to change the config.  I'll have to check again
(probably in the morning) if that halted operations or not.

This is 'operator error', but maybe there could be some kind of failsafe ..

Thanks!!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-10:
> if I replace a cooled camera (ASI294mc) with a non-cooled one (ASI120mc) I get errors in the log if I forget to change the config

There are checks to try to do the right thing if cooling is enabled with a non-cooled camera.  As long as the driver properly assigns the cooling properties with the read-only state, indi-allsky should continue running.

sifank, 2022-12-11:
It crashes if I have a ramp conf in the Indi Config box (while having non
cooled cam):
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#33: An uncaught exception occurred:
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#34: Type: <class 'indi_allsky.exceptions.TimeOutExc
eption'>
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#35: Value: Timeout finding control CCD_TEMP_RAMP
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/allsk
y.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/allsky.py", line 793, in run\n    self._initialize()\n'
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/allsky.py", line 519, in _initialize\n
   self.indiclient.configureCcdDevice(self.config[\'INDI_CONFIG_DEFAULTS\'])\n'

Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/camera/indi.py", line 638, in configureCcdDevice\n
   self.configureDevice(self._ccd_device, *args, **kwargs)\n'
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/camera/indi.py", line 625, in configureDevice\n
   self.set_number(device, k, v)\n'
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/camera/indi.py", line 1044, in set_number\n    c =
self.get_control(device, name, \'number\')\n'
Dec 10 18:11:06 allskycam [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/indi_
allsky/camera/indi.py", line 1035, in get_control\n    raise
TimeOutException(\'Timeout finding control {0}\'.format(name))\n'

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-11:
After thinking about it, I think I am going to leave the configuration processing as-is.  I would rather than verbosely crash than silently continue.

The problem is there is no way to validate the configuration.  There is no list of valid configuration options that are valid across all cameras.  Having the program generate exceptions is the only indication that the configuration is bad.  If I bypass invalid options, it gives the impression that the configuration was successful.

You can comment out the configuration to leave in place for later.  Prefix the first level options with `#`

```
{
    "#PROPERTIES" : {
        "CCD_TEMP_RAMP" : {
            "RAMP_SLOPE"     : 5,
            "RAMP_THRESHOLD" : 0.5
        }
    },
    "#SWITCHES" : {}
}
```

sifank, 2022-12-12:
Sounds good, especially since one doesn't change cameras all that often!

Thanks!

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-22:
Closing this for now.  Thanks!

closed 2022-12-22 by aaronwmorris