aaronwmorris/indi-allsky #444: ffmpeg quietly fails to generate timelapse

Elliot-Ford, 2022-12-05:
ffmpeg fails without warning to generate a timelapse. It does generate a file for the timelapse 64mb in size with the following contents:
```
00000000: 0000 0020 6674 7970 6973 6f6d 0000 0200  ... ftypisom....
00000010: 6973 6f6d 6973 6f32 6176 6331 6d70 3431  isomiso2avc1mp41
00000020: 0000 0008 6672 6565 0000 0000 6d64 6174  ....free....mdat
```
this looks to me like a empty mp4 file and doing some digging I think the issue is an out of memory situation, however the artifacts and report of success is misleading. Would it be possible to check the success of the subprocess command (similar to "echo $?") and mark the execution as a failure (and cleanup artifacts)?

### hardware & driver
- raspberrypi 4 2Gb
- raspberrypi hq cam
- libcamera_imx477

aaronwmorris, 2022-12-07:
I agree I need to be more verbose about the failures.  I really need to split the timelapse generation from the keogram in to separate task runs.  That way each can be tracked individually--currently they run as part of the same task.   So, even if the the video fail, the keogram may be successful but the overall execution is considered "successful".

As for cleaning up the artifacts, that is a little more difficult.  The timelapse viewer in the web interface relies on an entry in the "video" table in the database.  Without the DB entry, none of the keograms or startrails would display for that day.  If the file does not exist (even a remnant) the DB validator would delete the DB entry.  This is one of those "it's a feature, not a bug" situations.

As for the 2GB Raspi4... I think you would need to use the 50% ffmpeg scaling to be able to generate a timelapse video without running out of memory (for the imx477 camera).  75% might work, but you may need a lot of swap space.

Elliot-Ford, 2022-12-07:
Hmm, I've attempted to address this with #446, I've tried to handle the db cleanup as well. Not sure if I caught the keogram generation issue (didn't look as far). Have tested it locally and it does catch the file change (and doesn't complain about the database entry deletion)

aaronwmorris, 2022-12-08:
> Would the deletion of the files still be an issue?

It would only be a problem if you ran a "Validation" from the system page.  The validation would detect the missing file and delete the database entry.  This would result in none of the assets for the day displayed in the web interface.

I am thinking about adding a flag in the DB to indicate if the video was successful.  This would negate the need for the file to exist.

aaronwmorris, 2022-12-11:
I merged your PR, but I ended up making extensive changes.  I replaced the return codes in favor of using exceptions.

#452 

The tasks for timelapses and keogram/startrails are completely separated now.

Elliot-Ford, 2022-12-12:
Thanks for quick review and merge. Closing this Issue.

closed 2022-12-12 by Elliot-Ford