aaronwmorris/indi-allsky #449: allsky.unhandled_exception error after install of fresh OS install on RPI4.

zetafunc, 2022-12-10:
Hi,

This software looks amazing and I hope to be able to use it. I'm getting an error on startup.  My system is:

OS:  Raspberry Pi OS (64bit)  2022-09-22  - Debian GNU/Linux 11 (bullseye)
Hardware: Raspberry Pi 4 Model B Rev 1.5 - 2GB RAM.   Raspberry Pi HQ Camera
cat /proc/version
Linux version 5.15.76-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022

During setup.sh, I selected to use "libcamera_imx477", INDI version "v1.9.9" and there were no apparent errors.

cat /var/log/indi-allsky/indi-allsky.log

Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #347: indi-allsky release: 6.0
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #348: indi-allsky config version: 20221129.0
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #350: Python version: 3.9.2
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #351: Platform: aarch64
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #353: System CPUs: 4
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #358: System memory: 1849 MB
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #361: System uptime: 21s
Dec 10 18:44:02 indi-allsky [INFO] MainProcess indi.__init__() #134: creating an instance of IndiClient
Dec 10 18:44:02 indi-allsky [INFO] MainProcess indi.__init__() #141: PyIndi version: 1.9.7
Dec 10 18:44:02 indi-allsky [INFO] MainProcess allsky._initialize() #382: Connecting to indiserver
Dec 10 18:44:02 indi-allsky [INFO] MainProcess indi.serverConnected() #285: Server connected (localhost:7624)
Dec 10 18:44:02 indi-allsky [INFO] MainProcess indi.newDevice() #197: new device CCD Simulator
Dec 10 18:44:02 indi-allsky [INFO] MainProcess indi.newDevice() #197: new device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #555: Found device CCD Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #560:  Detected ccd
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #560:  Detected guider
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #560:  Detected filter
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #555: Found device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #560:  Detected telescope
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findTelescopes() #560:  Detected guider
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.findTelescope() #570: Found 1 Telescopess
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #586: Found device CCD Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #591:  Detected ccd
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #591:  Detected guider
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #591:  Detected filter
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #586: Found device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #591:  Detected telescope
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi._findGpss() #591:  Detected guider
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.findGps() #601: Found 0 GPSs
Dec 10 18:44:10 indi-allsky [WARNING] MainProcess allsky._initialize() #402: Connecting to CCD device libcamera_imx477
Dec 10 18:44:10 indi-allsky [WARNING] MainProcess allsky._initialize() #406: Connecting to Telescope device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.configureDevice() #619: Setting switch TELESCOPE_SLEW_RATE
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newMessage() #282: new Message 2022-12-10T07:44:10: [INFO] Telescope simulator is online.
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newSwitch() #269: new Switch CONNECTION for device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newMessage() #282: new Message 2022-12-10T07:44:10: [INFO] update mount and pier side: Pier Side On, mount type 2
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newMessage() #282: new Message 2022-12-10T07:44:10: [INFO] InitPark: No Park data in file /home/ade/.indi/ParkData.xml: No such file or directory
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newMessage() #282: new Message 2022-12-10T07:44:10: [INFO] Mount is unparked.
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newSwitch() #269: new Switch TELESCOPE_PARK for device Telescope Simulator
Dec 10 18:44:10 indi-allsky [INFO] MainProcess indi.newText() #276: new Text TIME_UTC for device Telescope Simulator
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #33: An uncaught exception occurred:
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #34: Type: <class 'TypeError'>
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #35: Value: Wrong number or type of arguments for overloaded function 'BaseClient_sendNewSwitch'.#012  Possible C/C++ prototypes are:#012    INDI::BaseClient::sendNewSwitch(ISwitchVectorProperty *)#012    INDI::BaseClient::sendNewSwitch(char const *,char const *,char const *)
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/allsky.py", line 128, in <module>\n    action_func(*args_list, **kwargs_dict)\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/indi_allsky/allsky.py", line 793, in run\n    self._initialize()\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/indi_allsky/allsky.py", line 492, in _initialize\n    self.indiclient.configureTelescopeDevice(telescope_config)\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/indi_allsky/camera/indi.py", line 646, in configureTelescopeDevice\n    self.configureDevice(self._telescope_device, *args, **kwargs)\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/indi_allsky/camera/indi.py", line 620, in configureDevice\n    self.set_switch(device, k, on_switches=v.get(\'on\', []), off_switches=v.get(\'off\', []))\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/indi_allsky/camera/indi.py", line 1084, in set_switch\n    self.sendNewSwitch(c)\n'
Dec 10 18:44:10 indi-allsky [ERROR] MainProcess allsky.unhandled_exception() #40: '  File "/home/ade/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/PyIndi.py", line 767, in sendNewSwitch\n    return _PyIndi.BaseClient_sendNewSwitch(self, *args)\n'

aaronwmorris, 2022-12-10:
You just selected the wrong version of indi during the setup.  The Astroberry APT repo has not been upgraded to indi 1.9.9 yet (still at version 1.9.7)

You can run `./misc/rebuild_pyindi-client.sh` and select 1.9.7, you should be all set.

zetafunc, 2022-12-14:
Thanks Aaron - that worked a treat!  Loving the software.

closed 2022-12-14 by zetafunc