aaronwmorris/indi-allsky #451: Workers sleeping in place of blocking

Elliot-Ford, 2022-12-11:
I noticed that the workers spinlooping on the task queue instead of blocking on reads from the queue. Is there a reason for this that I'm missing? I've made the change on my local version and haven't had an issue and have noticed a considerable decrease in system load during normal operation (in the range of maybe 20-30%), since the worker halts until a new task is available instead of looping. 

Thanks!

aaronwmorris, 2022-12-11:
It may have been due to pulling some of my queuing code from python 2.7 scripts I have written.  Or maybe I used to perform shutdowns of the workers in a different way.  To be honest, I do not recall.  :-)

There is this note in the queue module:

```
Prior to 3.0 on POSIX systems, and for all versions on Windows, if block is true and timeout is None, this operation goes into an uninterruptible wait on an underlying lock. This means that no exceptions can occur, and in particular a SIGINT will not trigger a [KeyboardInterrupt](https://docs.python.org/3/library/exceptions.html#KeyboardInterrupt).
```

I will do some testing.

aaronwmorris, 2022-12-13:
I will meet you in the middle with #457.  I have altered the way the workers wait on data.  Instead of using no_wait, I changed the code to use blocking gets with a timeout.

Elliot-Ford, 2022-12-22:
Okay, sounds like a good compromise, don't fully understand the choice of prime numbers though.

aaronwmorris, 2022-12-22:
So, my theory behind the prime numbers is to ensure the sleep cycles of the different processes align as little as possible.  Using prime numbers ensures they wake up at different offsets to reduce contention.

In reality, does it help?  Probably not.  :-)  But I do not think it hurts either.

Elliot-Ford, 2022-12-22:
Gotcha, yes in reality it won't help and/or matter. With the blocking operation the workers will just consume the provided tasks as fast as possible so "alignment" of various tasks isn't so much achieved, you could maybe do a "cooldown" after the completion of a task that would achieve this goal. But I wouldn't worry too much about alignment, this is something the OS should/does handle. Either way, thanks for making the change. Seeing much better load numbers now with the removal of the spinlock :)

closed 2022-12-22 by Elliot-Ford