aaronwmorris/indi-allsky #460: New config fields for "Site" and "Observer"

sifank, 2022-12-15:
Would it be possible to add two new fields to the Config page to ask for "Site" and "Observer" and use those fields in the fits headers?

aaronwmorris, 2022-12-17:
In #463 I added 5 custom FITS header fields in the configuration that can be added to the files.

sifank, 2022-12-18:
That works!  The fields are populated in the fits headers.

But something weird is happening:  the image on the Image tab is not
updating, even tho new fits images are being generated and saved.  Also
noted that no new /var/www/html/allsky/images/latest.png is not being
updated.  That part of the config was not changed, it still shows png as
the type and the correct path to store them.

Any ideas?

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-18:
The big refactor I just performed to add image stacking, I disabled generating images because I thought that was the way it was working before.

I re-enabled writing images in #464

sifank, 2022-12-18:
Wheeeee!!  Thank you!  Thought I was going nuts.  I was trying a number of
different configs to see if it was something I did.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-22:
Closing this for now.  Please reopen if you need anything else here.

closed 2022-12-22 by aaronwmorris