aaronwmorris/indi-allsky #468: INDI-Allsky configured for IP camera continues to try to read /dev/video0

vastmeridian, 2022-12-21:
**Processor**: Virtualbox VM configured with 2GB RAM and 20GB SSD
**Camera**: IP camera permanently running with indi-webcam driver (downloaded from ppa:mutlaqja/ppa)
**Operating System**: Ubuntu 20.04.5 LTS (GNU/Linux 5.4.0-131-generic x86_64)
**indi-allsky**: Release 6.0

Issue: When the server is initially started, the indi-allsky service stops after a short while and remains DOWN. Despite INDI config as shown below, indi-allsky persists in trying to read /dev/video0.

indi-config:
`{
    "PROPERTIES": {},
    "TEXT": {
        "ONLINE_PATH": {
            "URL_PATH": "http://username:password@10.11.12.13/jpg/image.jpg"
        }
    },
    "SWITCHES": {
        "CAPTURE_DEVICE": {
            "on": [
                "IP Camera"
            ]
        },
        "ONLINE_PROTOCOL": {
            "on": [
                "HTTP"
            ],
            "off": [
                "CUSTOM"
            ]
        }
    }
}`

indi-allsky log
`Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi._findGpss() #592:  Detected telescope
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi._findGpss() #592:  Detected guider
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.findGps() #602: Found 0 GPSs
Dec 21 06:32:56 allskyvideo [WARNING] MainProcess allsky._initialize() #408: Connecting to CCD device INDI Webcam
Dec 21 06:32:56 allskyvideo [WARNING] MainProcess allsky._initialize() #412: Connecting to Telescope device Telescope Simulator
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.configureDevice() #620: Setting switch TELESCOPE_SLEW_RATE
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-21T06:32:56: [INFO] Trying to connect to: /dev/video0, on device: video4linux2,v4l2 with 640x480 at 30 frames per second
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-21T06:32:56: [INFO] Attempting to connect
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.newMessage() #283: new Message 2022-12-21T06:32:56: [INFO] Failed to open source. Check your settings: No such file or directory
Dec 21 06:32:56 allskyvideo [INFO] MainProcess indi.newSwitch() #270: new Switch CONNECTION for device INDI Webcam
`

aaronwmorris, 2022-12-22:
I would delete all XML files in `~/.indi/` and then restart the indisever service.  If you have connected with KStars/EKOS, it will save a default configuration file in ~/indi/ which can cause the camera driver to start in a different default state than expected.

I just retested my Reolink camera and it worked fine, although, I did have to start indi-allsky twice to get it to fully start.  This was due to one of the camera parameters not being available on the first start.

vastmeridian, 2022-12-22:
There are no files in ~/.indi/

vastmeridian, 2022-12-22:
Here's the indi_getprop for the driver:
[IP-Camera-indi-getprop.txt](https://github.com/aaronwmorris/indi-allsky/files/10284010/IP-Camera-indi-getprop.txt)

I don't run KSTARS or EKOS on my systems - maybe I should run an instance as it may help with setting up or checking configs.  I do feel like I'm flying blind with the system otherwise.

aaronwmorris, 2022-12-22:
Can you check `/usr/share/indi/indi_webcam.xml` and make sure it is basically the minimal configuration?

This is the reference file in github:  https://github.com/indilib/indi-3rdparty/blob/master/indi-webcam/indi_webcam.xml.cmake

vastmeridian, 2022-12-23:
I checked that.  It's the same as the example you linked to on github.

I even tried adding another switch to the config and re-starting indiserver.  That didn't work either:

```
{
    "PROPERTIES": {},
    "TEXT": {
        "ONLINE_PATH": { "URL_PATH": "http://user:pass@11.22.33.44/jpg/image.jpg"  }
    },
    "SWITCHES": {
        "CAPTURE_DEVICE": {
            "on": [ "IP Camera"  ],
            "off": [ "video4linux2,v4l2"  ]
        },
        "ONLINE_PROTOCOL": {
            "on": [ "HTTP" ],
            "off": [ "CUSTOM"  ]
        }
    }
}
```
The logfile from stopping both services and restarting is attached.
[indi logfile 20221223.txt](https://github.com/aaronwmorris/indi-allsky/files/10292193/indi.logfile.20221223.txt)

aaronwmorris, 2023-01-27:
Merged #538 to make the CCD_FRAME_TYPE property optional during initialization.

vastmeridian, 2023-01-29:
My AllSky cameras are being re-housed, so they're not operational at the moment.  I will test this as soon as the IP camera is back online.  Thanks.

vastmeridian, 2023-04-23:
I don't think I will be resurrecting the video camera (IP-based) version of the allsky in the near future.  indi-allsky and indi-server have both moved on considerably too, so I think it best if we close this and I re-open it if I get problems in the future.

closed 2023-04-23 by vastmeridian