aaronwmorris/indi-allsky #471: Star trails image inclusion

alexdumitriu, 2022-12-21:
Hi,

I am trying to get the startrails to generate, but it seems like only 1 image is included most times. I am in a light polluted area, so what would be a good value to set for Star Trails Max ADU and Star Trails Mask Threshold? Also, do I need to change anything on the Star Trails Pixel Threshold?

Many thanks
Alex

aaronwmorris, 2022-12-21:
#473 and #474 include a new view that help determine the correct Max ADU.

Update the the latest code and go to https://hostname/indi-allsky/adu and it will show you some average stats over 15 minute increments for the past 3 days (night time only).  Basically, you need a Max ADU value that is just above your baseline average ADU.

If you want to paste the resulting table here, we can go through that.

aaronwmorris, 2023-06-22:
Today I merged #808 which excludes images from the star trails when the sun is above -15 degrees.  This can significantly improve the star trails.

StarGeezerPhil, 2023-08-21:
@aaronwmorris this is really useful information - would be good in the wiki/FAQ.

Please also see suggestion for sample data for testing/optimising config (in discussions) - i.e. 10 minutes of good RAW data saved for use in tweaking the Max ADU / Threshold/ Pixel settings.

aaronwmorris, 2024-01-19:
#1076 adds a new option in the image viewer to exclude individual images from the timelapse and startrails.

aaronwmorris, 2024-04-28:
#1223 Adds the ability to use star counts for filtering images for star trails.

closed 2024-04-28 by aaronwmorris