aaronwmorris/indi-allsky #472: /tmp filling up with .fits files that are not pruned

vastmeridian, 2022-12-21:
**Processor**: SECO picoITX industrial computer c/w Intel Atom N2800 - 4GB RAM and 32G SSD
**Camera**: Altair GPCam3 290C
**Operating System**: Ubuntu 20.04.5 LTS (GNU/Linux 5.4.0-131-generic x86_64)
**indi-allsky: Release** 6.0

I have been trying to get the Altair 290C working with indi-allsky and finally was able to get images from the camera thiese last two days.  However, within hours, the computer had filled up the disk space available.  Use of monitoring and the following command led me to look into the /tmp directory only to find a large number of .fit files 

`du -shx * | sort -rh | head -10`

Removing those .fit cured my issue for a short time but the runaway disk usage is happening again:
[image]

Is there perhaps some housekeeping that is not taking place?  I have fits files turned off in the config.

aaronwmorris, 2022-12-21:
Normally, what happens when indi-allsky receives an image from the camera, the Main process writes the image in FITS format to the /tmp file system.  The Image process then reads the FITS file and immediately deletes it.

The Image process will process each image in the order it is received.  If anything slows down the image processing, the whole process gets backed up and you can end up with a bunch of files in the /tmp folder.

Do you notice any exceptions in the log?  Can you post a bit of the log during the time the /tmp file system was filling up?

vastmeridian, 2022-12-22:
All of the above makes total sense, now I understand the process.  Mea culpa this time.  I think my issue was twofold:
- The SECO computer isn't a quick computer
- The computer was in focus mode.  

Extract from the log:
`Dec 21 19:29:46 allsky [INFO] MainProcess indi.newBLOB() #213: new BLOB CCD1
Dec 21 19:29:47 allsky [INFO] MainProcess allsky.run() #933: Exposure received in 0.5084 s (0.4276)
Dec 21 19:29:47 allsky [INFO] MainProcess allsky.shoot() #1515: Taking 0.08071957 s exposure (gain 2000)
Dec 21 19:29:47 allsky [INFO] ImageWorker001 stars.detectObjects() #84: Star detection in 0.6407 s
Dec 21 19:29:47 allsky [INFO] ImageWorker001 stars.detectObjects() #86: Found 0 objects
Dec 21 19:29:47 allsky [INFO] ImageWorker001 scnr.maximum_neutral() #89: SCNR maximum neutral in 0.0257 s
Dec 21 19:29:47 allsky [INFO] ImageWorker001 image.white_balance_manual_bgr() #1709: Applying manual color balance settings
Dec 21 19:29:47 allsky [WARNING] MainProcess allsky.run() #986: *** FOCUS MODE ENABLED ***
Dec 21 19:29:47 allsky [INFO] MainProcess allsky.run() #993: Total time since last exposure 1.0132 s
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image._image_text() #1912: Focus mode enabled, Image labels disabled
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.saferun() #337: Image processed in 1.8292 s
Dec 21 19:29:48 allsky [INFO] MainProcess indi.newBLOB() #213: new BLOB CCD1
Dec 21 19:29:48 allsky [INFO] MainProcess allsky.run() #933: Exposure received in 0.6151 s (0.5344)
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.write_img() #782: Image compressed in 0.2531 s
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image.write_img() #799: Focus mode enabled, not saving timelapse image
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.add() #1292: Image bits: 16, cfa: RGGB
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image._detectBitDepth() #1336: Image max value: 107
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image._detectBitDepth() #1361: Detected bit depth: 8
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image._calibrate() #1388: Searching for bad pixel map: gain 2000, exposure >= 0.0, temp >= 21.2c
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image._calibrate() #1405: Temperature matched bad pixel map not found: 21.20c
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image._calibrate() #1423: Bad Pixel Map not found: ccd1 16bit 0.0301478s gain 2000 bin 1 21.20c
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image._calibrate() #1435: Searching for dark frame: gain 2000, exposure >= 0.0, temp >= 21.2c
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image._calibrate() #1452: Temperature matched dark not found: 21.20c
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image._calibrate() #1470: Dark not found: ccd1 16bit 0.0301478s gain 2000 bin 1 21.20c
Dec 21 19:29:48 allsky [INFO] ImageWorker001 sqm.calculate() #21: Exposure: 0.030148, gain: 2000
Dec 21 19:29:48 allsky [INFO] ImageWorker001 sqm.calculate() #37: Raw SQM average: 44.70
Dec 21 19:29:48 allsky [INFO] ImageWorker001 sqm.calculate() #42: Weighted SQM average: 111.62
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.saferun() #240: Image: 1920 x 1080
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image._convert_16bit_to_8bit() #1631: Resampling image from 16 to 8 bits
Dec 21 19:29:48 allsky [INFO] MainProcess allsky.shoot() #1515: Taking 0.08071957 s exposure (gain 2000)
Dec 21 19:29:48 allsky [WARNING] MainProcess allsky.run() #986: *** FOCUS MODE ENABLED ***
Dec 21 19:29:48 allsky [INFO] MainProcess allsky.run() #993: Total time since last exposure 1.0183 s
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.calculate_histogram() #909: Greyscale mean: 44.79
Dec 21 19:29:48 allsky [INFO] ImageWorker001 image.calculate_histogram() #920: Brightness average: 44.79
Dec 21 19:29:48 allsky [WARNING] ImageWorker001 image.recalculate_exposure() #1010: New calculated exposure: 0.080777
Dec 21 19:29:48 allsky [INFO] ImageWorker001 detectLines.detectLines() #78: Line detection in 0.0425 s
Dec 21 19:29:48 allsky [INFO] ImageWorker001 detectLines.detectLines() #81: Detected 0 lines
Dec 21 19:29:49 allsky [INFO] MainProcess indi.newBLOB() #213: new BLOB CCD1
Dec 21 19:29:49 allsky [INFO] MainProcess allsky.run() #933: Exposure received in 0.5638 s (0.4830)
Dec 21 19:29:49 allsky [INFO] MainProcess allsky.shoot() #1515: Taking 0.08077709 s exposure (gain 2000)
Dec 21 19:29:49 allsky [INFO] ImageWorker001 stars.detectObjects() #84: Star detection in 0.6401 s
Dec 21 19:29:49 allsky [INFO] ImageWorker001 stars.detectObjects() #86: Found 0 objects
Dec 21 19:29:49 allsky [INFO] ImageWorker001 scnr.maximum_neutral() #89: SCNR maximum neutral in 0.0257 s
Dec 21 19:29:49 allsky [INFO] ImageWorker001 image.white_balance_manual_bgr() #1709: Applying manual color balance settings
Dec 21 19:29:49 allsky [WARNING] MainProcess allsky.run() #986: *** FOCUS MODE ENABLED ***
Dec 21 19:29:49 allsky [INFO] MainProcess allsky.run() #993: Total time since last exposure 1.0171 s
Dec 21 19:29:50 allsky [WARNING] ImageWorker001 image._image_text() #1912: Focus mode enabled, Image labels disabled
Dec 21 19:29:50 allsky [INFO] ImageWorker001 image.saferun() #337: Image processed in 1.7579 s
Dec 21 19:29:50 allsky [INFO] MainProcess indi.newBLOB() #213: new BLOB CCD1
Dec 21 19:29:50 allsky [INFO] MainProcess allsky.run() #933: Exposure received in 0.6238 s (0.5430)
Dec 21 19:29:50 allsky [INFO] ImageWorker001 image.write_img() #782: Image compressed in 0.2716 s
[... cut ...]
Dec 21 19:29:50 allsky [INFO] ImageWorker001 image.calculate_histogram() #909: Greyscale mean: 44.81
Dec 21 19:29:50 allsky [INFO] ImageWorker001 image.calculate_histogram() #920: Brightness average: 44.81
Dec 21 19:29:50 allsky [WARNING] ImageWorker001 image.recalculate_exposure() #1010: New calculated exposure: 0.080733
Dec 21 19:29:51 allsky [INFO] ImageWorker001 detectLines.detectLines() #78: Line detection in 0.0547 s
Dec 21 19:29:51 allsky [INFO] ImageWorker001 detectLines.detectLines() #81: Detected 0 lines
Dec 21 19:29:51 allsky [INFO] MainProcess indi.newBLOB() #213: new BLOB CCD1`

aaronwmorris, 2022-12-22:
Merged #475 to disable image processing during focus mode.

vastmeridian, 2022-12-23:
Thanks for that, Aaron.  Sounds much tidier.  You do kind of want the focus mode to just display images as quickly as possible, so disabling image processing will definitely help.  
You might also consider checking (and clearing) the focus mode flag on start-up as I hadn't realised the flag was still set when I resumed testing of my camera.

I think I can close this issue now.

closed 2022-12-23 by vastmeridian

aaronwmorris, 2022-12-23:
I added an indicator on the main view to show when focus mode is enabled.  Hopefully this makes things clearer in the future.