aaronwmorris/indi-allsky #476: unable to resolve host *: Name or service not known

dbainbridge, 2022-12-22:
When running ./setup.sh I get a lot of these dumped in all the setup steps:
`**** Setup rsyslog logging ****
sudo: unable to resolve host odroid: Name or service not known
sudo: unable to resolve host odroid: Name or service not known
sudo: unable to resolve host odroid: Name or service not known
`

`hostname -s` does return `odroid`

I can also ping `odroid.local` without issue.

aaronwmorris, 2022-12-22:
These are probably benign errors you can ignore.  However if you want to fix them, you can add `odroid` as an alias to the loopback (127.0.0.1) IP address.

`/etc/hosts` could look something like this:
```
127.0.0.1       localhost odroid
```

dbainbridge, 2022-12-22:
I created a [PR](https://github.com/aaronwmorris/indi-allsky/pull/478) that informs user of what to do so that others won't wonder what the issue is.

dbainbridge, 2022-12-22:
I did verify on my system this resolved the warnings.  I also did not see an issue even without updating /etc/host but I did not like seeing all those warnings.

aaronwmorris, 2022-12-22:
Undestood!  Merged #478