aaronwmorris/indi-allsky #477: No images on the Images page

sifank, 2022-12-22:
I'm on pr 475, images are appearing on the Latest page ok and are being stored, but there is no history on the Images page.  I do have timelapses, Meteor detection and stacking are disabled.  Daytime is enabled.  I also tried flushing images and validating the DB.

aaronwmorris, 2022-12-22:
If you have `Daytime Capture` enabled, but `Daytime Timelapse` disabled, it will lead to this behavior.  Images will be taken, but not stored.

sifank, 2022-12-22:
Is that the same for nighttime?

Sifan
Carpe Nocturnum ...

sifank, 2022-12-22:
Ah, I see that the daytime fits images are being stored, but not the png.
I turned on timelapse and yup, it's now showing up on the Images page.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-22:
I implemented it this way to replicate the behavior of TJ's allsky.  (At the time I used it) It basically had this behavior where it would display the latest daytime image, but not actually save it.

I do not have to keep it this way, its just a legacy behavior.

sifank, 2022-12-22:
I think this will work for us.  As part of their field procedure, I put
down that they should purge images and timelapses.

Thanks again for all your help!!

Happy Holidays,

Sifan
Carpe Nocturnum ...

closed 2022-12-28 by sifank