aaronwmorris/indi-allsky #479: indi-allsky service failed

pnorrised, 2022-12-22:
Apologies for what I'm sure is a really simple question.  What log files etc can we look at if indi-allsky is not functioning and we want to work out why?

I just did an install on a pi 4 (4GB) with Raspberry Pi HQ camera.  All seemed to go well, and I can log into the web interface.  However, no image is displayed.  On looking at the "Services" tab under "System" tab, I noticed that indiserver is active, but the indi-allsky service shows as "failed".  I've tried restarting the service, both via the web interface and via ssh but to no avail.

I'd welcome any pointers.

aaronwmorris, 2022-12-22:
You can look in `/var/log/indi-allsky/indi-allsky.log`.  Let me know if I can help further!

nealweston1, 2023-01-01:
Hi, I seem to have the same issue, the indi-allsky service will not run. I've installed on a Pi 4 (4GB), PiOS Bullseye 64 bit using a ZWO ASI385MC.

I've had a look a the log file but the errors mean little to me. Can someone help? Thanks.
[Allsky errors.txt](https://github.com/aaronwmorris/indi-allsky/files/10329031/Allsky.errors.txt)

aaronwmorris, 2023-01-01:
In your case @nealweston1 I think you just selected the wrong pyindi-client version.

Run `./misc/rebuild_pyindi-client.sh` and select indi 1.9.7.  I think that will fix your issue.

nealweston1, 2023-01-01:
That was it! Working perfectly now.
Thanks very much for your help.

pnorrised, 2023-01-04:
Apologies for not replying sooner, I put everything to one side over the holiday.  Many thanks for your reply.  Looks like it could have been a hardware issue, since the camera was not recognised, so I could not even use the normal pi camera application.  I switched to a different pi and all seems fine now.  Thanks again.

aaronwmorris, 2023-01-04:
Great!  Let me know if I can help further.

aaronwmorris, 2023-02-17:
Closing this issue for now.

closed 2023-02-17 by aaronwmorris