aaronwmorris/indi-allsky #482: Enhancement:  have a log viewer page

sifank, 2022-12-23:
I can see where it would be handy to have a page that allows one to scroll back to, oh say, the last couple hundred lines of the log file, but normally would show the latest, ala 'tail -f' type of function.  This would help me support the people in the field, as they would not be connected to the internet and would be phoning in from the field.  They would be connecting with a tablet or phone to the allskysqm's hotspot (web) only.

aaronwmorris, 2022-12-23:
Merged #486 with a log viewer

dbainbridge, 2023-01-03:
The log viewer is nice!  Very convenient checking issues.

aaronwmorris, 2023-01-03:
Fantastic!  I am glad it helps!

sifank, 2023-01-03:
It's a great enhancement.  Thanks!!

Sifan Kahale
Carpe Nocturnum ...

closed 2023-01-22 by sifank