aaronwmorris/indi-allsky #484: Getting BUSY exposure status and last ready keeps increasing

sifank, 2022-12-23:
After boot, it takes one image then goes into a sort of spin:
Dec 22 17:47:41 allskysqm [INFO] MainProcess allsky.run() #822: Camera last ready: 617.3s
Dec 22 17:47:41 allskysqm [INFO] MainProcess allsky.run() #823: Exposure state: BUSY
and this continues with the last ready getting larger and larger

aaronwmorris, 2022-12-23:
What kind of camera?  Can you show the log around the time the last ready was between 0 and 60 seconds?

sifank, 2022-12-23:
The camera is the ZWO ASI294mc pro.  Here's a full cycle:
Dec 22 22:50:33 allskysqm [INFO] MainProcess allsky.run() #823: Exposure
state: ALERT
Dec 22 22:50:33 allskysqm [INFO] MainProcess allsky.detectNight() #1297:
Sun altitude: -62:30:40.4
Dec 22 22:50:33 allskysqm [INFO] MainProcess allsky.detectMoonMode() #1314:
Moon altitide: -67:07:51.0
, phase 0.1%
Dec 22 22:50:33 allskysqm [INFO] MainProcess indi.getCcdTemperature() #778:
Sensor temperature: 18.6
Dec 22 22:50:33 allskysqm [INFO] MainProcess indi.getTelescopeRaDec() #763:
Telescope Coord: RA 4.69,
Dec 44.87
Dec 22 22:50:33 allskysqm [WARNING] MainProcess indi.getGpsPosition() #699:
GPS fix not found
Dec 22 22:50:33 allskysqm [INFO] MainProcess allsky._queueManualTasks()
#2066: Checking for manually s
ubmitted tasks
Dec 22 22:50:33 allskysqm [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
Dec 22 22:50:37 allskysqm [INFO] message repeated 2 times: [ MainProcess
indi.newText() #277: new Text
GPS_STATUS for device GPSD]
Dec 22 22:50:39 allskysqm [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
Dec 22 22:50:43 allskysqm [INFO] message repeated 2 times: [ MainProcess
indi.newText() #277: new Text GPS_STATU
S for device GPSD]
Dec 22 22:50:44 allskysqm [INFO] MainProcess allsky.run() #822: Camera last
ready: 559.2s
Dec 22 22:50:44 allskysqm [INFO] MainProcess allsky.run() #823: Exposure
state: ALERT
Dec 22 22:50:44 allskysqm [INFO] MainProcess allsky.detectNight() #1297:
Sun altitude: -62:32:06.7
Dec 22 22:50:44 allskysqm [INFO] MainProcess allsky.detectMoonMode() #1314:
Moon altitide: -67:09:14.1, phase 0.
1%
Dec 22 22:50:44 allskysqm [INFO] MainProcess indi.getCcdTemperature() #778:
Sensor temperature: 18.6
Dec 22 22:50:44 allskysqm [INFO] MainProcess indi.getTelescopeRaDec() #763:
Telescope Coord: RA 4.70, Dec 44.87
Dec 22 22:50:45 allskysqm [WARNING] MainProcess indi.getGpsPosition() #699:
GPS fix not found
Dec 22 22:50:45 allskysqm [INFO] MainProcess allsky._queueManualTasks()
#2066: Checking for manually submitted t
asks
Dec 22 22:50:45 allskysqm [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
Dec 22 22:50:53 allskysqm [INFO] message repeated 4 times: [ MainProcess
indi.newText() #277: new Text GPS_STATU
S for device GPSD]
Dec 22 22:50:55 allskysqm [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
Dec 22 22:50:56 allskysqm [INFO] MainProcess allsky.run() #822: Camera last
ready: 570.5s
Dec 22 22:50:56 allskysqm [INFO] MainProcess allsky.run() #823: Exposure
state: ALERT

Sifan
Carpe Nocturnum ...

sifank, 2022-12-23:
Hold on - I stopped the indiserver and indi-allsky, then restarted them and
now I'm getting:
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#33: An uncaught exception occurred:
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#34: Type: <class 'indi_allsky.exceptions.TimeOutE
xception'>
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#35: Value: Timeout finding control CCD_TEMP_RAMP
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/all
sky.py", line 128, in <module>\n    action_func(*args_list,
**kwargs_dict)\n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/allsky.py", line 804, in run\n    self._initialize()\n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/allsky.py", line 521, in _initialize\n
   self.indiclient.configureCcdDevice(self.config[\'INDI_CONFIG_DEFAULTS\'])\
n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/camera/indi.py", line 639, in configureCcdDevice\n
   self.configureDevice(self._ccd_device, *args, **kwargs)\n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/camera/indi.py", line 626, in configureDevice\n
   self.set_number(device, k, v)\n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/camera/indi.py", line 1089, in set_number\n    c =
self.get_control(device, name, \'number\')\n'
Dec 22 22:57:27 allskysqm [ERROR] MainProcess allsky.unhandled_exception()
#40: '  File "/home/sifan/Projects/indi-allsky/ind
i_allsky/camera/indi.py", line 1080, in get_control\n    raise
TimeOutException(\'Timeout finding control {0}\'.format(name))
\n'

I'm going to shut it down, reattach the cables and try again.

Sifan
Carpe Nocturnum ...

sifank, 2022-12-23:
That seems to have done it.  Shutting down, re-attaching the cables, then
the restart worked ok.  I wonder if the indiserver somehow got confused ...

I'm continuing to test to see if this happens again.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-23:
This was definitely a camera issue as the exposure control was in `Alert` state.  I will do some investigation about what can be done.

sifank, 2022-12-23:
Thx.  I think the larger issue was that this was only in the log.  It
would be nice if something was shown on the web pages.

The first case, it was reporting Busy, but for every attempt, and the
'last successful image' time kept increasing.  Maybe you could key off
of a combination of those to trigger an error.  The second case is
more straightforward, the Alert message.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2022-12-23:
On the main page, there is already a warning when the last image is more than 15 minutes old.  As of #489 I have reduced it to 5 minutes.

sifank, 2022-12-24:
Thx!

Sifan
Carpe Nocturnum ...

closed 2022-12-28 by sifank