aaronwmorris/indi-allsky #499: Error notifications (especially out of disk space)

dbainbridge, 2022-12-30:
It would be helpful if there was some type of banner/notification on the web page when the system ran out of disk space.  Right now the live image says something about it being out of date.  You then need to check the System Info page to see that it ran out.  Along those lines a configurable low disk space warning on the web page would be helpful so you might be able to do something before it runs out of disk space.

aaronwmorris, 2023-01-01:
Merged #503 with a new notification system that issues alerts for possible problems.

aaronwmorris, 2023-01-02:
One of my cameras stopped taking images this morning and I got a notification when I logged in.  This seems to be working pretty well.

vastmeridian, 2023-01-02:
Quick question, Aaron - does this notification pick up [ERROR] entries from the indi-allsky.log?

aaronwmorris, 2023-01-02:
Not from the log, but the notification entries are added in some of the same places where the ERROR logs are generated.  I added notifications related to shutdown, disk usage, swap usage, file transfer, video failures, etc.  I am sure I will add more notifications that can be helpful.

aaronwmorris, 2023-02-17:
Closing this issue for now.

closed 2023-02-17 by aaronwmorris