aaronwmorris/indi-allsky #500: Can not download Timelapses from browser

dbainbridge, 2022-12-30:
If you attempt to download any timelapse video there is a web authentication view from the browser that appears asking for username/password.  Even clicking cancel  brings the view up two more times.   I assume this is because of the nginx config preventing you from directly downloading the mp4 video:

[image]

aaronwmorris, 2022-12-30:
That is odd.  I have never implemented anything to prevent downloads.  There is authentication to view the web interface, but once you authenticate the first time, it should not prompt again.

You mentioned nginx, are you using Astroberry?

dbainbridge, 2022-12-30:
I'm not using astroberry.. Just an Ubuntu image for Odroid. I just tried a private/incognito window and had to sign in with the admin/secret, went to a timelapse video and control-clicked to try to download it and it prompted for a username/password as shown above.

Tried admin/secret and that does not work.  After clicking cancel 3 times it downloads the filename but it contains this:
```
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>401 Unauthorized</title>
</head><body>
<h1>Unauthorized</h1>
<p>This server could not verify that you
are authorized to access the document
requested.  Either you supplied the wrong
credentials (e.g., bad password), or your
browser doesn't understand how to supply
the credentials required.</p>
<hr>
<address>Apache/2.4.52 (Ubuntu) Server at 192.168.0.123 Port 443</address>
</body></html>

```

aaronwmorris, 2022-12-30:
You may try re-running setup.sh to refresh the apache configuration and see if that resolves it.

dbainbridge, 2022-12-30:
I am using https://github.com/0xERR0R/blocky which I wonder if is affecting this.  Will have to test later.

aaronwmorris, 2022-12-30:
I suppose it is possible, but I do not think it is likely that this would be caused by a DNS issue.

The authentication configuration is pretty basic, just a simple .htaccess type config.  The indi-allsky web application is not even aware of the authentication, the web server performs that function.

dbainbridge, 2022-12-31:
It appears that it is Safari that has the issue.. as though something is not working right with the cookies with Indi-allsky.  Manually downloading from Firefox works fine.

aaronwmorris, 2023-01-01:
What version of Safari and MacOS are you running?  I noticed the test was with Safari and I tested on one of my systems and it worked fine.

I have a theory, but I want to check everything.

dbainbridge, 2023-01-01:
Safari 16.2/macOS 13.1

dbainbridge, 2023-01-01:
What's interesting is Firefox does have this warning in console when viewing the Timelapse page: 

`Cookie “session” does not have a proper “SameSite” attribute value. Soon, cookies without the “SameSite” attribute or with an invalid value will be treated as “Lax”. This means that the cookie will no longer be sent in third-party contexts. If your application depends on this cookie being available in such contexts, please add the “SameSite=None“ attribute to it. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite`

Adding `"SESSION_COOKIE_SAMESITE" : "None",` to flask.json_template gets rid of the warning but still have issues with Safari not downloading the time lapse video.

aaronwmorris, 2023-01-01:
That was exactly what my theory revolves around.  In the #503 merge, I included a change that sets the samesite setting to "Lax".

I am not exactly sure of the correct setting, so I am flexible on that.

Let me know if it helps.

dbainbridge, 2023-01-03:
Unfortunately this didn't fix it.  Got the latest changes (see the new log page) and also notice Firefox doesn't show a warning but Safari is still broken trying to download any time lapse.

aaronwmorris, 2023-01-04:
I have tested multiple Macs running Ventura and the latest Safari.  I do get re-prompted for credentials, but if I use the credentials the download works.

aaronwmorris, 2023-01-27:
Web server realm based authentication has been removed from indi-allsky as of #519.  Hopefully, this will bypass the issue with file downloads.

aaronwmorris, 2023-03-10:
Closing this issue for now.  Let me know if this pops up again.

closed 2023-03-10 by aaronwmorris