aaronwmorris/indi-allsky #507: Error running setup.sh

dbainbridge, 2023-01-02:
For some reason started getting this error after updating to latest master and  running setup.sh:
```
.
.
.
File "/home/allsky/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/pip/_internal/locations/_distutils.py", line 20, in <module>
    from distutils.cmd import Command as DistutilsCommand
ModuleNotFoundError: No module named 'distutils.cmd'
```

aaronwmorris, 2023-01-02:
Its possible the virtualenv is messed up.

You may want to see if any system OS packages are half installed.  Use this to finish any half installed packages.
```
apt-get -f install
```

If the virtualenv is messed up, you can always regenerate it.  Just remove the folder and re-run setup.sh to regenerate things.

```
deactivate
rm -fR virtualenv/indi-allsky

./setup.sh
```

dbainbridge, 2023-01-03:
I ended up having to remove the entire virtualenv directory.. not sure if this is related but service won't start up trying to read the config:

```
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #439: Connecting to indiserver
Jan  2 20:23:20 localhost [INFO] MainProcess indi.__init__() #142: PyIndi version: 1.9.8
Jan  2 20:23:20 localhost [INFO] MainProcess indi.__init__() #135: creating an instance of IndiClient
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #418: System uptime: 699197s
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #415: System memory: 3712 MB
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #410: System CPUs: 6
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #408: Platform: aarch64
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #407: Python version: 3.10.6
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #405: indi-allsky config version: 20221230.0
Jan  2 20:23:20 localhost [INFO] MainProcess allsky._initialize() #404: indi-allsky release: 6.0
Jan  2 20:21:44 localhost [ERROR] message repeated 3 times: [ MainProcess views.check_config() #133: Unable to get CONFIG_MD5]
Jan  2 13:51:12 localhost [ERROR] MainProcess views.check_config() #133: Unable to get CONFIG_MD5
```

aaronwmorris, 2023-01-03:
I do not think there is a problem with the config.  The webapp logs are currently getting intermingled with the indi-allsky logs which is why you are seeing those errors at the end.  (Those errors will go away once it completely starts up).

You may want to try rebooting.  Also, do you recall if you selected indi 1.9.8 when you ran the setup.sh?

dbainbridge, 2023-01-03:
It's defaulting to 1.9.9 (which I used), you are saying we need to use 1.9.8 instead?

aaronwmorris, 2023-01-03:
Yes, I do not have a good way of detecting the version that is installed.  You can rebuild the module with the following command:

```
./misc/rebuild_pyindi-client.sh
```

dbainbridge, 2023-01-03:
I did try just now running setup again and selecting 1.9.8 and status is still DOWN after trying to start everything back up.

aaronwmorris, 2023-01-03:
If you did not reboot, please try that.  If you are unable to, at least try restarting the indiserver.

```
systemctl --user stop indi-allsky
systemctl --user restart indiserver
systemctl --user start indi-allsky
```

dbainbridge, 2023-01-03:
I tried rebooting and even re-running setup again since it is still saying DOWN.. I had done the `./misc/rebuild_pyindi-client.sh` and selected 1.9.9 in the setup but it seems like the service is still trying to use 1.9.8? 

```
Jan  2 21:24:03 localhost [INFO] MainProcess indi.__init__() #135: creating an instance of IndiClient
Jan  2 21:24:03 localhost [INFO] MainProcess indi.__init__() #142: PyIndi version: 1.9.8
Jan  2 21:24:03 localhost [INFO] MainProcess allsky._initialize() #438: Connecting to indiserver
Jan  2 21:24:03 localhost [ERROR] MainProcess allsky._initialize() #443: No indiserver available at localhost:7624
Jan  2 21:24:03 localhost [WARNING] MainProcess miscDb.addNotification() #466: Not adding existing notification

```

aaronwmorris, 2023-01-03:
The "PyIndi version" is actually the version of indilib that is installed.  Unfortunately, there are compatibility problems with different versions (git commits) of pyindi-client, so you have to select the correct version in order for it to start correctly.

I do not recall, which camera are you using?

dbainbridge, 2023-01-03:
ASI178MC

dbainbridge, 2023-01-03:
I can't seem to get the service back running.  I've rebooted.. Also reran setup and made sure to select 1.9.8 but still stays stuck at DOWN with basically showing the log output as above.

aaronwmorris, 2023-01-03:
Hmm.  Maybe try upgrading indi to the latest version.  (It will be necessary to rebuild pyindi-client).

```
./misc/build_indi_noansible.sh
```

If you do this, I would suggest fully rebooting after you are finished.

dbainbridge, 2023-01-03:
That fixed it!  So what I did was:
`./misc/build_indi_noansible.sh` (took 1104s)
`./misc/rebuild_pyindi-client.sh` (took 84s)
then rebooted, and it was working.

aaronwmorris, 2023-01-03:
Great!  That is a really good compile time for an ARM SoC!  It normally takes about 45 minutes on a Rpi4.

closed 2023-01-10 by dbainbridge