aaronwmorris/indi-allsky #511: Documenting Config differences

dbainbridge, 2023-01-03:
Do you have any thoughts on documenting/showing what the different config settings do?  I am wondering if it would make sense to have some type of clickable link after each setting/section in the config that goes to a part of the GitHub wiki showing/explaining what it does.  i.e. showing what "Apply CLAHE contrast enhancement" does:

No daytime contrast enhancement:
[image]

daytime contrast enhancement on: (note the file size nearly doubled)
[image]

dbainbridge, 2023-01-03:
That would require internet connection though..

aaronwmorris, 2023-01-05:
Time permitting, I would love to have definitions for each config setting.

The CLAHE algorithm is pretty cool, especially with daytime clouds (as you can see).  Being perfectly honest, I only know what I have read about CLAHE.  OpenCV has a CLAHE method baked in, so I just implemented that.

The (code)[https://github.com/aaronwmorris/indi-allsky/blob/master/indi_allsky/image.py#L1971] is fairly simple.  What it is really doing, I have no idea.  :-)

closed 2025-08-24 by aaronwmorris