aaronwmorris/indi-allsky #512: indi-allsky service fails at daybreak

sifank, 2023-01-04:
Just after daybreak here (7:40am on the west coast) both the indi-server and the indi-allsky say 'failed' on the Service page.  It happens at other times as well, but the morning daybreak is consistent.  I've rebooted, I've installed a new Ubuntu OS and installed PR 510, still the same.  File system is about 80% full, even tho the log says it went through a delete cycle, reporting 0 files removed.  I've attached the log and the config.json.
[indi-allsky.log](https://github.com/aaronwmorris/i
[config.json.txt](https://github.com/aaronwmorris/indi-allsky/files/10347180/config.json.txt)
ndi-allsky/files/10347171/indi-allsky.log)

sifank, 2023-01-04:
It starts back up fine and runs for a number of hours, then, same thing happens.

aaronwmorris, 2023-01-04:
The log file did not come through correctly.

sifank, 2023-01-04:
[indi-allsky.log](https://github.com/aaronwmorris/indi-allsky/files/10347493/indi-allsky.log)
Is there a log for the indiserver?  Perhaps that's what failed.

sifank, 2023-01-06:
> There is probably very little logged by the indiserver. You can see if it
> logged anything with systemctl --user status indiserver
>
> The log looks perfectly normal. If the indi-allsky process is crashing, it
> could be a segfault, which would not appear in the log. You could try
> running indi-allsky from the console and see if it eventually crashes.
>
> I would suggest power cycling the camera (unplug the cables and reconnect)
> then reboot and see if that helps.

The indi-allsky service stops at sunset and just after sunrise.  Both this
morning and this evening, the indiserver was still running, just the allsky
service stopped.  Here's the result of a systemctl on the allsky service
once it failed:
systemctl --user status indi-allsky.service
× indi-allsky.service - Indi AllSky Service
    Loaded: loaded (/home/sqm/.config/systemd/user/indi-allsky.service;
enabled; vendor preset: enabled)
    Active: failed (Result: signal) since Thu 2023-01-05 17:29:05 PST; 2h
23min ago
   Process: 3352024
ExecStart=/home/sqm/indi-allsky/virtualenv/indi-allsky/bin/python3
allsky.py --config /etc/indi-allsky/conf>
  Main PID: 3352024 (code=killed, signal=KILL)
       CPU: 41min 13.669s

Jan 05 17:27:24 allskysqm python3[3352024]: [INFO] MainProcess
allsky.detectNight() #1458: Sun altitude: -6:38:34.0
Jan 05 17:27:24 allskysqm python3[3352024]: [INFO] MainProcess
allsky.detectMoonMode() #1475: Moon altitide: 16:05:40.2, phase 9>
Jan 05 17:27:24 allskysqm python3[3352024]: [INFO] MainProcess
allsky.detectMoonMode() #1479: Moon Mode conditions detected
Jan 05 17:27:25 allskysqm python3[3352024]: [INFO] MainProcess
indi.getCcdTemperature() #778: Sensor temperature: 0.8
Jan 05 17:27:25 allskysqm python3[3352024]: [INFO] MainProcess
indi.getTelescopeRaDec() #763: Telescope Coord: RA 0.21, Dec 44.87
Jan 05 17:27:25 allskysqm python3[3352024]: [INFO] MainProcess
indi.getGpsPosition() #706: GPS location: lat 44.87, long -124.04>
Jan 05 17:27:25 allskysqm python3[3352024]: [INFO] MainProcess
allsky._queueManualTasks() #2251: Checking for manually submitted>
Jan 05 17:28:45 allskysqm systemd[1047]: indi-allsky.service: Main process
exited, code=killed, status=9/KILL
Jan 05 17:29:05 allskysqm systemd[1047]: indi-allsky.service: Failed with
result 'signal'.
Jan 05 17:29:05 allskysqm systemd[1047]: indi-allsky.service: Consumed
41min 13.669s CPU time.

Tonight I'll run it manually and see what happens at sunrise.

Sifan
Carpe Nocturnum ...

sifank, 2023-01-06:
When starting allsky manually, the first entries were Expiring orphaned
tasks:
./allsky.py run
2023-01-05 20:04:37,820 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 32
2023-01-05 20:04:37,821 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 33
2023-01-05 20:04:37,821 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 34
2023-01-05 20:04:37,821 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 35
2023-01-05 20:04:37,821 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 36
2023-01-05 20:04:37,822 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 37
2023-01-05 20:04:37,822 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 38
2023-01-05 20:04:37,822 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 39
2023-01-05 20:04:37,822 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 40
2023-01-05 20:04:37,822 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 41
2023-01-05 20:04:37,823 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 42
2023-01-05 20:04:37,823 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 43
2023-01-05 20:04:37,823 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 44
2023-01-05 20:04:37,823 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 45
2023-01-05 20:04:37,823 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 46
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 47
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 48
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 49
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 50
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 51
2023-01-05 20:04:37,824 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 52
2023-01-05 20:04:37,825 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 53
2023-01-05 20:04:37,825 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 29
2023-01-05 20:04:37,825 [WARNING] MainProcess allsky._expireOrphanedTasks()
#2233: Expiring orphaned task 31
2023-01-05 20:04:37,873 [INFO] MainProcess allsky._initialize() #403:
indi-allsky release: 6.0
2023-01-05 20:04:37,873 [INFO] MainProcess allsky._initialize() #404:
indi-allsky config version: 20221230.0
2023-01-05 20:04:37,874 [INFO] MainProcess allsky._initialize() #406:
Python version: 3.10.6
2023-01-05 20:04:37,874 [INFO] MainProcess allsky._initialize() #407:
Platform: aarch64
2023-01-05 20:04:37,874 [INFO] MainProcess allsky._initialize() #409:
System CPUs: 4
2023-01-05 20:04:37,875 [INFO] MainProcess allsky._initialize() #414:
System memory: 3789 MB
2023-01-05 20:04:37,876 [INFO] MainProcess allsky._initialize() #417:
System uptime: 247619s
2023-01-05 20:04:37,877 [INFO] MainProcess indi.__init__() #135: creating
an instance of IndiClient
2023-01-05 20:04:37,877 [INFO] MainProcess indi.__init__() #142: PyIndi
version: 1.9.9
2023-01-05 20:04:37,877 [INFO] MainProcess allsky._initialize() #438:
Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
2023-01-05 20:04:37,882 [INFO] MainProcess indi.serverConnected() #286:
Server connected (localhost:7624)
2023-01-05 20:04:37,884 [INFO] MainProcess indi.newDevice() #198: new
device GPSD
2023-01-05 20:04:37,885 [INFO] MainProcess indi.newDevice() #198: new
device GPSD
Dispatch command error(-1): INDI: Could not find property GEOGRAPHIC_COORD
in GPSD
<setNumberVector device="GPSD" name="GEOGRAPHIC_COORD" state="Idle"
timeout="60" timestamp="2023-01-06T04:04:37">
   <oneNumber name="LAT">
44.867798333000003197
   </oneNumber>
   <oneNumber name="LONG">
235.9597400000000107
   </oneNumber>
And also got this GPSD error (above).  However, after that, it is working
ok.

Not sure if this helps but it's some additional information.  Also
unplugging/plugging the camera didn't seem to have any effect.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-01-06:
According to those log messages, the process received a KILL (9) signal.

Can you check your system log to see if there are any out of memory (OOM) type messages?

`sudo dmesg`

sifank, 2023-01-06:
Just to follow up on the GPS: running cgps shows a 3D fix and the
indi-allsky logs has entries from the indiserver:
2023-01-05 20:09:37,885 [INFO] MainProcess indi.newMessage() #283: new
Message 2023-01-06T04:09:37: [INFO] GPS fix obtained.
2023-01-05 20:09:37,886 [INFO] MainProcess indi.newMessage() #283: new
Message 2023-01-06T04:09:37: [INFO] GPS fix obtained.
2023-01-05 20:09:37,886 [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
2023-01-05 20:09:37,886 [INFO] MainProcess indi.newText() #277: new Text
GPS_STATUS for device GPSD
2023-01-05 20:09:37,886 [INFO] MainProcess indi.newSwitch() #270: new
Switch GPS_REFRESH for device GPSD
2023-01-05 20:09:37,887 [INFO] MainProcess indi.newText() #277: new Text
TIME_UTC for device GPSD
2023-01-05 20:09:37,888 [INFO] MainProcess indi.newMessage() #283: new
Message 2023-01-06T04:09:37: [INFO] Observer location upda
ted: Latitude 44:52:04.6 (44.87) Longitude -124:02:25.5 (-124.04)

Sifan
Carpe Nocturnum ...

sifank, 2023-01-06:
Ah, yes indeed:
Jan  2 22:00:20 allskysqm systemd[1641]: dbus.service: Failed with
result 'signal'.
Jan  2 22:00:20 allskysqm CRON[7383]: (root) CMD (/opt/KOBS/dynSysInfo)
Jan  2 22:00:20 allskysqm systemd[1641]: xdg-permission-store.service:
Main process exited, code=exited, status=1/FAILURE
Jan  2 22:00:20 allskysqm systemd[1641]: xdg-permission-store.service:
Failed with result 'exit-code'.

This is a RPI4 with 4GB RAM and 1gb swap.

Sifan
Carpe Nocturnum ...

sifank, 2023-01-06:
Bad example, here's OOM on the allsky proc (grep of syslog):
Jan  2 20:35:58 allskysqm systemd[1]: Started Userspace Out-Of-Memory
(OOM) Killer.
Jan  2 23:10:17 allskysqm systemd[1]: Starting Userspace Out-Of-Memory
(OOM) Killer...
Jan  2 23:10:17 allskysqm systemd[1]: Started Userspace Out-Of-Memory
(OOM) Killer.
Jan  2 23:17:23 allskysqm systemd[1]: Starting Userspace Out-Of-Memory
(OOM) Killer...
Jan  2 23:17:23 allskysqm systemd[1]: Started Userspace Out-Of-Memory
(OOM) Killer.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-01-06:
Check the kernel log using `sudo dmesg` and see what is being killed.

I think I remember you mentioned using a ASI294MC Pro.   I assume you are running in the normal mode which has a 4144x2822 resolution.  You might be hitting the OOM condition when ffmpeg is generating the video.  Even 4GB of memory can be a limit when generating a video close to 4K resolution.

You can try setting the FFMPEG scaling in the config to 75% (or lower) and see if that resolves the OOM condition.

sifank, 2023-01-06:
Yup, it's ffmpeg:
[22729.595318]
oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=user.slice,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1002.slice/
***@***.***
rvice/app.slice/indi-allsky.service,task=ffmpeg,pid=21701,uid=1002
[22729.595476] Out of memory: Killed process 21701 (ffmpeg)
total-vm:4184572kB, anon-rss:2755428kB, file-rss:1008kB, shmem-rss:0kB,
UID:1002 pgtables:6228kB oom_s
core_adj:0
And this was on a system that I had just rebuilt - using Ubuntu server
instead of the desktop version, in hopes of reducing the system's
footprint.  And this is happening just about when it would be generating
the video.

I reset the image scaling for FFMPEG to 50 and Image scale for timelapse to
70 and will see what happens in the morning.

Thanks!

Sifan
Carpe Nocturnum ...

sifank, 2023-01-07:
Resetting the ffmpeg scaling did the trick.  It survived this morning!

Thanks again for all your help Aaron!!

Sifan
Carpe Nocturnum ...

closed 2023-01-22 by sifank