aaronwmorris/indi-allsky #514: Remote web interface

dbainbridge, 2023-01-04:
Is it possible to have a second web interface and dataset that is remote from the main system that is controlling the camera?  Similar to what TJ's allsky had with a separate web interface but that is very cumbersome when you have a lot of days.

What I want is the system connected to the camera to work as it is now but also have a remotely hosted web interface that can view all the time lapses, keograms, etc that were uploaded to it (and also has a lot more storage then what the system controlling the camera has).

aaronwmorris, 2023-01-05:
This is something that I have given a lot of thought to, but I have not had a eureka moment.

My current high level plans were to harden the existing web interface to allow it to be self hosted on your own internet connection.  Faster internet connections able to host multimedia files are becoming more common.   This is probably a controversial idea... even my own current internet connection is not quite up to that standard.

I would really like to do a public interface using Python Flask (or Django) like the current interface.  This would allow a lot of code reuse.  However, a lot of hosting environments do not provide Python native app server support.  It is better than it was a couple of years ago.  I am seeing more environments allowing Python apps using Passenger (a la Ruby on Rails).  PHP has very broad support, but I am not a big PHP fan.

dbainbridge, 2023-01-05:
I've been using a raw VM with ubuntu image to run TJ's web UI and a custom one before that so you could use the current Python Flask UI if going that route since I agree that it would be preferable to have one common UI, both visually and code wise.

If I were to try that is it currently possible to install indi-allsky without any of the driver/indi related components since this installation wouldn't be talking to a camera directly?  The interface would use the data that is uploaded from the system running indi-allsky connected to the camera.  If it is simpler at first  I would just upload every "raw" image and the hosted indi-allsky system process the images as though they came from a camera.  I realize there would need to be some changes to support this minimum functionality but wondered if this sounds reasonable for me to try?

aaronwmorris, 2023-01-05:
When I was writing my reply, I was considering something like that.

I was considering was just adding the ability to have an upload folder.  Image files, timelapses, startrails, etc all have a distinct naming convention.  I could just have code look at the upload folder, pattern match the file name, and import the files to the DB based on the name alone.

It could work...

aaronwmorris, 2023-01-05:
I had considered a RESTful interface.  I just worry about how complex that might become.  But a REST interface could be helpful in other ways.

dbainbridge, 2023-01-05:
An endpoint only for PUTing/POSTing images and then going through the same workflow as if it was taken from the camera would be useful.  You'd then have ability to basically have a local and remote instance running with same camera data.. can eventually expand that later but that would at least make is feasible to easily have a remotely hosted indi-allsky fully working that isn't directly connected to a camera.

dbainbridge, 2023-01-05:
Thinking through this further, in the future you could potentially have multiple cameras reporting to that same remote instance..

aaronwmorris, 2023-01-05:
It could be possible.  Supporting multiple cameras on a single indi-allsky install was actually one of the original design goals I had, but I abandoned the idea after deciding trying to handle the timing of multiple cameras was going to significantly increase complexity.

All of the media assets are linked to the camera table, so the basis for this is already in place.

Vertikar, 2023-01-06:
> I would really like to do a public interface using Python Flask (or Django) like the current interface. This would allow a lot of code reuse. However, a lot of hosting environments do not provide [...]

Could containerise it to partly avoid that issue? I see AWS does Lightsail containers now

https://aws.amazon.com/blogs/aws/lightsail-containers-an-easy-way-to-run-your-containers-in-the-cloud/

dbainbridge, 2023-01-06:
I don't know if Aaron is looking into this at all but I started looking at it and I don't view the problem of where to run it as a big issue.  It doesn't even have to be remote, it could simply be a beefier computer you have that isn't directly connected to the camera that does all the processing so that the device connected to the camera itself can be less specced out/cheaper.  There is also much simpler services compared to AWS like Linode that could eventually be made very easy to deploy the web ui/processing portion of indi-allsky to.  The code is written to expect a camera and an Indi-server running locally so I was looking at what it would take to make it more modularized.

aaronwmorris, 2023-01-07:
indi-allsky can already run with the camera and indiserver running on a remote server.  The indi-allsky is technically an indi client, you can specify a remote indiserver in the config.  Of course, this only works with cameras connected via indi, libcamera does not work remotely.

In any case, I am thinking about how to support a remote indi-allsky web site that could just host the media files.

aaronwmorris, 2023-01-08:
I am going to make this my next major sprint.  This is going to be a multi-part enhancement:

1.  Implement real identity and access management support.  Adding a User [database] model to authenticate users that require privileges.  Viewing images and movies should remain anonymous, but configuration will be placed behind real authentication managed by the application.  This will include all of the necessary security considerations necessary for a modern Internet facing application.
2. Add a REST endpoint for file uploads.  Authentication for the endpoint will be through an API key for the users in the User model.  The validation of the API key will probably include some time-based component to prevent credential reuse.

I will update my Security information wiki as I go with details:
https://github.com/aaronwmorris/indi-allsky/wiki/Security-considerations

aaronwmorris, 2023-01-11:
Merged #519 with the new internal user management.  Authentication is no longer handled at the web server level.