aaronwmorris/indi-allsky #520: MQTT monitoring: A couple of additional topics to publish.

vastmeridian, 2023-01-12:
Hi Aaron

I've successfully managed to get MQTT running and I can incorporate the topics already documented into Home Assistant and also to my 'phone using MQTT-Dash.  Thank you.
Given the debugging I have had to do, is it worth adding a couple of further topics to the list such as available disk space, memory and CPU usage along the lines of

<html><body>

Topic | Type | Data
-- | -- | --
indi-allsky/disk | float | disk space used %
indi-allsky/memory | float | memory utilisation %
indi-allsky/cpu | float | cpu utilisation %

</body>
</html>

I've also got a question about the units you've used for SQM.  At the moment we just get a large number that varies.  What's the intended scale for this?  I appreciate it's hard for you to produce a general method to get a direct figure, but the definition of SQM I am more familiar with is the one discussed on Wikipedia, which is:

> Scientists, in the process of creating the SQM, devised a scale: between the numbers of 16.00-22.00. At the lowest number—16.00—the sky is the brightest. Customarily, this class would transpire in [urbanized areas](https://en.wikipedia.org/wiki/Urban_area). Meanwhile, a number of 22.00 represents the least [luminance](https://en.wikipedia.org/wiki/Luminance)—in other words, the least [light pollution](https://en.wikipedia.org/wiki/Light_pollution). Typically, this reading would materialize in [rural areas](https://en.wikipedia.org/wiki/Rural_area).

https://en.wikipedia.org/wiki/Sky_quality_meter

Is there a correlation between your SQM data and this scale at all?

vastmeridian, 2023-01-12:
Regarding SQM - or jSQM that you refer to, I've just seen your post on Cloudy Nights which explains the basis of your number.

https://www.cloudynights.com/topic/788720-all-sky-camera-as-a-sqm/

aaronwmorris, 2023-01-12:
Unless the exposure and gain are at the max (moonless night), the SQM number is basically meaningless.  Currently, the SQM value is really just a placeholder until I can get something better in there.

I'll work on getting the system stats added to the MQTT data.

aaronwmorris, 2023-01-26:
Merged #535 for the new mqtt topics.  I updated the wiki page with the new topics.

https://github.com/aaronwmorris/indi-allsky/wiki/MQTT-Broker-Publishing

vastmeridian, 2023-02-13:
Thanks Aaron.  Good selection of additional values.  I've tested most of them on MQTT Dashboard and they work fine.  I'll have a go at integrating into Home Assistant and see how they look.

Any chance of CPU Temp being recorded too?  This might be useful during the summer months if the sun is on the system on hot days.

aaronwmorris, 2023-02-13:
I can add all of the temperatures that are reported by the Python psutil module, which are the same values under the System page.  You would have to find the topic names on the system page.

The Raspberry Pi is pretty easy as the topic is usually `cpu_thermal`, but different systems report different names.  My Rock Pi 3 system reports `soc_thermal` and `gpu_thermal`.  The Le Potato system reports `scpi_sensors`.

I am trying to make this as platform agnostic as possible.

aaronwmorris, 2023-02-13:
Merged #562 to publish temperature data to mqtt.

I had to do things a little different than I mentioned due to the way the temperature is reported.  Raspi is fairly simple with a single item on a single subsystem.  However, my Thinkpad reports 14 temperatures across 3 subsystems.

Update the wiki:  https://github.com/aaronwmorris/indi-allsky/wiki/MQTT-Broker-Publishing

vastmeridian, 2023-03-14:
Many thanks for this, Aaron.
Sorry I've been out of the loop for a while.  I've been engineering a proper case/mount for my camera.  Finally, it is ready to be mounted outside (which is happening today) so I will be able to complete testing and maybe add some comments.  I'll post the camera on the CN thread  as soon as it's formally 'live'.

vastmeridian, 2023-04-04:
Aaron.  This is working fine - and the documentation is good, too.  If you're happy, I'll close this enhancement as 'complete'