aaronwmorris/indi-allsky #529: Can't logon

jkoenig72, 2023-01-24:
Hi Arron,

really strange - I tried to update my allsky cams to your latest release.

All good - but I can not logon via the web interface.

A user has been created during install and all looks good. Website comes up (I choosed 'no' during install, so the password is only for the reserved sections.)

Strangely I can not login, I tried a few times, I changed password of that user with your ./misc/usertool.py - nothing work. Then I tried a fresh install, here I can also not logon. No error, nothing, press the button - nothing happens. 

Could there anything wrong with the code? 

Thanks!
Joerg

jkoenig72, 2023-01-24:
I found the reason for that - when I switch on to https again its working. If I commented out those 2 lines for http instead https
in /etc/apache2/sites-enabled/indi-allsky.conf the whole user managment is not working at all. It worked before. Minor thing.

### Comment this section to permit HTTP access to indi-allsky
    ###  SESSION_COOKIE_SECURE will have to be set to "false" in flash config
here->    RewriteCond "%{HTTPS}" off
and here->   RewriteRule "^/(.*)" "https://%{SERVER_NAME}:443/$1" [R,L]
    ###

aaronwmorris, 2023-01-24:
If you try to use HTTP (no SSL/TLS) you will also have to disable the secure cookie flag in `/etc/indi-allsky/flask.json`

```
"SESSION_COOKIE_SECURE" : false,
...
"REMEMBER_COOKIE_SECURE" : false,
```

aaronwmorris, 2023-02-17:
Closing this issue for now.  Let me know if I can do anything else.

closed 2023-02-17 by aaronwmorris