aaronwmorris/indi-allsky #530: Switch off logon at web interface?

jkoenig72, 2023-01-24:
Hi Aaron,

just asking - is there a way to disable logon completly?

Thanks!

aaronwmorris, 2023-01-24:
You should be able to add the following to /etc/indi-allsky/flask.json.  I have not tested this, so let me know.

```
"LOGIN_DISABLED" : true
```
Note:  may require a trailing comma depending on where you add it.

aaronwmorris, 2023-01-27:
I have added the `LOGIN_DISABLED` option (default false) in the repo for reference.

In merge #532 I also updated the processing of the flask.json config file.  Previously, the setup.sh script would overwrite the flask.json, losing any customizations.  In this change, the setup.sh script now attempts to merge the changes so hopefully, any customizations you make will be maintained.

jkoenig72, 2023-01-27:
Awesome, thanks!

aaronwmorris, 2023-03-10:
Closing this for now.

closed 2023-03-10 by aaronwmorris