aaronwmorris/indi-allsky #554: After manual time update, allsky doesn't take images

sifank, 2023-02-05:
RPI is not connected to the internet so it starts up with a date about a month ago.  Allsky starts up and is taking images but dated a month ago (of course).  When I click on the 'Update Time', the correct time (from the browser) is set, but then allsky starts reporting Exposure State: BUSY and Camera last ready: keeps climbing.

I have to restart the indiserver and then the allsky services to get everything taking pictures with the right time.

I just updated to PR553 to check if this is still happening - it is.  

I also disabled GPS and ran setup.py telling it no GPS, just to isolate this behavior.

This sequence keeps repeating itself in the log:
Feb  4 17:06:08 allskysqm [INFO] MainProcess allsky.run() #932: Camera last ready: 2345846.2s
Feb  4 17:06:08 allskysqm [INFO] MainProcess allsky.run() #933: Exposure state: BUSY
Feb  4 17:06:08 allskysqm [INFO] MainProcess allsky.detectNight() #1502: Sun altitude: 3:01:25.0
Feb  4 17:06:08 allskysqm [INFO] MainProcess allsky.detectMoonMode() #1519: Moon altitide: 5:46:13.7, phase 99.3%
Feb  4 17:06:08 allskysqm [INFO] MainProcess indi.getCcdTemperature() #849: Sensor temperature: 28.2
Feb  4 17:06:08 allskysqm [INFO] MainProcess indi.getTelescopeRaDec() #834: Telescope Coord: RA 1.82, Dec 44.87
Feb  4 17:06:08 allskysqm [INFO] MainProcess allsky._queueManualTasks() #2295: Checking for manually submitted tasks

aaronwmorris, 2023-02-05:
Remind me what camera you are using?  ASI294?

sifank, 2023-02-05:
Yes, ASI294.  But I've noticed this on an ASI120 as well.

Sifan
Carpe Nocturnum ...

aaronwmorris, 2023-02-17:
I will have to add a note in the time configuration page that the indiserver and indi-allsky may need a full start when changing the time.

If you notice above it says the exposure state is busy.  That message is because the camera server is still reporting the camera is still taking an exposure.  Depending on how the camera driver is tracking the time, suddenly changing the time can cause a race condition where things never complete.  And each camera server has different code, so the behavior would not be consistent between vendors.

Specific to indi-allsky:  Changing the time forward in indi-allsky should be safe, but setting time backwards could cause some very strange issues.

aaronwmorris, 2023-02-18:
Merged #572 with the additional warning.

sifank, 2023-02-26:
Would it be possible to restart both the indiserver and the indi-allsky when updating the time?

aaronwmorris, 2023-02-26:
It would be possible, but I am not sure how many problems that could cause.

I have considered updating the code to only set the time when the camera is not actively taking a frame, but that is a fairly complex workflow, as well.

aaronwmorris, 2023-04-04:
I tried updating the code to only update time between exposures, but it not help.  Moving forward in time works consistently, but moving backwards almost always causes the exposure control to get stuck in the Busy state.

aaronwmorris, 2023-08-24:
Closing for now.

closed 2023-08-24 by aaronwmorris