aaronwmorris/indi-allsky #555: updating program

chudson1984, 2023-02-05:
I am installing the software and it says already installed, do you want to update. I choose yes, it goes through the processes. Then I get a error:

Detected INDI version: 2.0.0
Collecting pyindi-client
  Cloning https://github.com/indilib/pyindi-client.git (to revision ce808b7) to /tmp/pip-install-jan6wtwl/pyindi-client_ec5c66b0431d4379ac2053533afcc7bb
  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-jan6wtwl/pyindi-client_ec5c66b0431d4379ac2053533afcc7bb
  WARNING: Did not find branch or tag 'ce808b7', assuming revision or ref.
  Running command git checkout -q ce808b7
  Resolved https://github.com/indilib/pyindi-client.git to commit ce808b7
  Preparing metadata (setup.py) ... done

! Bypassing indiserver setup
**** Setting up indi-allsky service ****
**** Setting up gunicorn ****
**** Enabling services ****
Failed to enable unit: Unit file indiserver.service does not exist.

The script exited abnormally, please try to run again...

aaronwmorris, 2023-02-05:
Merged #557 .

I have been messing with the build process and I found a logic error I created.

chudson1984, 2023-02-05:
Ok. That would explain I guess. I am not real savvy in Linux. Just enough to get by.

chudson1984, 2023-02-05:
I am using a ZWO camera. The 224MC to be exact. I've got the server up and running but still not seeing my camera. I would assume it is not supported? I downloaded all the indi drivers fro ZWO and even libcam

aaronwmorris, 2023-02-06:
The ASI224 will work just fine.  Did you select the `indi_asi_ccd` when running setup.sh?

chudson1984, 2023-02-06:
I did. But it stills says camera not detected when logging into the browser

chudson1984, 2023-02-06:
So, ill have to revisit my steps to remember what I did but I did get it running with terminal command. Believe it was the allsky.py run. It starts to process images but says something about a threshold and doesn't store them. I'm gonna start from scratch tomorrow and see if that resolves my issues.

Vertikar, 2023-03-22:
I think I had a similar issue when I set mine up initially, and had to manually edit one of the config files.
Was going to report it as a bug, but didn't get around to it.

I think it was within the indi config, rather than the indi-allsky side, but I can't remember exactly.

Probably slightly different now too with the config being in the database.

aaronwmorris, 2023-04-30:
Closing this issue for now.  Let me know if there is anything else I can do.

closed 2023-04-30 by aaronwmorris