aaronwmorris/indi-allsky #574: Setup routine stops - not finding flask.json

zero00zero, 2023-02-18:
Trying to get this fine product installed on my Raspberry Pi 3B+ with Debian Bullseye 64bit. 
Setup script fails with following error:

--------------
Collecting pyindi-client
  Cloning https://github.com/indilib/pyindi-client.git (to revision ffd939b) to /tmp/pip-install-w5p98wo5/pyindi-client_c67b22d719344ae4a2778814e6f76f5a
  Running command git clone --filter=blob:none --quiet https://github.com/indilib/pyindi-client.git /tmp/pip-install-w5p98wo5/pyindi-client_c67b22d719344ae4a2778814e6f76f5a
  **WARNING: Did not find branch or tag 'ffd939b', assuming revision or ref.**
  Running command git checkout -q ffd939b
  Resolved https://github.com/indilib/pyindi-client.git to commit ffd939b
  Preparing metadata (setup.py) ... done

**** Setting up indiserver service ****
**** Setting up indi-allsky service ****
**** Setting up gunicorn ****
**** Enabling services ****
**** Setup policy kit permissions ****
**** Ensure user is a member of the systemd-journal group ****
**** Setup rsyslog logging ****
**** Indi-allsky config ****
**** Setup DB ****
Traceback (most recent call last):
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/bin/flask", line 8, in <module>
    sys.exit(main())
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 1050, in main
    cli.main()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1651, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/click/core.py", line 1698, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 578, in get_command
    app = info.load_app()
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 312, in load_app
    app = locate_app(import_name, None, raise_if_not_found=False)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/cli.py", line 218, in locate_app
    __import__(module_name)
  File "/home/pi/indi-allsky/app.py", line 3, in <module>
    from indi_allsky.flask import create_app
  File "/home/pi/indi-allsky/indi_allsky/__init__.py", line 3, in <module>
    from .allsky import IndiAllSky
  File "/home/pi/indi-allsky/indi_allsky/allsky.py", line 32, in <module>
    from . import camera as camera_module
  File "/home/pi/indi-allsky/indi_allsky/camera/__init__.py", line 1, in <module>
    from .indi import IndiClient as indi
  File "/home/pi/indi-allsky/indi_allsky/camera/indi.py", line 30, in <module>
    app = create_app()
  File "/home/pi/indi-allsky/indi_allsky/flask/__init__.py", line 84, in create_app
    app.config.from_file(p_flask_config, load=json.load)
  File "/home/pi/indi-allsky/virtualenv/indi-allsky/lib/python3.9/site-packages/flask/config.py", line 264, in from_file
    with open(filename) as f:
FileNotFoundError: [Errno 2] **Unable to load configuration file (No such file or directory): '/etc/indi-allsky/flask.json'**

The script exited abnormally, please try to run again...
------------

File is indeed not existing. 
Copying the flask.json template file didn't work either. 

Thanks for looking into this!

aaronwmorris, 2023-02-18:
Merged #575 .

Sorry about that.  I just made a big change to indi-allsky and I did not test the setup process for new installs.  Hopefully this should fix it.

zero00zero, 2023-02-18:
Working flawlessly now! Thanks a ton for the quick fix.

closed 2023-02-18 by zero00zero