aaronwmorris/indi-allsky #584: Deb 11 Install Fail

bsbachert, 2023-02-24:
WARNING: Did not find branch or tag '6145b648', assuming revision or ref.
  Running command git checkout -q 6145b648
  Resolved https://salsa.debian.org/apt-team/python-apt.git to commit 6145b648
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      Traceback (most recent call last):
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/_normalization.py", line 62, in safe_version
          return str(packaging.version.Version(v))
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/_vendor/packaging/version.py", line 197, in __init__
          raise InvalidVersion(f"Invalid version: '{version}'")
      setuptools.extern.packaging.version.InvalidVersion: Invalid version: 'None'
      
      During handling of the above exception, another exception occurred:
      
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-ddjat8h3/python-apt_8154848c7d3f4dbeabfdef76ba334a61/setup.py", line 100, in <module>
          setup(name="python-apt",
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/__init__.py", line 108, in setup
          return distutils.core.setup(**attrs)
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/_distutils/core.py", line 147, in setup
          _setup_distribution = dist = klass(attrs)
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/dist.py", line 483, in __init__
          self.patch_missing_pkg_info(attrs)
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/dist.py", line 472, in patch_missing_pkg_info
          dist._version = _normalization.safe_version(str(attrs['version']))
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/_normalization.py", line 65, in safe_version
          return str(packaging.version.Version(attempt))
        File "/home/dell-pi/indi-allsky/virtualenv/ansible/lib/python3.9/site-packages/setuptools/_vendor/packaging/version.py", line 197, in __init__
          raise InvalidVersion(f"Invalid version: '{version}'")
      setuptools.extern.packaging.version.InvalidVersion: Invalid version: 'None'
      W: [python3.9] DistUtilsExtra import error.
      W: [python3.9] Sphinx import error.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
dell-pi@raspberry:~/indi-allsky $

aaronwmorris, 2023-02-24:
Use the `./misc/build_indi_noansible.sh` for now.  I will take a look at the other script tomorrow.

bsbachert, 2023-02-24:
Thank you Aaron.

bsbachert, 2023-02-25:
got thru ./misc/build_indi_noansible.sh, then had this failure in
./setup.sh.

 Using cached
Bottleneck-1.3.6-cp39-cp39-manylinux_2_5_i686.manylinux1_i686.manylinux_2_17_i686.manylinux2014_i686.whl
(358 kB)
Collecting python-dateutil
  Using cached
https://www.piwheels.org/simple/python-dateutil/python_dateutil-2.8.2-py2.py3-none-any.whl
(247 kB)
Collecting ephem
  Using cached
ephem-4.1.4-cp39-cp39-manylinux_2_12_i686.manylinux2010_i686.manylinux_2_17_i686.manylinux2014_i686.whl
(1.8 MB)
Collecting paramiko
  Using cached
https://www.piwheels.org/simple/paramiko/paramiko-3.0.0-py3-none-any.whl
(210 kB)
Collecting pycurl
  Using cached pycurl-7.45.2.tar.gz (234 kB)
  Preparing metadata (setup.py) ... done
Collecting imageio
  Using cached
https://www.piwheels.org/simple/imageio/imageio-2.25.1-py3-none-any.whl
(3.4 MB)
Collecting imageio-ffmpeg
  Using cached
https://www.piwheels.org/simple/imageio-ffmpeg/imageio_ffmpeg-0.4.8-py3-none-any.whl
(16 kB)
ERROR: Could not find a version that satisfies the requirement rawpy (from
versions: none)
ERROR: No matching distribution found for rawpy

The script exited abnormally, please try to run again...

bsbachert, 2023-02-25:
Using Raspberry PI OS for PC. Debian 11

aaronwmorris, 2023-02-25:
What architecture is reported by `uname -m`?  I just need to know if its i386, i586, or i686 so I can code this correctly.

bsbachert, 2023-02-25:
x86_64

aaronwmorris, 2023-02-25:
Okay, this must be a hybrid distribution running a 64-bit kernel but 32-bit binaries.  The wheels being loaded by python are not 64-bit.

The rawpy python module will not build on a 32-bit platform, but my script is detecting the system is 64-bit.  Just manually edit `requirements_debian11.txt` and remove (or comment) the `rawpy` line and re-run setup.sh

bsbachert, 2023-02-27:
I thing image handler is broke. Two different setups, one raspberry OS and
one Ubuntu Mate, and neither will upload to my website.

Thanks
Brian

bsbachert, 2023-02-27:
Image handler if fine. I rebuilt both , and they are working fine.

Thanks
Brian

aaronwmorris, 2023-04-03:
Closing this for now.  Let me know if there is anything else I can do.

closed 2023-04-03 by aaronwmorris